tzapu/WiFiManager #1007: How to tell if the WIFI manager just went through AP mode?

KiavashS, 2020-02-10:
I'm writing a SPIFFS module to save and retrieve additional parameters through WifiManager. I'm having trouble working out the logic of how to best know if I should retrieve stored values from the flash or from the parameters entered on the web page. The parameters aren't mandatory, so I can't just check if there is a value.

I'm thinking if I can detect that the WifiManager went into AP mode the last time it booted, then it means values should be retrieved from the parameters. Otherwise, get them from the stored SPIFFS file.

Any thoughts? Is there a way to tell if on a given boot up, the AP was used to retrieve values?

lbussy, 2020-02-27:
Assuming the user made a configuration commit, one of the various callbacks would work.

KiavashS, 2020-03-01:
> Assuming the user made a configuration commit, one of the various callbacks would work.

Sorry I don't understand

lbussy, 2020-03-01:
> Sorry I don't understand

Documentation on [callbacks](https://github.com/tzapu/WiFiManager/tree/development#callbacks).