tzapu/WiFiManager #1010: Wifimanger do not remember pass nor SSID in Confriguation mode, and save empty "strings" when closing after changing other parameters?

kbaggen, 2020-02-17:
Hi there, not sure if this is an issue !
Shouldent Wifimanger remmeber the old pass/ssid if I just shnage the mqtt port or whatever?

.Anyone know what to do, e.g. to ensure to display the last used SSID and last use password so when saving other parameters the last used pass/ssis get saved automtically?

Some code to do this whould be nice......if possible! Might be I removed something I shouldent, but all my 5 other parameters remembers from time to time....just not pass nor SSID and it is a bugger to remember setting SSID and pass each time you just wish to change another parameter!

/Klaus

tablatronix, 2020-02-17:
What version?

kbaggen, 2020-02-17:
Ups.....that part I did forget to write....I am using the version Ken Taloy made, so maybe that is the reason?

tablatronix, 2020-02-17:
yes

kbaggen, 2020-02-17:
Just tried with 0.15 and same results.....the SSID nor pass is stored from time to time in confriguation! If resetting ESP it works fine and hence the SSID and pass is saved but jsut not get scripted into confirguation moe....any clue to what might be wrong!

tablatronix, 2020-02-18:
can you explain exactly what you are doing and expecting?

If you save config and leave ssid blank it should keep the one previously saved.  Are you expecting it to reconnect and it does not and you have to reboot?

kbaggen, 2020-02-18:
Yes, that is what I want and assumed, eg. that if you enter confriguaion and do not change of SSI nor pass but only some additinal parameters I expect it to keep the "in-the-patst" set ssid/pass.
 
I have 6-7 parameters I included in wifimanger there sometimes is changed according till the function of the device (it is a temperature/activity logger for fermentation and hence I set setpoint, volumes, gravity of worts, etc. before each fermentation as parameters).

When I just change one parameters  (and hence not changing the SSID nor pass) after reboot the device do not logon. Opposite, if I when I change the parameter remember to set the SSID and pass it functions....so it seems the Wifimanger do not remember the SSID/pass in confriguation from time to time unless they are written each time. All my untoched other parameters are finely remembered.

So if you have an idea where I should look, then, thanks, else I musr expplore. Will start fresh on 0.15 and just check with nofthing besides one parameter.

As said I started out on you script, and ended (not recalling why anymore) using Ken version, so even all besdies the pass/ssid now works, it is a mismatch I guess! 

Thanks for responding, much appriciated.

tablatronix, 2020-02-18:
I suggest you switch to the development branch and start using it, it is the next version.

0.15 should remember your wifi..

Can you provide 0.15 serial logs?

kbaggen, 2020-02-18:
Hi Agian, tried both 0.15 and development branch........I will try a bit more the enxt days and see to provide some serials for it, but my code for the area of connection is:

```
  wifiManager.setTimeout(180);

  //fetches ssid and pass and tries to connect
  //if it does not connect it starts an access point with the specified name
  //here  "AutoConnectAP"
  //and goes into a blocking loop awaiting configuration
  if (!wifiManager.startConfigPortal("CONNECT")) {
    pinMode(PIN_LED, OUTPUT);
    digitalWrite(PIN_LED, LOW); 
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP. restart();
  }
```
Hence, as I use "startConfigPortal" this might be the reason?

aguilaair, 2020-02-18:
I am using the D1 mini, same issue here.

aguilaair, 2020-02-18:
I'm getting inconsistent results. It seems that 1/2 of the time it connects correctly whilst the other half it start the ap

Cat30Mk3, 2020-02-24:
I'm experiencing the exact same issue after switching to latest development version of WiFiManager (Feb 24 2020). Hardware is Adafruit HUZZAH ESP8266 , Connecting to access point with Win10  laptop OR with Apple Tablet IOS 12. gives same results. Did not experience this with latest WiFiManager Master branch.

When I connect to access point and open "Configure WiFi" SSID is correctly populated and Password shows ********
If I attempt to change a custom parameter WITHOUTre-entering ssid & password WM reports:
*WM: [3] Sent wifi save page 
*WM: [2] process connect 
*WM: [2] No ssid, skipping wifi save 
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA
..... 

But if I enter the unchanged ssid & psk it works every time:
*WM: [3] Sent wifi save page 
*WM: [2] process connect 
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping 
*WM: [1] CONNECTED:
*WM: [1] Connecting to NEW AP: JohnAndLinda
*WM: [3] Using Password: 999999999
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] Connect to new AP [SUCCESS] 
*WM: [1] Got IP Address: 
*WM: [1] 192.168.2.43 
Should save config
*WM: [2] disconnect configportal 
*WM: [2] restoring usermode STA+AP
*WM: [2] wifi status: WL_CONNECTED
*WM: [2] wifi mode: STA+AP
*WM: [1] config portal exiting 
connected to wifi after wifiManager AP ...yeey :)
...

tablatronix, 2020-02-24:
What works? What is not working

Cat30Mk3, 2020-02-24:
wifi save page (custom parameter saving) only works when SSID and Password are entered.

wifi save page  (custom parameter saving) does not work if ssid and password are left blank.

Your description from above:
"If you save config and leave ssid blank it should keep the one previously saved. "
is failing.

tablatronix, 2020-02-24:
Did you pull recently? I tested this the other day I think. There is no save if its blank

Cat30Mk3, 2020-02-24:
Yes today from development branch.

Cat30Mk3, 2020-02-24:
It is not actually blank, it is populated with existing ssid and password (shows ********)

tablatronix, 2020-02-24:
Ok so are you saying your wifi is wiped and no longer connects after rebooting?

Or that your savecallback is not being called ?

Set breakaftercpnfig true

tablatronix, 2020-02-24:
Also esp8266 or esp32? You can make a new issue for this if you want

tablatronix, 2020-02-24:
I will take a look asap at this

Cat30Mk3, 2020-02-25:
Sorry if i am not being clear ..  
(1) this is ESP8266 - have not tried on anything else
(2) using On Demand Access Point -- when I change a custom parameter and press Save, 
 [WM] reports "WM: [2] No ssid, skipping wifi save"  BUT SSID and hidden psk were populated and are displayed - 
(3) If I re-enter or pick from scan list the SSID and provide psk and change custom parameter, the save page functions correctly and the custom parameters are being saved and the ESP reboots correctly.

tablatronix, 2020-02-25:
Thats how it is supposed to work. No wifi change so skipping the wifi save

Cat30Mk3, 2020-02-25:
So are you saying that even if the SSIS and PSK dont change I cant trigger a save of custom parameters without touching the SSID & PSK?

Is this a change from current production version?

Cat30Mk3, 2020-02-25:
OK, I went back to existing production (master) and found same functionality - so my mistake -- I agree this was original design. But I question if this makes sense -  if you only need to change a custom parameter why do you have to re-enter ssid and psk ?

Thanks for your patience

tablatronix, 2020-02-25:
you have to set shouldbreakafterconfig right now for it to work, I do not have it refactored yet

kbaggen, 2020-02-25:
Hi there,
what Cat30Mk3 describes is the same I see! And  also propsoe that the saving of a a parameter should NOT need us also to set the SSID/password..........if possible! 

I aosl thought this was a faoult and did not assume it was the normal behavier! 
Anyway...thanks to all for support and disccussing!

kbaggen, 2020-02-25:
REgarding,
"Thats how it is supposed to work. No wifi change so skipping the wifi save"!
That is not what I see.......it seems if entering the poral I need to set SSID and pass else it braks, so even we wdo not change in SSID nor pass, jsut entering and saving even no changing it breaks the connection later!

tablatronix, 2020-02-25:
are you using .15?

Cat30Mk3, 2020-02-25:
Well, for what it's worth, and with the dislaimer that I am C++ incompetent and really don't appreciate the complexities of this remarkable library,  I tried a small edit to the latest development version WiFiManager.cpp which seems to solve my issue: (inserted at line 626) : 
`      if((WiFi_SSID().c_str() != "") && ( _ssid == "")){    //only if truly not blank ..
         _ssid = WiFi_SSID().c_str();   //set internal _ssid to last WiFi_SSID
         _pass = WiFi_psk().c_str();    //same for internal _psk
       }

    // skip wifi if no ssid
`

I tested and was successful with the following:
(1) saves changes to custom parameter without touching SSID or PSK
(2) saves changes to SSID & PSK as previous 
(3) will not save any changes if SSID is truly blank as previous (I'm assuming this is fundamental)

I'm not sure if i've broken anything else or missed cases but it seems solid for my purposes.

tablatronix, 2020-02-25:
which version?

tablatronix, 2020-02-25:
you are posting 2 different issues, 

savecallbacks do not occur if not saving wifi 
```
you have to set     //if this is set, it will exit after config, even if connection is unsuccessful.
    void          setBreakAfterConfig(boolean shouldBreak);
```

erasing wifi credentials if blank is a bug and was fixed in 0.15 and should not exist in development branch

Cat30Mk3, 2020-02-25:
Not sure about the two issues you mention .. I am only focused on the fact that I can not save parameters without touching SSID/PSK.

I should have mentioned that I was unsuccesssful with setBreakAfterConfig(true);  on latest development version.  I tried this again, and get the same result...

With  setBreakAfterConfig(true); It change the way the AP exited , but still failed to save parameters UNLESS  SSID/PSK were touched.

Here's the result of Save without touching SSID/PSK (parameter temp_Label2 was provided value "Attic") but that value never reached the program. You can see the "*WM: [2] No ssid, skipping wifi save " diagnostic print partway through.

*WM: [2] <- HTTP WiFi save  
*WM: [3] Method: POST
*WM: [2] Parameters 
*WM: [2] -------------------- 
*WM: [2] localDeviceName: 1527Rawlings
*WM: [2] remoteDeviceName: RSU-B
*WM: [2] mqttServer: broker.mqtt-dashboard.com
*WM: [2] mqttUserId: 
*WM: [2] mqttPassword: 
*WM: [2] tempLabel_0: Outside
*WM: [2] tempLabel_1: Attic
*WM: [2] tempLabel_2: 
*WM: [2] tempLabel_3: 
*WM: [2] tempLabel_4: 
*WM: [2] tempLabel_5: 
*WM: [2] tempLabel_6: 
*WM: [2] -------------------- 
*WM: [3] Sent wifi save page 
*WM: [2] process connect 
*WM: [2] No ssid, skipping wifi save 
Should save config
*WM: [2] disconnect configportal 
*WM: [2] restoring usermode STA
*WM: [2] wifi status: WL_CONNECTED
*WM: [2] wifi mode: STA
*WM: [1] config portal exiting 
failed to connect to wifiManager and hit timeout

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1392, room 16 
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld

Here's the full Diagnostic print from initiating AP to Reboot with setBreakAfterConfig(true) and not touching SSID/PSK

tablatronix, 2020-02-25:
#1013

tablatronix, 2020-02-25:
works for me

```PHP
*WM: [3] -> captive.apple.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.4.1 
*WM: [3] lastconxresulttmp: WL_NO_SSID_AVAIL
*WM: [3] lastconxresult: WL_NO_SSID_AVAIL
*WM: [2] Scan is cached 5811 ms ago
*WM: [2] <- HTTP WiFi save  
*WM: [3] Method: POST
*WM: [3] Sent wifi save page 
*WM: [2] process connect 
*WM: [2] No ssid, skipping wifi save 
*WM: [1] WiFi/Param save callback 
[CALLBACK] saveCallback fired
*WM: [2] disconnect configportal 
*WM: [2] restoring usermode STA
*WM: [2] wifi status: WL_DISCONNECTED
*WM: [2] wifi mode: STA
*WM: [1] config portal exiting 
failed to connect and hit timeout
Mode: STA
Channel: 1
SSID (5): Shawn
Passphrase (8): 12345678
BSSID set: 0
YES
SSID: 
PASS: 12345678
OTA server at: esp32-a4cf1224c810.local:3232
```

tablatronix, 2020-02-25:
I updated the logging messages a little

Cat30Mk3, 2020-02-26:
Ah Ha! Found it! Your results helped me track it down... now it works like yours..
THANKS!

Here's the issue I had :
when setBreakAfterConfig(true) is used without touching SSID/PSK  
wifiManager.startConfigPortal(wmConfigDb.ssid,wmConfigDb.password) return FALSE,
and when SSID/PSK is touched return TRUE

heres where my code failed .. on return from .startConfigPortal I would force a reset without capturing parameters assuming FALSE was a failure to connect

if (!wifiManager.startConfigPortal(wmConfigDb.ssid,wmConfigDb.password)) {
      Serial.println("failed to connect to wifiManager and hit timeout");
      nonBlockingDelay(3000);
      ESP.reset();
    }

tablatronix, 2020-02-26:
Yeah its a known issue and has always been a problem, I need to replace it all but there is no way to change without breaking 1 thing, either the return result or when the callback occurs.

Cat30Mk3, 2020-02-26:
Thanks again for your help. I can proceed with my next project now. I do use WM extensively  and this is important to me.  Obviously this fix is only temporary as the returned result is no longer useful to detect failed connections. I'll give it some thought as I proceed and if i have any revelations i will share them in #1013. Personally i don't have problem with adapting my code to accommodate your fix, but I can appreciate  the impact you are trying to avoid. I'm happy to test if  that helps I typically work with ESP8266 and ESP32.