tzapu/WiFiManager #1014: Setting the hostname

fmarzocca, 2020-02-26:
I am wondering when is the right point to set the hostname ( `WiFi.hostname(BASE_HOST_NAME);`)of the esp8266 to avoid it show itself in the LAN with ESP_xxxxx? 

```
WiFiManager wifiManager;
wifiManager.setConfigPortalTimeout(300);
if (!wifiManager.autoConnect(BASE_HOST_NAME)) {
		delay(9000);
		ESP.reset();
	}
if (WiFi.status() != WL_CONNECTED) {
		delay(3000);
		ESP.reset();
	}
```

tablatronix, 2020-02-27:
Well if you set the mode to OFF it will remember it and not auto start the ap

fmarzocca, 2020-02-27:
@tablatronix , sorry but it wasn't clear to me. Which mode should I set to OFF? I need the AP the first time.... Where should I include WiFi.hostname(BASE_HOST_NAME) ??

tablatronix, 2020-02-27:
setup()
WiFi.mode(WIFI_OFF);

fmarzocca, 2020-02-27:
I am confused... how can WiFi.mode() be
able to set the device hostname?

- Sent from my mobile phone. Sorry for typos -

tablatronix, 2020-02-27:
Your problem has nothing to do with hostname

fmarzocca, 2020-02-27:
Sorry, maybe I wasn't clear in my question.

My device is working very well with WifiManager. The first time it starts in AP mode, I set the credentials and then every time it starts again it connects to the WiFi AP. 
My problem is that at this point the device is showing itself on the LAN with ESP-xxxxx hostname, while I'd like to assign it to a more specific hostname.

How to assign it the hostname and where in the sequence of setup() instructions?

tablatronix, 2020-02-27:
oh are you talking about mdns hostnames? I thought you mean the ssid was showing up.
Sorry I misunderstood

If you are using autoconnect, you can set it before you call autoconnect, it doesn't get set until you call autoconnect, there have been some changes to mdns, I think you might have to service it in loop now, not sure.

tablatronix, 2020-02-27:
depending on the version of your libs, you probably have to add this to your loop to get hostname resolution to work properly

```C++
  // Allow MDNS processing
  MDNS.update();
```

closed 2020-02-27 by fmarzocca

tablatronix, 2020-02-27:
Here is the code if you are curious or want to do it yourself in your code,

```c++
  if((String)_hostname != ""){
    DEBUG_WM(DEBUG_VERBOSE,"Setting hostname:",_hostname);
    bool res = true;
    #ifdef ESP8266
      res = WiFi.hostname(_hostname);
      #ifdef ESP8266MDNS_H
        DEBUG_WM(DEBUG_VERBOSE,"Setting MDNS hostname");
        if(MDNS.begin(_hostname)){
          MDNS.addService("http", "tcp", 80);
        }
      #endif
    #elif defined(ESP32)
      // @note hostname must be set after STA_START
      delay(200); // do not remove, give time for STA_START
      res = WiFi.setHostname(_hostname);
      #ifdef ESP32MDNS_H
        DEBUG_WM(DEBUG_VERBOSE,"Setting MDNS hostname");
        if(MDNS.begin(_hostname)){
          MDNS.addService("http", "tcp", 80);
        }
      #endif
    #endif
```

thijstriemstra, 2025-01-22:
For an ESP32 I only had to:

```cpp
#include <ESPmDNS.h>
```

and add `wifiManager.setHostname("MyHostname");` in `setup()`