tzapu/WiFiManager #1015: Captive portal doesn't pop up after scan QR code

bfaliszek, 2020-02-27:
#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [X] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [X] ESP32

WifiManager works without any problem if you have selected network networks in Settings> WiFi. A captive portal appears and everything works as it does.
I generated a QR code with the name of my WiFi network at https://qifi.org.
After scanning the code on the iPhone, nothing happens. Captive portal does not appear. Only after going into Settings>WiFi does the captive portal pop up.
I just don't know if it's a problem with WiFiManager or iOS.

tablatronix, 2020-02-27:
Doesn't seem to work for me at all, says unable to join after I scan, IOS

tablatronix, 2020-02-27:
Oops I had it anonymous, let me try again

tablatronix, 2020-02-27:
It only works for me if I accept, close the camera goto settings wifi then it pops up

tablatronix, 2020-02-27:
adding qr code for testing dev examples
`  if(!wm.autoConnect("WM_AutoConnectAP","12345678")) {`

[image]

bfaliszek, 2020-02-27:
> It only works for me if I accept, close the camera goto settings wifi->then it auto pops the portal. It seems to be waiting but gives you no notification that it needs further info etc.. Seems buggy

It works exactly the same for me.

adamgweeks, 2021-07-05:
I'm working with a completely different captive network system and I'm getting the same effect in iOS.  I think it's just an iOS security thing?

If anyone confidently says I'm wrong that would be very useful to me! :)