tzapu/WiFiManager #1018: WM fails to reconnect to known AP after wake up from sleep

MauiKano, 2020-03-07:
I use the latest development branch, cloned it on March 7th 2020 with
git clone -b development https://github.com/tzapu/WiFiManager.git
HW is a LiLygo Wristband (https://github.com/Xinyuan-LilyGO/LilyGO-T-Wristband)
WM configuring works but when I send my HW to sleep and rewake it WM has issues with reconnecting to the know AP. It remembers credentials correctly however but fails back to the setup procedure with the access point in the ESP for initial configuration.

This is what I see of DEBUG messages:

eFuse Vref:1107 mV*WM: [1] AutoConnect
*WM: [2] ESP32 event handler enabled
*WM: [2] Connecting as wifi client...
*WM: [1] STA static IP:
*WM: [2] setSTAConfig static ip not set
*WM: [3] WIFI station disconnect
E (26425) wifi: esp_wifi_disconnect 1153 wifi not start
*WM: [1] Connecting to SAVED AP: xxxxx
*WM: [3] Using Password: xxxxxxxxxx
*WM: [3] WiFi station enable
E (26506) wifi: esp_wifi_connect 1145 wifi not start
*WM: [1] connectTimeout not set, ESP waitForConnectResult...
*WM: [2] Connection result: Unknown
*WM: [3] lastconxresult: Unknown
*WM: [1] AutoConnect: FAILED
*WM: [2] Starting Config Portal
*WM: [3] WIFI station disconnect
E (36509) wifi: esp_wifi_disconnect 1153 wifi not start
*WM: [3] WiFi station enable
*WM: [2] Disabling STA
*WM: [2] Enabling AP
*WM: [1] StartAP with SSID:  T-Wristband
*WM: [2] AP has anonymous access!
*WM: [1] SoftAP Configuration
*WM: [1] --------------------
*WM: [1] ssid:             T-Wristband

tablatronix, 2020-03-07:
This looks like wifi is not initializing or has not had time to.

Do you get similar issues with plain sketch and starting wifi ?

E (26425) wifi: esp_wifi_disconnect 1153 wifi not start
*WM: [1] Connecting to SAVED AP: xxxxx
*WM: [3] Using Password: xxxxxxxxxx
*WM: [3] WiFi station enable
E (26506) wifi: esp_wifi_connect 1145 wifi not startp

MauiKano, 2020-03-08:
Shame on me: It actually happens only when I set
#define FACTORY_HW_TEST     //! Test RTC and WiFi scan when enabled
Without that setting WiFi works as expected. I have not yet diged in to understand the difference. However. I have a work-around so this can be closed.

closed 2020-03-08 by MauiKano