tzapu/WiFiManager #1019: Config Portal fails but Auto Connect works

tgreening, 2020-03-09:
### Basic Infos
  wifiManager.setTimeout(120);
  if (!wifiManager.startConfigPortal("OnDemandAP"))
#### Hardware
**WiFimanager Branch/Release:**
- [x ] Master
- [ ] Development

**Esp8266/Esp32:**
- [x ] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [x ] 2.4.0
- [ ] staging (master/dev)

### Description
If you setup the on demand config portal it will fail to connect to the wifi but changing to auto connect and it will connect just fine.   
Problem description

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch
#include <FS.h> //this needs to be first, or it all crashes and burns...

#include <ESP8266WiFi.h> //https://github.com/esp8266/Arduino

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h> //https://github.com/tzapu/WiFiManager
#include <WiFi.h>
#include <ArduinoJson.h> //https://github.com/bblanchon/ArduinoJson

//define your default values here, if there are different values in config.json, they are overwritten.
char mqtt_server[40];
char mqtt_port[6] = "8080";
char blynk_token[34] = "YOUR_BLYNK_TOKEN";

IPAddress server(74,125,115,105);  // Google

// Initialize the client library
WiFiClient client;

//flag for saving data
bool shouldSaveConfig = false;

//callback notifying us of the need to save config
void saveConfigCallback()
{
  Serial.println("Should save config");
  shouldSaveConfig = true;
}

void setup()
{
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println();

  //clean FS, for testing
  //SPIFFS.format();

  //read configuration from FS json
  Serial.println("mounting FS...");

  if (SPIFFS.begin())
  {
    Serial.println("mounted file system");
    if (SPIFFS.exists("/config.json"))
    {
      //file exists, reading and loading
      Serial.println("reading config file");
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile)
      {
        Serial.println("opened config file");
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);

        configFile.readBytes(buf.get(), size);
        StaticJsonDocument<500> jsonBuffer;
        auto error = deserializeJson(jsonBuffer, buf.get());
        if (error)
        {
          Serial.print(F("deserializeJson() failed with code "));
          Serial.println(error.c_str());
          return;
        }
        else
        {
          Serial.println(F("deserializeJson() successful:"));

          serializeJsonPretty(jsonBuffer, Serial);
          Serial.println();

          strcpy(mqtt_server, jsonBuffer["mqtt_server"]);
          strcpy(mqtt_port, jsonBuffer["mqtt_port"]);
          strcpy(blynk_token, jsonBuffer["blynk_token"]);
         }
        configFile.close();
      }
    }
  }
  else
  {
    Serial.println("failed to mount FS");
  }
  //end read

  // The extra parameters to be configured (can be either global or just in the setup)
  // After connecting, parameter.getValue() will get you the configured value
  // id/name placeholder/prompt default length
  WiFiManagerParameter custom_mqtt_server("server", "mqtt server", mqtt_server, 40);
  WiFiManagerParameter custom_mqtt_port("port", "mqtt port", mqtt_port, 6);
  WiFiManagerParameter custom_blynk_token("blynk", "blynk token", blynk_token, 32);

  //WiFiManager
  //Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wifiManager;
  //wifiManager.resetSettings();

  //set config save notify callback
  wifiManager.setSaveConfigCallback(saveConfigCallback);

  //set static ip
  //wifiManager.setSTAStaticIPConfig(IPAddress(10, 0, 1, 99), IPAddress(10, 0, 1, 1), IPAddress(255, 255, 255, 0));

  //add all your parameters here
  wifiManager.addParameter(&custom_mqtt_server);
  wifiManager.addParameter(&custom_mqtt_port);
  wifiManager.addParameter(&custom_blynk_token);

  //reset settings - for testing
  //wifiManager.resetSettings();

  //set minimu quality of signal so it ignores AP's under that quality
  //defaults to 8%
  //wifiManager.setMinimumSignalQuality();

  //sets timeout until configuration portal gets turned off
  //useful to make it all retry or go to sleep
  //in seconds
  //wifiManager.setTimeout(120);

  //fetches ssid and pass and tries to connect
  //if it does not connect it starts an access point with the specified name
  //here  "AutoConnectAP"
  //and goes into a blocking loop awaiting configuration
  wifiManager.setTimeout(120);
  if (!wifiManager.startConfigPortal("OnDemandAP"))
  //if (!wifiManager.autoConnect("OnDemandAP"))
  {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(5000);
  }

  while (WiFi.status() != WL_CONNECTED) { // Wait for the Wi-Fi to connect
    delay(1000);
    Serial.print(".");
  }

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");

  //read updated parameters
  strcpy(mqtt_server, custom_mqtt_server.getValue());
  strcpy(mqtt_port, custom_mqtt_port.getValue());
  strcpy(blynk_token, custom_blynk_token.getValue());

  //save the custom parameters to FS
  if (shouldSaveConfig)
  {
    Serial.println("saving config");

    StaticJsonDocument<500> jsonBuffer;
    jsonBuffer["mqtt_server"] = mqtt_server;
    jsonBuffer["mqtt_port"] = mqtt_port;
    jsonBuffer["blynk_token"] = blynk_token;
    

    File configFile = SPIFFS.open("/config.json", "w");
    if (!configFile)
    {
      Serial.println("failed to open config file for writing");
    }

    serializeJson(jsonBuffer, Serial);
    Serial.println();
    serializeJson(jsonBuffer, configFile);
    configFile.close();
    //end save
  }

  Serial.println("local ip");
  Serial.println(WiFi.localIP());
  Serial.println("Connected to wifi");
  Serial.println("\nStarting connection...");
  // if you get a connection, report back via serial:
  if (client.connect(server, 80))
  {
    Serial.println("connected");
    // Make a HTTP request:
    client.println("GET /search?q=arduino HTTP/1.0");
    client.println();
    } else {
      Serial.println("Failed....");
    }
}

void loop()
{
  // put your main code here, to run repeatedly:
}

```

mounting FS...
mounted file system
reading config file
opened config file
deserializeJson() successful:
{
  "mqtt_server": "192.168.0.99",
  "mqtt_port": "8080",
  "blynk_token": "1A"
}
*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: blynk
*WM:
*WM: Configuring access point... 
*WM: OnDemandAP
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
failed to connect and hit timeout

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1392, room 16
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld

```
messages here
```

tablatronix, 2020-03-09:
Why would starting the config portal connect to wifi?

tgreening, 2020-03-09:
I am adding configuration parameters, like MQTT or Thinkgspeak API Key, and the config portal allows that.  It's a common task as far as I know.

tablatronix, 2020-03-09:
startconfigportal does not connect to wifi

tgreening, 2020-03-09:
Then why does it let you config the WiFi settings?  See code above.  What is the correct flow.  

BTW, previous versions of WiFiManager with this exact same code worked to connect to WiFi using config portal

tgreening, 2020-03-09:
Also the example for config Portal shows this:

```
if (!wifiManager.startConfigPortal("OnDemandAP")) {
      Serial.println("failed to connect and hit timeout");
      delay(3000);
      //reset and try again, or maybe put it to deep sleep
      ESP.reset();
      delay(5000);
    }

    //if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");
  }
```

Is this example wrong?

tablatronix, 2020-03-09:
Yeah but you have to start wifi still if you want to connect to wifi.
ESP8266 will auto connect to wifi on is own.

But you can use autoconnect and `setEnableConfigPortal`

```C++
    // if true (default) then start the config portal from autoConnect if connection failed
    void          setEnableConfigPortal(boolean enable);
```

tgreening, 2020-03-09:
By using setEnableConfigPortal(true) will it always stop and wait for the portal so I can enter my configuration?  Even if the Wifi credentials have been entered?

tablatronix, 2020-03-09:
No only if it cannot connect does autoconnect start configportal. If you disable that false the. It will not start the configportal automatically ever.

tgreening, 2020-03-09:
I know that wifiManager.startConfigPortal("OnDemandAP") used to show the config portal every time and connect to WiFI.  Something changed in the last two versions.  

If I did: wifiManager.startConfigPortal("OnDemandAP") then wifiManager.autoConnect("OnDemandAP") will that work?

tablatronix, 2020-03-09:
Not really sure what you want to do maybe create a new issue and explain it.

Starting the configportal all the time, only if first time, only if wifi fails, only ondemand button etc? 

Autoconnect connects and start cp if it fails.
Esp will auto connect usually even if you do not call autoconnect provided you gave set the wifi mode properly

tgreening, 2020-03-09:
Maybe I haven't explained it correctly.  

I want what I saw in prior versions of WifIManager.  Namely, the config portal starts every time the ESP8266 restarts.  If the timeout, set in code, passes it should try and use existing credentials to connect to WiFi.  If I enter new credentials in the portal it should use those and connect to WiFi.  As long as I have valid credentials to connect to WiFi startConfigPortal should connect to the WiFi.

What I have seen is that if I start the config portal it will not connect to WiFi when I have valid credentials.  If I change the connect method to autoConnect, and don't change the credentials or touch anything, it connects to WiFi just fine, so I know the saved credentials are fine.  I believe that there is something going on in the config portal that is preventing it from working.  It used to work just fine.

Let me know what I can do to help.

tablatronix, 2020-03-10:
add this to setup and let me know if it changes.

wifiManager.startConfigPortal("OnDemandAP");
WiFi.begin();

tablatronix, 2020-03-10:
I also suggest you use the development branch now

tgreening, 2020-03-11:
I took the development branch and sometimes when I configure the WiFi credentials it connects.  But not always.  If I don't re-enter the credentials and it hits the timeout it fails to connect every time.  Especially if I reboot the ESP8266, Wemos D1 mini - here's the serial output:
`20:24:47.670 -> mounting FS...
20:24:47.740 -> mounted file system
20:24:47.740 -> reading config file
20:24:47.740 -> opened config file
20:24:47.740 -> deserializeJson() successful:
20:24:47.774 -> {
20:24:47.774 ->   "mqtt_server": "192.168.0.1",
20:24:47.774 ->   "mqtt_port": "8080",
20:24:47.774 ->   "blynk_token": "1AB"
20:24:47.774 -> }
20:24:47.774 -> *WM: [1] <form action='/wifi'    method='get'><button>Configure WiFi</button></form><br/>
20:24:47.774 -> <form action='/info'    method='get'><button>Info</button></form><br/>
20:24:47.774 -> <form action='/exit'    method='get'><button>Exit</button></form><br/>
20:24:47.774 ->  
20:24:47.774 -> *WM: [3] allocating params bytes: 20
20:24:47.808 -> *WM: [2] Added Parameter: server
20:24:47.808 -> *WM: [2] Added Parameter: port
20:24:47.808 -> *WM: [2] Added Parameter: blynk
20:24:47.808 -> *WM: [2] Starting Config Portal 
20:24:47.808 -> *WM: [3] WIFI station disconnect 
20:24:47.808 -> *WM: [3] WiFi station enable 
20:24:47.808 -> *WM: [2] Disabling STA 
20:24:47.808 -> *WM: [2] Enabling AP 
20:24:47.808 -> *WM: [1] StartAP with SSID:  OnDemandAP
20:24:48.323 -> *WM: [2] AP has anonymous access! 
20:24:48.323 -> *WM: [1] SoftAP Configuration 
20:24:48.323 -> *WM: [1] -------------------- 
20:24:48.323 -> *WM: [1] ssid:             OnDemandAP
20:24:48.323 -> *WM: [1] password:         
20:24:48.323 -> *WM: [1] ssid_len:         10
20:24:48.323 -> *WM: [1] channel:          1
20:24:48.323 -> *WM: [1] authmode:        
20:24:48.323 -> *WM: [1] ssid_hidden:     
20:24:48.323 -> *WM: [1] max_connection:   4
20:24:48.323 -> *WM: [1] country:          CN
20:24:48.323 -> *WM: [1] beacon_interval:  100(ms)
20:24:48.323 -> *WM: [1] -------------------- 
20:24:48.836 -> *WM: [1] AP IP address: 192.168.4.1
20:24:48.836 -> *WM: [3] setupConfigPortal 
20:24:48.836 -> *WM: [1] Starting Web Portal 
20:24:48.836 -> *WM: [3] dns server started with ip:  192.168.4.1
20:24:48.836 -> *WM: [2] HTTP server started 
20:24:51.026 -> *WM: [2] WiFi Scan completed in 2185 ms
20:24:51.026 -> *WM: [2] Config Portal Running, blocking, waiting for clients... 
20:25:17.600 -> *WM: [2] Portal Timeout In 90 seconds
20:25:47.606 -> *WM: [2] Portal Timeout In 60 seconds
20:26:17.624 -> *WM: [2] Portal Timeout In 30 seconds
20:26:47.619 -> *WM: [2] Portal Timeout In 0 seconds
20:26:47.791 -> *WM: [1] config portal has timed out 
20:26:47.825 -> *WM: [3] configportal abort 
20:26:47.825 -> *WM: [2] disconnect configportal 
20:26:48.821 -> *WM: [2] restoring usermode STA+AP
20:26:48.821 -> *WM: [2] WiFi Reconnect, was idle 
20:26:48.821 -> *WM: [2] wifi status: WL_DISCONNECTED
20:26:48.821 -> *WM: [2] wifi mode: STA+AP
20:26:48.821 -> *WM: [1] config portal exiting 
20:26:48.821 -> failed to connect and hit timeout
20:26:51.826 -> 
20:26:51.826 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
20:26:51.826 -> 
20:26:51.826 -> load 0x4010f000, len 1392, room 16 
20:26:51.860 -> tail 0
20:26:51.860 -> chksum 0xd0
20:26:51.860 -> csum 0xd0
20:26:51.860 -> v3d128e5c
20:26:51.860 -> ~ld
20:26:51.928 -> 
20:26:51.928 -> mounting FS...
20:26:51.995 -> mounted file system
20:26:51.995 -> reading config file
20:26:51.995 -> opened config file
20:26:51.995 -> deserializeJson() successful:
20:26:51.995 -> {
20:26:51.995 ->   "mqtt_server": "192.168.0.1",
20:26:51.995 ->   "mqtt_port": "8080",
20:26:51.995 ->   "blynk_token": "1AB"
20:26:51.995 -> }
20:26:51.995 -> *WM: [1] <form action='/wifi'    method='get'><button>Configure WiFi</button></form><br/>
20:26:52.030 -> <form action='/info'    method='get'><button>Info</button></form><br/>
20:26:52.030 -> <form action='/exit'    method='get'><button>Exit</button></form><br/>
20:26:52.030 ->  
20:26:52.030 -> *WM: [3] allocating params bytes: 20
20:26:52.030 -> *WM: [2] Added Parameter: server
20:26:52.030 -> *WM: [2] Added Parameter: port
20:26:52.030 -> *WM: [2] Added Parameter: blynk
20:26:52.030 -> *WM: [2] Starting Config Portal 
20:26:52.030 -> *WM: [3] WIFI station disconnect 
20:26:52.030 -> *WM: [3] WiFi station enable 
20:26:52.030 -> *WM: [2] Disabling STA 
20:26:52.064 -> *WM: [2] Enabling AP 
20:26:52.064 -> *WM: [1] StartAP with SSID:  OnDemandAP
20:26:52.541 -> *WM: [2] AP has anonymous access! 
20:26:52.541 -> *WM: [1] SoftAP Configuration 
20:26:52.541 -> *WM: [1] -------------------- 
20:26:52.541 -> *WM: [1] ssid:             OnDemandAP
20:26:52.541 -> *WM: [1] password:         
20:26:52.541 -> *WM: [1] ssid_len:         10
20:26:52.576 -> *WM: [1] channel:          1
20:26:52.576 -> *WM: [1] authmode:        
20:26:52.576 -> *WM: [1] ssid_hidden:     
20:26:52.576 -> *WM: [1] max_connection:   4
20:26:52.576 -> *WM: [1] country:          CN
20:26:52.576 -> *WM: [1] beacon_interval:  100(ms)
20:26:52.576 -> *WM: [1] -------------------- 
20:26:53.054 -> *WM: [1] AP IP address: 192.168.4.1
20:26:53.088 -> *WM: [3] setupConfigPortal 
20:26:53.088 -> *WM: [1] Starting Web Portal 
20:26:53.088 -> *WM: [3] dns server started with ip:  192.168.4.1
20:26:53.088 -> *WM: [2] HTTP server started 
20:26:55.263 -> *WM: [2] WiFi Scan completed in 2185 ms
20:26:55.263 -> *WM: [2] Config Portal Running, blocking, waiting for clients... 
`

tablatronix, 2020-03-11:
When was the last time you erased flash?

That is odd that it is failing, also what esp lib version?

tgreening, 2020-03-11:
I used a brand new Wemos and an old one.  Both failed.
I believe I'm on the latest but I'll check when I get home.
Thansk for all your help!

tablatronix, 2020-03-11:
Try full erase, make sure flash is not corrupt.

Also why is your esp restarting ?

```
20:26:48.821 -> *WM: [2] wifi status: WL_DISCONNECTED
20:26:48.821 -> *WM: [2] wifi mode: STA+AP
20:26:48.821 -> *WM: [1] config portal exiting
20:26:48.821 -> failed to connect and hit timeout
20:26:51.826 ->
20:26:51.826 -> ets Jan 8 2013,rst cause:2, boot mode:(3,6)
20:26:51.826 ->
20:26:51.826 -> load 0x4010f000, len 1392, room 16
20:26:51.860 -> tail 0
20:26:51.860 -> chksum 0xd0
20:26:51.860 -> csum 0xd0
20:26:51.860 -> v3d128e5c
20:26:51.860 -> ~ld
20:26:51.928 ->
20:26:51.928 -> mounting FS...
20:26:51.995 -> mounted file system
20:26:51.995 -> reading config file
20:26:51.995 -> opened config file
```

tgreening, 2020-03-12:
I'm using esp8266 board library version 2.6.3

Here's the results:
Global variables use 30372 bytes (37%) of dynamic memory, leaving 51548 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.8
Serial port COM4
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 4c:11:ae:0d:d1:1a
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Erasing flash (this may take a while)...
Chip erase completed successfully in 5.9s
Compressed 378912 bytes to 267599...

Writing at 0x00000000... (5 %)
Writing at 0x00004000... (11 %)
Writing at 0x00008000... (17 %)
Writing at 0x0000c000... (23 %)
Writing at 0x00010000... (29 %)
Writing at 0x00014000... (35 %)
Writing at 0x00018000... (41 %)
Writing at 0x0001c000... (47 %)
Writing at 0x00020000... (52 %)
Writing at 0x00024000... (58 %)
Writing at 0x00028000... (64 %)
Writing at 0x0002c000... (70 %)
Writing at 0x00030000... (76 %)
Writing at 0x00034000... (82 %)
Writing at 0x00038000... (88 %)
Writing at 0x0003c000... (94 %)
Writing at 0x00040000... (100 %)
Wrote 378912 bytes (267599 compressed) at 0x00000000 in 6.1 seconds (effective 497.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

and the serial monitor:

21:19:26.262 -> mounting FS...
21:19:44.059 -> mounted file system
21:19:44.126 -> *WM: [1] <form action='/wifi'    method='get'><button>Configure WiFi</button></form><br/>
21:19:44.126 -> <form action='/info'    method='get'><button>Info</button></form><br/>
21:19:44.126 -> <form action='/exit'    method='get'><button>Exit</button></form><br/>
21:19:44.126 ->  
21:19:44.126 -> *WM: [3] allocating params bytes: 20
21:19:44.126 -> *WM: [2] Added Parameter: server
21:19:44.126 -> *WM: [2] Added Parameter: port
21:19:44.126 -> *WM: [2] Added Parameter: blynk
21:19:44.159 -> *WM: [2] Starting Config Portal 
21:19:44.159 -> *WM: [3] WiFi station enable 
21:19:44.159 -> *WM: [2] Disabling STA 
21:19:44.159 -> *WM: [2] Enabling AP 
21:19:44.159 -> *WM: [1] StartAP with SSID:  OnDemandAP
21:19:44.636 -> *WM: [2] AP has anonymous access! 
21:19:44.636 -> *WM: [1] SoftAP Configuration 
21:19:44.671 -> *WM: [1] -------------------- 
21:19:44.671 -> *WM: [1] ssid:             OnDemandAP
21:19:44.671 -> *WM: [1] password:         
21:19:44.671 -> *WM: [1] ssid_len:         10
21:19:44.671 -> *WM: [1] channel:          1
21:19:44.671 -> *WM: [1] authmode:        
21:19:44.671 -> *WM: [1] ssid_hidden:     
21:19:44.671 -> *WM: [1] max_connection:   4
21:19:44.671 -> *WM: [1] country:          CN
21:19:44.671 -> *WM: [1] beacon_interval:  100(ms)
21:19:44.671 -> *WM: [1] -------------------- 
21:19:45.181 -> *WM: [1] AP IP address: 192.168.4.1
21:19:45.181 -> *WM: [3] setupConfigPortal 
21:19:45.181 -> *WM: [1] Starting Web Portal 
21:19:45.181 -> *WM: [3] dns server started with ip:  192.168.4.1
21:19:45.181 -> *WM: [2] HTTP server started 
21:19:47.372 -> *WM: [2] WiFi Scan completed in 2183 ms
21:19:47.372 -> *WM: [2] Config Portal Running, blocking, waiting for clients... 
21:19:55.902 -> *WM: [2] Portal Timeout In 108 seconds
21:20:25.917 -> *WM: [2] Portal Timeout In 78 seconds
21:20:55.921 -> *WM: [2] Portal Timeout In 48 seconds
21:21:25.943 -> *WM: [2] Portal Timeout In 18 seconds
21:21:44.166 -> *WM: [1] config portal has timed out 
21:21:44.166 -> *WM: [3] configportal abort 
21:21:44.166 -> *WM: [2] disconnect configportal 
21:21:45.151 -> *WM: [2] restoring usermode AP
21:21:45.151 -> *WM: [2] wifi status: WL_DISCONNECTED
21:21:45.151 -> *WM: [2] wifi mode: AP
21:21:45.151 -> *WM: [1] config portal exiting 
21:21:45.151 -> failed to connect and hit timeout
21:21:48.171 -> 
21:21:48.171 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
21:21:48.171 -> 
21:21:48.171 -> load 0x4010f000, len 1392, room 16 
21:21:48.171 -> tail 0
21:21:48.171 -> chksum 0xd0
21:21:48.171 -> csum 0xd0
21:21:48.171 -> v3d128e5c
21:21:48.171 -> ~ld
21:21:48.239 -> 
21:21:48.239 -> mounting FS...
21:21:48.308 -> mounted file system
21:21:48.342 -> *WM: [1] <form action='/wifi'    method='get'><button>Configure WiFi</button></form><br/>
21:21:48.342 -> <form action='/info'    method='get'><button>Info</button></form><br/>
21:21:48.377 -> <form action='/exit'    method='get'><button>Exit</button></form><br/>
21:21:48.377 ->  
21:21:48.377 -> *WM: [3] allocating params bytes: 20
21:21:48.377 -> *WM: [2] Added Parameter: server
21:21:48.377 -> *WM: [2] Added Parameter: port
21:21:48.377 -> *WM: [2] Added Parameter: blynk
21:21:48.377 -> *WM: [2] Starting Config Portal 
21:21:48.377 -> *WM: [3] WiFi station enable 
21:21:48.377 -> *WM: [2] Disabling STA 
21:21:48.377 -> *WM: [2] Enabling AP 
21:21:48.377 -> *WM: [1] StartAP with SSID:  OnDemandAP
21:21:48.895 -> *WM: [2] AP has anonymous access! 
21:21:48.895 -> *WM: [1] SoftAP Configuration 
21:21:48.895 -> *WM: [1] -------------------- 
21:21:48.895 -> *WM: [1] ssid:             OnDemandAP
21:21:48.895 -> *WM: [1] password:         
21:21:48.895 -> *WM: [1] ssid_len:         10
21:21:48.895 -> *WM: [1] channel:          1
21:21:48.895 -> *WM: [1] authmode:        
21:21:48.895 -> *WM: [1] ssid_hidden:     
21:21:48.895 -> *WM: [1] max_connection:   4
21:21:48.895 -> *WM: [1] country:          CN
21:21:48.895 -> *WM: [1] beacon_interval:  100(ms)
21:21:48.930 -> *WM: [1] -------------------- 
21:21:49.405 -> *WM: [1] AP IP address: 192.168.4.1
21:21:49.405 -> *WM: [3] setupConfigPortal 
21:21:49.405 -> *WM: [1] Starting Web Portal 
21:21:49.405 -> *WM: [3] dns server started with ip:  192.168.4.1
21:21:49.405 -> *WM: [2] HTTP server started 
21:21:51.614 -> *WM: [2] WiFi Scan completed in 2184 ms
21:21:51.614 -> *WM: [2] Config Portal Running, blocking, waiting for clients...

tablatronix, 2020-03-12:
ok now add WiFi.mode(WiFi_STA); to setup

tgreening, 2020-03-12:
> ok now add WiFi.mode(WiFi_STA); to setup

Where?  After configPortal or before?

tablatronix, 2020-03-12:
early in setup

tgreening, 2020-03-13:
Still have the same problem.  Here's my current codes:

`#include <FS.h> //this needs to be first, or it all crashes and burns...

#include <ESP8266WiFi.h> //https://github.com/esp8266/Arduino

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h> //https://github.com/tzapu/WiFiManager
#include <ArduinoJson.h> //https://github.com/bblanchon/ArduinoJson

//define your default values here, if there are different values in config.json, they are overwritten.
char mqtt_server[40];
char mqtt_port[6] = "8080";
char blynk_token[34] = "YOUR_BLYNK_TOKEN";

IPAddress server(74, 125, 115, 105); // Google

// Initialize the client library
WiFiClient client;

//flag for saving data
bool shouldSaveConfig = false;

//callback notifying us of the need to save config
void saveConfigCallback()
{
  Serial.println("Should save config");
  shouldSaveConfig = true;
}

void setup()
{
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println();

  //clean FS, for testing
  //SPIFFS.format();

  //read configuration from FS json
  Serial.println("mounting FS...");

  if (SPIFFS.begin())
  {
    Serial.println("mounted file system");
    if (SPIFFS.exists("/config.json"))
    {
      //file exists, reading and loading
      Serial.println("reading config file");
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile)
      {
        Serial.println("opened config file");
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);

        configFile.readBytes(buf.get(), size);
        StaticJsonDocument<500> jsonBuffer;
        auto error = deserializeJson(jsonBuffer, buf.get());
        if (error)
        {
          Serial.print(F("deserializeJson() failed with code "));
          Serial.println(error.c_str());
          return;
        }
        else
        {
          Serial.println(F("deserializeJson() successful:"));

          serializeJsonPretty(jsonBuffer, Serial);
          Serial.println();

          strcpy(mqtt_server, jsonBuffer["mqtt_server"]);
          strcpy(mqtt_port, jsonBuffer["mqtt_port"]);
          strcpy(blynk_token, jsonBuffer["blynk_token"]);
        }
        configFile.close();
      }
    }
  }
  else
  {
    Serial.println("failed to mount FS");
  }
  //end read

  // The extra parameters to be configured (can be either global or just in the setup)
  // After connecting, parameter.getValue() will get you the configured value
  // id/name placeholder/prompt default length
  WiFiManagerParameter custom_mqtt_server("server", "mqtt server", mqtt_server, 40);
  WiFiManagerParameter custom_mqtt_port("port", "mqtt port", mqtt_port, 6);
  WiFiManagerParameter custom_blynk_token("blynk", "blynk token", blynk_token, 32);

  //WiFiManager
  //Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wifiManager;
  //wifiManager.resetSettings();

  //set config save notify callback
  wifiManager.setSaveConfigCallback(saveConfigCallback);

  //set static ip
  //wifiManager.setSTAStaticIPConfig(IPAddress(10, 0, 1, 99), IPAddress(10, 0, 1, 1), IPAddress(255, 255, 255, 0));

  //add all your parameters here
  wifiManager.addParameter(&custom_mqtt_server);
  wifiManager.addParameter(&custom_mqtt_port);
  wifiManager.addParameter(&custom_blynk_token);

  //reset settings - for testing
  //wifiManager.resetSettings();

  //set minimu quality of signal so it ignores AP's under that quality
  //defaults to 8%
  //wifiManager.setMinimumSignalQuality();

  //sets timeout until configuration portal gets turned off
  //useful to make it all retry or go to sleep
  //in seconds
  //wifiManager.setTimeout(120);

  //fetches ssid and pass and tries to connect
  //if it does not connect it starts an access point with the specified name
  //here "AutoConnectAP"
  //and goes into a blocking loop awaiting configuration
  wifiManager.setTimeout(120);
  if (!wifiManager.startConfigPortal("OnDemandAP"))
    //if (!wifiManager.autoConnect("OnDemandAP"))
  {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(5000);
  }

  while (WiFi.status() != WL_CONNECTED) { // Wait for the Wi-Fi to connect
    delay(1000);
    Serial.print(".");
  }

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");

  //read updated parameters
  strcpy(mqtt_server, custom_mqtt_server.getValue());
  strcpy(mqtt_port, custom_mqtt_port.getValue());
  strcpy(blynk_token, custom_blynk_token.getValue());

  //save the custom parameters to FS
  if (shouldSaveConfig)
  {
    Serial.println("saving config");

    StaticJsonDocument<500> jsonBuffer;
    jsonBuffer["mqtt_server"] = mqtt_server;
    jsonBuffer["mqtt_port"] = mqtt_port;
    jsonBuffer["blynk_token"] = blynk_token;

    File configFile = SPIFFS.open("/config.json", "w");
    if (!configFile)
    {
      Serial.println("failed to open config file for writing");
    }

    serializeJson(jsonBuffer, Serial);
    Serial.println();
    serializeJson(jsonBuffer, configFile);
    configFile.close();
    //end save
  }

  Serial.println("local ip");
  Serial.println(WiFi.localIP());
  Serial.println("Connected to wifi");
  Serial.println("\nStarting connection...");
  // if you get a connection, report back via serial:
  if (client.connect(server, 80))
  {
    Serial.println("connected");
    // Make a HTTP request:
    client.println("GET /search?q=arduino HTTP/1.0");
    client.println();
  } else {
    Serial.println("Failed....");
  }
}

void loop()
{
  // put your main code here, to run repeatedly:
}

`

tgreening, 2020-03-13:
Here's the log output:

20:43:50.537 -> mounting FS...
20:43:50.606 -> mounted file system
20:43:50.708 -> <form action='/info'    method='get'><button>Info</button></form><br/>
20:43:50.708 -> <form action='/exit'    method='get'><button>Exit</button></form><br/>
20:43:50.708 ->  
20:43:50.708 -> *WM: [3] allocating params bytes: 20
20:43:50.708 -> *WM: [2] Added Parameter: server
20:43:50.708 -> *WM: [2] Added Parameter: port
20:43:50.741 -> *WM: [2] Added Parameter: blynk
20:43:50.741 -> *WM: [2] Starting Config Portal 
20:43:50.741 -> *WM: [3] WIFI station disconnect 
20:43:50.741 -> *WM: [3] WiFi station enable 
20:43:50.741 -> *WM: [2] Disabling STA 
20:43:50.741 -> *WM: [2] Enabling AP 
20:43:50.741 -> *WM: [1] StartAP with SSID:  OnDemandAP
20:43:51.250 -> *WM: [2] AP has anonymous access! 
20:43:51.250 -> *WM: [1] SoftAP Configuration 
20:43:51.250 -> *WM: [1] -------------------- 
20:43:51.250 -> *WM: [1] ssid:             OnDemandAP
20:43:51.250 -> *WM: [1] password:         
20:43:51.250 -> *WM: [1] ssid_len:         10
20:43:51.250 -> *WM: [1] channel:          1
20:43:51.250 -> *WM: [1] authmode:        
20:43:51.250 -> *WM: [1] ssid_hidden:     
20:43:51.250 -> *WM: [1] max_connection:   4
20:43:51.250 -> *WM: [1] country:          CN
20:43:51.250 -> *WM: [1] beacon_interval:  100(ms)
20:43:51.284 -> *WM: [1] -------------------- 
20:43:51.757 -> *WM: [1] AP IP address: 192.168.4.1
20:43:51.757 -> *WM: [3] setupConfigPortal 
20:43:51.757 -> *WM: [1] Starting Web Portal 
20:43:51.757 -> *WM: [3] dns server started with ip:  192.168.4.1
20:43:51.757 -> *WM: [2] HTTP server started 
20:43:53.960 -> *WM: [2] WiFi Scan completed in 2184 ms
20:43:53.960 -> *WM: [2] Config Portal Running, blocking, waiting for clients... 
20:43:55.693 -> *WM: [3] -> connectivitycheck.gstatic.com 
20:43:55.693 -> *WM: [2] <- Request redirected to captive portal 
20:44:20.462 -> *WM: [2] Portal Timeout In 90 seconds
20:44:50.486 -> *WM: [2] Portal Timeout In 60 seconds
20:45:20.454 -> *WM: [2] Portal Timeout In 30 seconds
20:45:50.483 -> *WM: [2] Portal Timeout In 0 seconds
20:45:50.754 -> *WM: [1] config portal has timed out 
20:45:50.754 -> *WM: [3] configportal abort 
20:45:50.754 -> *WM: [2] disconnect configportal 
20:45:51.739 -> *WM: [2] restoring usermode STA+AP
20:45:51.739 -> *WM: [2] WiFi Reconnect, was idle 
20:45:51.739 -> *WM: [2] wifi status: WL_DISCONNECTED
20:45:51.739 -> *WM: [2] wifi mode: STA+AP
20:45:51.739 -> *WM: [1] config portal exiting 
20:45:51.773 -> failed to connect and hit timeout
20:45:54.757 -> 
20:45:54.757 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
20:45:54.757 -> 
20:45:54.757 -> load 0x4010f000, len 1392, room 16 
20:45:54.791 -> tail 0
20:45:54.791 -> chksum 0xd0
20:45:54.791 -> csum 0xd0
20:45:54.791 -> v3d128e5c
20:45:54.791 -> ~ld
20:45:54.858 -> 
20:45:54.858 -> mounting FS...
20:45:54.927 -> mounted file system
20:45:54.996 -> *WM: [1] <form action='/wifi'    method='get'><button>Configure WiFi</button></form><br/>
20:45:54.996 -> <form action='/info'    method='get'><button>Info</button></form><br/>
20:45:55.029 -> <form action='/exit'    method='get'><button>Exit</button></form><br/>
20:45:55.029 ->  
20:45:55.029 -> *WM: [3] allocating params bytes: 20
20:45:55.029 -> *WM: [2] Added Parameter: server
20:45:55.029 -> *WM: [2] Added Parameter: port
20:45:55.029 -> *WM: [2] Added Parameter: blynk
20:45:55.029 -> *WM: [2] Starting Config Portal 
20:45:55.029 -> *WM: [3] WIFI station disconnect 
20:45:55.029 -> *WM: [3] WiFi station enable 
20:45:55.029 -> *WM: [2] Disabling STA 
20:45:55.029 -> *WM: [2] Enabling AP 
20:45:55.029 -> *WM: [1] StartAP with SSID:  OnDemandAP
20:45:55.537 -> *WM: [2] AP has anonymous access! 
20:45:55.537 -> *WM: [1] SoftAP Configuration 
20:45:55.537 -> *WM: [1] -------------------- 
20:45:55.537 -> *WM: [1] ssid:             OnDemandAP
20:45:55.537 -> *WM: [1] password:         
20:45:55.537 -> *WM: [1] ssid_len:         10
20:45:55.537 -> *WM: [1] channel:          1
20:45:55.573 -> *WM: [1] authmode:        
20:45:55.573 -> *WM: [1] ssid_hidden:     
20:45:55.573 -> *WM: [1] max_connection:   4
20:45:55.573 -> *WM: [1] country:          CN
20:45:55.573 -> *WM: [1] beacon_interval:  100(ms)
20:45:55.573 -> *WM: [1] -------------------- 
20:45:56.078 -> *WM: [1] AP IP address: 192.168.4.1
20:45:56.078 -> *WM: [3] setupConfigPortal 
20:45:56.078 -> *WM: [1] Starting Web Portal 
20:45:56.078 -> *WM: [3] dns server started with ip:  192.168.4.1
20:45:56.078 -> *WM: [2] HTTP server started 
20:45:58.259 -> *WM: [2] WiFi Scan completed in 2184 ms
20:45:58.259 -> *WM: [2] Config Portal Running, blocking, waiting for clients... 
20:46:24.768 -> *WM: [2] Portal Timeout In 90 seconds
20:46:54.757 -> *WM: [2] Portal Timeout In 60 seconds

tgreening, 2020-03-14:
I have now tried this same code on four different esp8266 chips and it get the same results.  When I add the wifi credentials it connects fine.  If I let the config portal timeout it will not connect to wifi.  Any ideas?

tablatronix, 2020-03-14:
Did you set wifi mode? Printdiag() and see what the mode is?

tgreening, 2020-03-15:
yes to both.

Here's the serial output:
22:48:09.426 -> Mode: STA
22:48:09.426 -> PHY mode: N
22:48:09.426 -> Channel: 1
22:48:09.426 -> AP id: 0
22:48:09.426 -> Status: 1
22:48:09.426 -> Auto connect: 1
22:48:09.426 -> SSID (8): xxxxxxxx
22:48:09.426 -> Passphrase (8): xxxxxxxx
22:48:09.426 -> BSSID set: 0
22:48:09.460 -> mounting FS...
22:48:09.494 -> mounted file system
22:48:09.528 -> *WM: [1] <form action='/wifi'    method='get'><button>Configure WiFi</button></form><br/>
22:48:09.528 -> <form action='/info'    method='get'><button>Info</button></form><br/>
22:48:09.562 -> <form action='/exit'    method='get'><button>Exit</button></form><br/>
22:48:09.562 ->  
22:48:09.562 -> *WM: [3] allocating params bytes: 20
22:48:09.562 -> *WM: [2] Added Parameter: server
22:48:09.562 -> *WM: [2] Added Parameter: port
22:48:09.562 -> *WM: [2] Added Parameter: blynk
22:48:09.562 -> Mode: STA
22:48:09.562 -> PHY mode: N
22:48:09.562 -> Channel: 1
22:48:09.562 -> AP id: 0
22:48:09.562 -> Status: 1
22:48:09.562 -> Auto connect: 1
22:48:09.562 -> SSID (8): xxxxxxxx
22:48:09.562 -> Passphrase (8): xxxxxx
22:48:09.596 -> BSSID set: 0
22:48:09.596 -> *WM: [2] Starting Config Portal 
22:48:09.596 -> *WM: [3] WIFI station disconnect 
22:48:09.596 -> *WM: [3] WiFi station enable 
22:48:09.596 -> *WM: [2] Disabling STA 
22:48:09.596 -> *WM: [2] Enabling AP 
22:48:09.596 -> *WM: [1] StartAP with SSID:  OnDemandAP
22:48:10.102 -> *WM: [2] AP has anonymous access! 
22:48:10.102 -> *WM: [1] SoftAP Configuration 
22:48:10.102 -> *WM: [1] -------------------- 
22:48:10.102 -> *WM: [1] ssid:             OnDemandAP
22:48:10.102 -> *WM: [1] password:         
22:48:10.102 -> *WM: [1] ssid_len:         10
22:48:10.102 -> *WM: [1] channel:          1
22:48:10.102 -> *WM: [1] authmode:        
22:48:10.102 -> *WM: [1] ssid_hidden:     
22:48:10.102 -> *WM: [1] max_connection:   4
22:48:10.102 -> *WM: [1] country:          CN
22:48:10.102 -> *WM: [1] beacon_interval:  100(ms)
22:48:10.136 -> *WM: [1] -------------------- 
22:48:10.608 -> *WM: [1] AP IP address: 192.168.4.1
22:48:10.608 -> *WM: [3] setupConfigPortal 
22:48:10.608 -> *WM: [1] Starting Web Portal 
22:48:10.608 -> *WM: [3] dns server started with ip:  192.168.4.1
22:48:10.642 -> *WM: [2] HTTP server started 
22:48:12.810 -> *WM: [2] WiFi Scan completed in 2184 ms
22:48:12.810 -> *WM: [2] Config Portal Running, blocking, waiting for clients... 
22:48:39.350 -> *WM: [2] Portal Timeout In 90 seconds
22:51:13.614 -> *WM: [2] Portal Timeout In 60 seconds
22:49:39.350 -> *WM: [2] Portal Timeout In 30 seconds
22:50:09.377 -> *WM: [2] Portal Timeout In 0 seconds
22:50:09.581 -> *WM: [1] config portal has timed out 
22:50:09.581 -> *WM: [3] configportal abort 
22:50:09.581 -> *WM: [2] disconnect configportal 
22:50:10.600 -> *WM: [2] restoring usermode STA
22:50:10.600 -> *WM: [2] WiFi Reconnect, was idle 
22:50:10.600 -> *WM: [2] wifi status: WL_DISCONNECTED
22:50:10.600 -> *WM: [2] wifi mode: STA
22:50:10.600 -> *WM: [1] config portal exiting 
22:50:10.600 -> Mode: STA
22:50:10.600 -> PHY mode: N
22:50:10.600 -> Channel: 1
22:50:10.600 -> AP id: 0
22:50:10.600 -> Status: 1
22:50:10.600 -> Auto connect: 1
22:50:10.600 -> SSID (8): xxxxxxxx
22:50:10.600 -> Passphrase (8): xxxxxxxx
22:50:10.600 -> BSSID set: 0
22:50:10.633 -> failed to connect and hit timeout
22:50:13.619 -> 
22:50:13.619 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
22:50:13.619 -> 
22:50:13.619 -> load 0x4010f000, len 1392, room 16 
22:50:13.619 -> tail 0
22:50:13.619 -> chksum 0xd0
22:50:13.619 -> csum 0xd0
22:50:13.653 -> v3d128e5c
22:50:13.653 -> ~ld
22:50:13.687 ->

tablatronix, 2020-03-15:
So strange. Maybe we need to add a delay?

tgreening, 2020-03-16:
Do a delay before starting the config portal or after?  Maybe checking the wifi status?

tgreening, 2020-03-19:
I added a delay before starting the config portal and it still fails.

I even tried looping retrying to see if it would eventually connect...still no connection.

Here's a code snippet:
`
 wifiManager.setTimeout(45);
  WiFi.printDiag(Serial);
  delay(1000);
  while(!wifiManager.startConfigPortal("OnDemandAP")) {
    Serial.println("Failed trying again...");
  }
  `

tgreening, 2020-03-20:
So I tried something different.

I started the config portal then if it fails to connects try atuoConnect.  See code below:

 wifiManager.setTimeout(45);
  delay(1000);
  if (!wifiManager.startConfigPortal("OnDemandAP")) {
    WiFi.printDiag(Serial);
    if (!wifiManager.autoConnect("OnDemandAP"))
    {
      WiFi.printDiag(Serial);
      Serial.println("failed to connect and hit timeout");
      delay(3000);
      //reset and try again, or maybe put it to deep sleep
      ESP.reset();
      delay(5000);
    }
  }

The logs show after the config portal the ssid and passphrase are gone.  Is config portal erasing it?

19:36:32.626 -> Mode: STA
19:36:32.626 -> PHY mode: N
19:36:32.626 -> Channel: 1
19:36:32.626 -> AP id: 0
19:36:32.626 -> Status: 1
19:36:32.626 -> Auto connect: 1
19:36:32.626 -> SSID (8): xxxxxx  **-correct**
19:36:32.626 -> Passphrase (8): xxxxxx **- correct**
19:36:32.626 -> BSSID set: 0
19:36:32.626 -> mounting FS...
19:36:32.660 -> mounted file system
19:36:32.660 -> reading config file
19:36:32.694 -> opened config file
19:36:32.694 -> deserializeJson() successful:
19:36:32.694 -> {
19:36:32.694 ->   "mqtt_server": "192",
19:36:32.694 ->   "mqtt_port": "8080",
19:36:32.694 ->   "blynk_token": "YOUR_BLYNK_TOKEN"
19:36:32.694 -> }
19:36:32.694 -> *WM: Adding parameter
19:36:32.694 -> *WM: server
19:36:32.694 -> *WM: Adding parameter
19:36:32.694 -> *WM: port
19:36:32.694 -> *WM: Adding parameter
19:36:32.694 -> *WM: blynk
19:36:33.679 -> *WM: 
19:36:33.679 -> *WM: Configuring access point... 
19:36:33.713 -> *WM: OnDemandAP
19:36:34.189 -> *WM: AP IP address: 
19:36:34.189 -> *WM: 192.168.4.1
19:36:34.189 -> *WM: HTTP server started
19:37:18.709 -> Mode: AP
19:37:18.709 -> PHY mode: N
19:37:18.709 -> Channel: 1
19:37:18.709 -> AP id: 0
19:37:18.709 -> Status: 255
19:37:18.709 -> Auto connect: 1
19:37:18.709 -> SSID (0):  **---empty**
19:37:18.709 -> Passphrase (0):  **--empty**
19:37:18.709 -> BSSID set: 0
19:37:18.709 -> *WM: 
19:37:18.709 -> *WM: AutoConnect
19:37:18.709 -> *WM: Connecting as wifi client...
19:37:18.709 -> *WM: Status:
19:37:18.709 -> *WM: 0
19:37:18.709 -> *WM: No saved credentials
19:37:18.709 -> *WM: Connection result: 
19:37:18.709 -> *WM: 0
19:37:18.709 -> *WM: 
19:37:18.709 -> *WM: Configuring access point... 
19:37:18.709 -> *WM: OnDemandAP
19:37:19.217 -> *WM: AP IP address: 
19:37:19.217 -> *WM: 192.168.4.1
19:37:19.217 -> *WM: HTTP server started

tablatronix, 2020-03-21:
The most success people have is doing a disconnect first. There is something in the sdk that gets stuck here and no one has been able to narrow it down. Ill do some testing but it might be router specific

tablatronix, 2020-03-21:
Let me see what could be erasing it,

tablatronix, 2020-03-21:
Try setting _shouldBreakAfterConfig

tablatronix, 2020-03-21:
We are doing this

```CPP
  if(!WiFi.isConnected()){
    WiFi.persistent(false);
    // disconnect sta, start ap
    WiFi.disconnect(); //  this alone is not enough to stop the autoconnecter
    WiFi.mode(WIFI_AP);
    WiFi.persistent(true);
  }
```
But it should not be erasing creds since it toggles persistent..
Do the creds come back if you reboot ?