tzapu/WiFiManager #1020: Question : Esp8266 12E

Ruandv, 2020-03-12:
Hi,

I am using the WiFiManager to save a few of my settings to SPIFF.
I noticed that when i store this type of data then it saves fine to the SPIFF
<pre>
*WM: [2] <- HTTP WiFi save  
*WM: [3] Method: POST
*WM: [2] Parameters 
*WM: [2] -------------------- 
*WM: [2] APName: Deployment Blinker
*WM: [2] FireBaseRoot: https://fakeConnection.firebaseio.com/
*WM: [2] Secret: NG2ekdzPZ6Zz1c05rNhCDN6si7CvRcWMTmSsdiAf
*WM: [2] Organization: BA
*WM: [2] SystemId: 1439
*WM: [2] -------------------- 
...
...
...
*WM: [1] Got IP Address: 
*WM: [1] 192.168.1.34 
saving config
Values retrieved from Variables
firebase_Root	https://fakeConnection.firebaseio.com/
firebase_Secret	NG2ekdzPZ6Zz1c05rNhCDN6si7CvRcWMTmSsdiAf
AP_Names	Deployment Blinker
firebase_Organization	BA
firebase_SystemId	1439
Values that will be saved to config.json
{"firebase_Root":"https://fakeConnection.firebaseio.com/","firebase_Secret":"NG2ekdzPZ6Zz1c05rNhCDN6si7CvRcWMTmSsdiAf","AP_Names":"Deployment Blinker","firebase_Organization":"BA","firebase_SystemId":"1439"}
*WM: [2] disconnect configportal 
*WM: [2] restoring usermode STA
*WM: [2] wifi status: WL_CONNECTED
*WM: [2] wifi mode: STA
*WM: [1] config portal exiting 
Connected to the WIFI...192.168.1.34
Type reset to Reset ALL the WIFI Settings the ESP
Type restart to restart the ESP
</pre>

but when i try and and change any parameter value to be a bit longer than what was orignally in the data then it saves my data like this 
** note ** I changed the field firebase_Organization from BA to MyOrg
and all of a sudden when i call
` strcpy(firebase_Secret, firebase_Secret_param.getValue());` it returns rg
however as you can see in the line what is written by WM is correct.

**What am i doing wrong**

<pre>
*WM: [2] <- HTTP WiFi save
*WM: [3] Method: POST
*WM: [2] Parameters 
*WM: [2] --------------------  
*WM: [2] APName: Deployment Blinker 
*WM: [2] FireBaseRoot: https://fakeConnection.firebaseio.com/ 
*WM: [2] Secret: NG2ekdzPZ6Zz1c05rNhCDN6si7CvRcWMTmSsdiAf 
*WM: [2] Organization: MyOrg 
*WM: [2] SystemId: 1439 
*WM: [2] -------------------- 
...
...
...
 *WM: [1] Got IP Address: 
*WM: [1] 192.168.1.34  
saving config 
Values retrieved from Variables 
firebase_Root	https://fakeConnection.firebaseio.com/
firebase_Secret	rg
AP_Names	Deployment Blinker
firebase_Organization	MyOrg
firebase_SystemId	1439
Values that will be saved to config.json
{"firebase_Root":"https://fakeConnection.firebaseio.com/","firebase_Secret":"rg","AP_N 
ames":"Deployment Blinker","firebase_Organization":"MyOrg","firebase_SystemId":"1439"}
*WM: [2] disconnect configportal  
*WM: [2] restoring usermode STA 
*WM: [2] wifi status: WL_CONNECTED 
*WM: [2] wifi mode: STA 
*WM: [1] config portal exiting  
Connected to the WIFI...192.168.1.34 
Type reset to Reset ALL the WIFI Settings the ESP 
Type restart to restart the ESP 
</pre>
Attached is my exact .ino(in .txt format) file that i used to generate this data.

[WiFiManager_Issue.txt](https://github.com/tzapu/WiFiManager/files/4326541/WiFiManager_Issue.txt)

Ruandv, 2020-03-16:
Is there anyone that can please assist me with my problem? I am still having issues with the getValue() function.

vaz83, 2020-03-25:
Can you show how are you declaring your parameters?

Ruandv, 2020-03-25:
My whole ino file is attached to the issue.
And it shows how I declare the parameters. 

https://github.com/tzapu/WiFiManager/files/4326541/WiFiManager_Issue.txt

closed 2021-09-27 by Ruandv