tzapu/WiFiManager #1021: How can I return to Ap mode when I press the reset button?

PozitronikTech, 2020-03-13:
How can I return to Ap mode when I press the reset button?

**WiFimanager Branch/Release:**
- [x] Master

**Esp8266/Esp32:**
- [x] ESP8266

**Hardware:			ESP-12e, esp01, esp25**
- [x] ESP12 E/F/S (nodemcu, wemos, feather)

** Using esp version
 - 2.2.2-dev(38a443e)

Ap Mode  -> STA Mode  -> Web Server -> Void Loop (server->handleClient(); or button.read();)
It starts in Ap mode.
I include Esp8266 in a network and the system is rebooting.
When it restarts (in STA mode) the Web server gets up so that I can control the devices with json data.

I will put the device in a box. Outside this box, there will be a button to reset the device. When I press the button, I want to delete the SSID and password in the eeprom and start in AP mode.
I have tried many methods.
Pressing the button continues to connect to the network, starting with the information in memory each time. Do you have a solution suggestion for this? We can do without going out of the wifimanager library. Thank you

### Settings in IDE

Module:  	NodeMcu  1.0 (ESP-12e Module)
Builtin led: 2
upload speed : 115200
cpu freq: 80mhz
flash size: 4mb(fs:2mb ota: ~1019kb)
debug port: disabled
debug level: noting
ivip varinat: v2 lover memory
vtables: flash
exceptions: legacy (new can return nullptr)
erase flash : only sketch
ssl support: all ssl ciphers (most compatible)
port: 8 com

** Home Automation

//SPI
#include <SPI.h>

// Esp8266 ana kütüphanesi
#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino
#include <EasyButton.h>

//Dns, web server bir de Otomatik bağlanmayı sağlayan wifi manager kütüphanesi
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include "WiFiManager.h"         //https://github.com/tzapu/WiFiManager

std::unique_ptr<ESP8266WebServer> server;

#include <ArduinoJson.h>

// Bu kısımda dijital pin 4 için giriş olarak ayarlanıyor.
#define ButtonReset 2 // dijital 4 e denk geliyor.
EasyButton button(ButtonReset);

//Bu zaman aşımı kadar ağa bağlanmaya çalışır. Bağlanamadığında hotspot yayını yapan ağ ayağa kaldırır.
//int timeoutDelayTime = 30;

//Sunucu çalıştığında aldığı ip adresi üzerinden yayın yaptığı web sayfası içeriği
const String postForms = "<html>\
  <head>\
    <title>ESP8266 Web Server POST handling</title>\
    <meta Content-Type: application/json; charset=utf-8>\
    <style>\
[... 252 lines cut ...]
09:27:37.244 -> scandone
09:27:38.202 -> state: 0 -> 2 (b0)
09:27:38.202 -> state: 2 -> 3 (0)
09:27:38.270 -> state: 3 -> 5 (10)
09:27:38.270 -> add 0
09:27:38.270 -> aid 1
09:27:38.270 -> cnt 
09:27:38.270 -> 
09:27:38.270 -> connected with HomeNetwork, channel 1
09:27:38.340 -> dhcp client start...
09:27:41.368 -> ip:192.168.43.245,mask:255.255.255.0,gw:192.168.43.1
09:27:41.368 -> *WM: Connection result: 
09:27:41.368 -> *WM: 3
09:27:41.368 -> station: xx:xx:xx:xx:xx:x leave, AID = 1
09:27:41.368 -> rm 1
09:27:41.368 -> bcn 0
09:27:41.368 -> del if1
09:27:41.368 -> pm open,type:2 0
09:27:41.368 -> mode : sta(xx:xx:xx:xx:xx:x)
09:27:41.402 -> bağlantı başarılı.
09:27:41.402 -> HTTP sunucu başlatılıyor
09:27:41.436 -> Alınan IP adresi: 192.168.43.245
09:27:41.436 -> *WM: freeing allocated params!
09:28:03.113 -> Butona basıldı.
09:28:03.113 -> button press but how I can :)

PozitronikTech, 2020-03-13:
All information is deleted when you clear the Eeprom memory. Thus, it can be started in Ap mode.
How do we do this through wifimanager?

tablatronix, 2020-03-13:
or just startConfigPortal, its in the ondemandconfigportal examples...

closed 2020-03-16 by PozitronikTech

PozitronikTech, 2020-03-16:
Thank u