tzapu/WiFiManager #1022: error: no matching function for call to 'WiFiManager::WiFiManager(Stream&)'

globus1000, 2020-03-13:
Howdy!

I am not a connoisseur of the arduino, I only rebuild projects that already work. In this case "WeMos-Matrix-Shield" by Charles Hallard (https://github.com/hallard/WeMos-Matrix-Shield), based on myClock by them (https://github.com/dragondaud/myClock). When compiling, the following error always occurs with the WiFi.ino, which cannot be solved with google either:

Arduino: 1.8.9 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), dtr (aka nodemcu), 26 MHz, 40MHz, DOUT (compatible), 1MB (FS:64KB OTA:~470KB), 2, nonos-sdk 2.2.1+100 (190703), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Jürgen\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Jürgen\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Jürgen\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=legacy,ssl=all,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=1M64,led=2,sdk=nonosdk_190703,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -ide-version=10809 -build-path C:\Users\Jürgen\AppData\Local\Temp\arduino_build_350357 -warnings=none -build-cache C:\Users\Jürgen\AppData\Local\Temp\arduino_cache_675892 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Jürgen\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=C:\Users\Jürgen\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.mklittlefs.path=C:\Users\Jürgen\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-69bd9e6 -prefs=runtime.tools.mklittlefs-2.5.0-4-69bd9e6.path=C:\Users\Jürgen\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-69bd9e6 -prefs=runtime.tools.mkspiffs.path=C:\Users\Jürgen\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=C:\Users\Jürgen\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.python3.path=C:\Users\Jürgen\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Jürgen\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -verbose E:\MyClock\Sketch\myClock\myClock.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Jürgen\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Jürgen\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Jürgen\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=legacy,ssl=all,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=1M64,led=2,sdk=nonosdk_190703,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -ide-version=10809 -build-path C:\Users\Jürgen\AppData\Local\Temp\arduino_build_350357 -warnings=none -build-cache C:\Users\Jürgen\AppData\Local\Temp\arduino_cache_675892 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Jürgen\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=C:\Users\Jürgen\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.mklittlefs.path=C:\Users\Jürgen\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-69bd9e6 -prefs=runtime.tools.mklittlefs-2.5.0-4-69bd9e6.path=C:\Users\Jürgen\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-69bd9e6 -prefs=runtime.tools.mkspiffs.path=C:\Users\Jürgen\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=C:\Users\Jürgen\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.python3.path=C:\Users\Jürgen\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Jürgen\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -verbose E:\MyClock\Sketch\myClock\myClock.ino
Using board 'generic' from platform in folder: C:\Users\Jürgen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3
Using core 'esp8266' from platform in folder: C:\Users\Jürgen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3
Detecting libraries used...
"C:\\Users\\Jürgen\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\Jürgen\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3/tools/sdk/include" "-IC:\\Users\\Jürgen\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3/tools/sdk/lwip2/include" "-IC:\\Users\\Jürgen\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\Jürgen\\AppData\\Local\\Temp\\arduino_build_350357/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10809 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DLED_BUILTIN=2 -DFLASHMODE_DOUT -DESP8266 "-IC:\\Users\\Jürgen\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3\\cores\\esp8266" "-IC:\\Users\\Jürgen\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3\\variants\\generic" "C:\\Users\\Jürgen\\AppData\\Local\\Temp\\arduino_build_350357\\sketch\\myClock.ino.cpp" -o nul
"C:\\Users\\Jürgen\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\Jürgen\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3/tools/sdk/include" "-IC:\\Users\\Jürgen\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3/tools/sdk/lwip2/include" "-IC:\\Users\\Jürgen\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\Jürgen\\AppData\\Local\\Temp\\arduino_build_350357/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10809 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DLED_BUILTIN=2 -DFLASHMODE_DOUT -DESP8266 "-IC:\\Users\\Jürgen\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3\\cores\\esp8266" "-IC:\\Users\\Jürgen\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3\\variants\\generic" "-IC:\\Users\\Jürgen\\Documents\\Arduino\\libraries\\ArduinoJson-5.x\\src" "C:\\Users\\Jürgen\\AppData\\Local\\Temp\\arduino_build_350357\\sketch\\myClock.ino.cpp" -o nul
"C:\\Users\\Jürgen\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\Jürgen\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3/tools/sdk/include" "-IC:\\Users\\Jürgen\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3/tools/sdk/lwip2/include" "-IC:\\Users\\Jürgen\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\Jürgen\\AppData\\Local\\Temp\\arduino_build_350357/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -f [... cut ...]
Bibliothek Ticker in Version 1.0 im Ordner: C:\Users\Jürgen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\Ticker  wird verwendet
Bibliothek ESP8266mDNS in Version 1.2 im Ordner: C:\Users\Jürgen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266mDNS  wird verwendet
Bibliothek ESP8266HTTPClient in Version 1.2 im Ordner: C:\Users\Jürgen\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266HTTPClient  wird verwendet
exit status 1
no matching function for call to 'WiFiManager::WiFiManager(Stream&)'

can you help me?

kindest regards

tablatronix, 2020-03-13:
I am not reading that

tablatronix, 2020-03-13:
Maybe you have wrong version, this seems to be the development branch

frederic-kleinert, 2020-03-26:
I have tested the WifiManager libraries from version 0.5.0 to 0.15.0, all bring errors:
WiFi: 45: 15: error: 'class WiFiManager' has no member named 'setClass'
'class WiFiManager' has no member named 'getConfigPortalSSID'
and
no matching function for call to 'WiFiManager :: WiFiManager (Stream &)'

What's wrong?

tablatronix, 2020-03-26:
That is the development branch

luto65, 2020-03-29:
Hi, I'm using version 0.15.0 as provided thorugh the Arduino Library manager.
I get indeed the error message:

WiFiManager.cpp:200:12: error: 'class ESP8266WiFiClass' has no member named 'isConnected'

lucatoldo, 2020-03-29:
By using an older version then the problem goes away.

tablatronix, 2020-03-29:
Well what in your sketch is calling wm?