tzapu/WiFiManager #1023: AP Webserver not responding after WiFiManager connected to WiFi

TMaxedCode, 2020-03-15:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [x] Master
- [ ] Development

**Esp8266/Esp32:**
- [x] ESP8266
- [ ] ESP32

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [x] 2.4.0
- [ ] staging (master/dev)

### Description
I changed  your code wherein it makes the esp8266 go STA mode after successfully connecting to a WiFi. I changed it to AP_STA mode instead of just sta mode so I can update a text in the webserver. However, this didn't go as planned because the webserver stopped responding once the esp8266 connected to the wifi. How would this be fixed?

tablatronix, 2020-03-15:
The webserver shutsdown when cp exits use non blocking mode in development branch

TMaxedCode, 2020-03-16:
@tablatronix Thanks, will check it out.

closed 2020-03-16 by TMaxedCode