tzapu/WiFiManager #1028: Library Crashing on ESP-07

geek1111, 2020-03-30:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [x] Development 0.15-beta

**Esp8266/Esp32:**
- [x] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [x] Other ( ESP-07 )

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [x] staging (master/dev)

### Description

Hello I'm facing issue OnDemandConfigPortal sketch without any changes on ESP - 07
After a successful upload, I triggered the OnDemand event by pulling pin 0 low.
when I try to connect the ESP - 07 it crashes and my mobile/laptop wifi disconnects with OnDemanAP and tries to connect again and it crashes in a loop.

- [x] Tripple checked there are no power issues.
- [x] Re-Flashed AT Firmware and uploaded the same sketch.
- [x] Tried other examples available in the library. ( Similar Crash )

### Settings in IDE

- Board: Generic ESp8266 Module.
- Upload Speed: 115200.
- CPU Frequency: 80MHz.
- Crystal Frequency: 26MHz.
- Flash Size: 1MB (FS:64KB OTA:~470KB).
- Flash Mode: Dout (compatible).
- Flash Frequency: 40MHz.
- Rest Method: dtr(aka nodemcu).
- VTables: Flash.
- Exceptions: Enabled.
- Erase Flash: All Flash Contents.
- Espressif FW: nonos-sdk 2.2.1 + 100 (190703)

### Sketch

```cpp

#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino

//needed for library
#include <ESP8266WebServer.h>
#include <DNSServer.h>
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager

// select which pin will trigger the configuration portal when set to LOW
// ESP-01 users please note: the only pins available (0 and 2), are shared 
// with the bootloader, so always set them HIGH at power-up
#define TRIGGER_PIN 0

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("\n Starting");

  pinMode(TRIGGER_PIN, INPUT);
}

void loop() {
  // is configuration portal requested?
  if ( digitalRead(TRIGGER_PIN) == LOW ) {
    //WiFiManager
    //Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wifiManager;

    //reset settings - for testing
    //wifiManager.resetSettings();

    //sets timeout until configuration portal gets turned off
    //useful to make it all retry or go to sleep
    //in seconds
    //wifiManager.setTimeout(120);

    //it starts an access point with the specified name
    //here  "AutoConnectAP"
    //and goes into a blocking loop awaiting configuration

    //WITHOUT THIS THE AP DOES NOT SEEM TO WORK PROPERLY WITH SDK 1.5 , update to at least 1.5.1
    //WiFi.mode(WIFI_STA);
    
    if (!wifiManager.startConfigPortal("OnDemandAP")) {
      Serial.println("failed to connect and hit timeout");
      delay(3000);
      //reset and try again, or maybe put it to deep sleep
      ESP.reset();
      delay(5000);
    }

    //if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");
  }

  // put your main code here, to run repeatedly:

}
```

### Debug Messages

```
Starting
*WM: 
*WM: Configuring access point... 
*WM: OnDemandAP
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started

Exception (9):
epc1=0x40000e62 epc2=0x00000000 epc3=0x00000000 excvaddr=0x01000001 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffefb10 end: 3ffefdf0 offset: 01a0
3ffefcb0:  00000000 00000000 00000001 40100190  
3ffefcc0:  4020d5e0 00000000 3ffefd40 401001b1  
3ffefcd0:  00000000 00000000 3ffefd40 40208966  
3ffefce0:  00000000 3ffeed88 3ffefd40 4020530e  
3ffefcf0:  3ffefd50 3ffe87b9 00000020 401009cc  
3ffefd00:  3ffefd50 3ffefd5c 3ffe87b9 402078fa  
3ffefd10:  00000000 3ffe87b9 3ffefd40 40204de0  
3ffefd20:  00000000 3fffdad0 3ffee8ec 3ffee8ec  
3ffefd30:  3fffdad0 00000000 00000000 402010a0  
3ffefd40:  3fff1254 3fff128c 3ffe8673 00000000  
3ffefd50:  3ffe8600 00000000 80000a0d 40209800  
3ffefd60:  3ffe86e6 80000000 00000000 00000000  
3ffefd70:  00002f0e 4020d3a0 00000000 4020d3a0  
3ffefd80:  00000000 4020d3a0 00000000 4020d3a0  
3ffefd90:  00000000 4020d3a0 00000000 4020d3a0  
3ffefda0:  00000000 00000000 ffffffff 40000001  
3ffefdb0:  3ffe87b9 00000000 00010035 00000000  
3ffefdc0:  00000000 0000000a 3fff1224 401001b1  
3ffefdd0:  feefeffe feefeffe 3ffee8ac 40208a58  
3ffefde0:  feefeffe feefeffe 3ffe84e0 40100c59  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbc204a9b
~ld
⸮cl {⸮ndph
 Starting

```
### Decoded Exceptions 
```
Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
PC: 0x40000e62
EXCVADDR: 0x01000001

Decoding stack results
0x40100190: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\SURAJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 160
0x401001b1: esp_schedule() at C:\Users\SURAJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 113
0x40208966: __yield() at C:\Users\SURAJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 99
0x4020530e: WiFiManager::startConfigPortal(char const*, char const*) at C:\Users\SURAJ\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp line 271
0x401009cc: malloc(size_t) at C:\Users\SURAJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 511
0x402078fa: String::String(char const*) at C:\Users\SURAJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\WString.cpp line 36
0x40204de0: WiFiManager::WiFiManager() at C:\Users\SURAJ\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp line 72
0x402010a0: loop() at C:\Users\SURAJ\Documents\Arduino\libraries\WiFiManager\examples\OnDemandConfigPortal/OnDemandConfigPortal.ino line 45
0x40209800: uart_write(uart_t*, char const*, size_t) at C:\Users\SURAJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\uart.cpp line 509
0x401001b1: esp_schedule() at C:\Users\SURAJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 113
0x40208a58: loop_wrapper() at C:\Users\SURAJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 180
```

tablatronix, 2020-04-01:
Interesting, have you erased flash ?

geek1111, 2020-04-01:
Yeah, Flash erased with esptool.py 
after erase OnDemandPortal sketch worked for the first time and connected to AP as expected.
I tried one more time, It started HTTP Server and when I tried to connect with ESP-07 with Mobile/Laptop its started crashing with the same exception ( 9 ).

tablatronix, 2020-04-01:
Might you have the wrong memory size selected, it sounds like running out of flash

geek1111, 2020-04-02:
Can you please suggest to me what Flash Size needs to be selected? Tried 1MB size with different partition schemes available in Arduino IDE but no luck.

tablatronix, 2020-04-02:
hmm, I dont know it looks like an sdk thing

```
Decoding stack results
0x40100190: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\SURAJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 160
0x401001b1: esp_schedule() at C:\Users\SURAJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 113
0x40208966: __yield() at C:\Users\SURAJ\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266\core_esp8266_main.cpp line 99
```