tzapu/WiFiManager #1029: Alternate method for startWebPortal

GiPe66, 2020-04-04:
Hi,

It should be nice to have an alternate method startWebPortal(const char *title) to specify the title line of the menu when portal is in web mode.

In this case the local IP address si displayed and may not be necessary considering it is also displayed in the footer. Replacing it by a text given when starting the webPortal may give a more focussed information for the application.

Also the current startWebPortal() will go on giving the localIp address.

Thanks for your attention.

tablatronix, 2020-04-05:
The ip only shows if you do not use a hostname

GiPe66, 2020-04-05:
@tablatronix, Thanks for the fast answer.

If I well understand, in the code

`  
str.replace(FPSTR(T_v),configPortalActive ? _apName : WiFi.localIP().toString()); // use ip if ap is not active for heading
`
as I am in webPortal mode, configPortalActive is false, so you always get the IP address.

tablatronix, 2020-04-05:
What version is this? Ill check

tablatronix, 2020-04-05:
Looks like I never implemented that, oops

GiPe66, 2020-04-06:
Maybe a method to set a name should be simpler and enough...

tablatronix, 2020-04-06:
I will work on it tomorrow , but it will be development branch only

GiPe66, 2020-04-06:
No problem, I'm only using this branch considering the non-blocking mode.
Thanks for all this nice job.

tablatronix, 2020-04-06:
hmm best option would be to allow replacing the entire

`const char HTTP_ROOT_MAIN[]        PROGMEM = "<h1>{v}</h1><h3>WiFiManager</h3>";`

GiPe66, 2020-04-06:
Yes, your option seems interesting and gives more flexibility to customize the pages, mainly avoiding the <h1> tag. Thanks.

GiPe66, 2020-04-06:
H1 is really BIG... It should be nice to let the user define the header and the size of the font; So your proposal to possibly replace the full header is for me a good option.