tzapu/WiFiManager #1030: Cannot change the html of a custom parameter

GiPe66, 2020-04-06:
Hi all,

I did not find any way to change the html of a custom parameter.

It is possibletyo get the html code of a custom parameter (getCustomHTML)  but not to change it when it has been created.

I added a simple method to set if.
.
void WiFiManagerParameter::setCustomHTML(const char *custom) {
  _customHTML     = custom;
}

.

This allows changing a select /option in my case bat also will help changing radio bttons or any configurable input in a non blocking mode web portal.

Thanks for your attention.

dontsovcmc, 2020-05-18:
@GiPe66 please, see here: https://github.com/tzapu/WiFiManager/blob/master/WiFiManager.cpp#L15 

there are some examples of using: https://github.com/dontsovcmc/waterius/blob/master/ESP8266/src/setup_ap.cpp#L113

tablatronix, 2020-05-18:
Also check out the childparamsclass example, it shows how to create custom params

TheNetStriker, 2020-06-03:
I'm having the same problem. I've added a custom checkbox parameter, but I want to add the "checked" html attribute dynamically after the WiFiManagerParameter is initialized depending on the stored settings. The examples above only show how the custom html can be set when initializing the WiFiManagerParameter. Is there a way to change this after initialization or could this setter be implemented?

tablatronix, 2020-06-03:
you can use param.setValue and change it at anytime via the param reference

tablatronix, 2020-06-03:
see the dev example

```
  // set values later if you want
  custom_html.setValue("test",4);
````

TheNetStriker, 2020-06-03:
@tablatronix I've already tried this, but this seem to only change the value of my custom parameter. The checkbox on the web page is still not checked when it is loaded. As a workarround I'm now initializing the custom WiFiManagerParameter in the setup() method after reading the json settings and set the checked attribute if needed:

custom_setting = new WiFiManagerParameter("id", "description", "value", 6, setting_enabled ? "type=\"checkbox\" checked" : "type=\"checkbox\"");

tablatronix, 2020-06-03:
Ill have to check but you should be able to just override or add your own methods to change the custom html. I thought someone already made a checkbox example somewhere. But its not in the example