tzapu/WiFiManager #1033: Crash in wifi_softap_start?

cmarrin, 2020-04-08:
I'm running the development build of WiFiManager, mostly so I can use function pointers for the setAPCallback() method. I'm using this same version in another project that is build in the Arduino IDE. But I'm building this project in VSCode using PlatformIO, still with the Arduino SDK.

I have a dead simple case of: 

```
		WiFiManager wifiManager;
		wifiManager.setAPCallback([this](WiFiManager* wifiManager) {
			Serial.printf("Entered config mode:ip=%s, ssid='%s'\n", WiFi.softAPIP().toString().c_str(), wifiManager->getConfigPortalSSID().c_str());
			_enteredConfigMode = true;
		});

		if (!wifiManager.autoConnect()) {
			Serial.printf("*** Failed to connect and hit timeout\n");
			ESP.reset();
			delay(1000);
		}

```

This is in setup(). I'm running this on a Wemos D1 Mini.

I decorated the coredump and it showed me:

```
0x40100d26: malloc at /Users/cmarrin/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/umm_malloc/umm_malloc.cpp:511
0x401005ac: pvPortMalloc at /Users/cmarrin/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/heap.cpp:291
0x40253e33: wifi_softap_start at ??:?
0x4026a48a: wifi_set_opmode_current at ??:?
0x40201ba3: ESP8266WiFiGenericClass::mode(WiFiMode, WiFiState*) at /Users/cmarrin/.platformio/packages/toolchain-xtensa@src-71082da57b6781e9b0698b5fba9df356/xtensa-lx106-elf/include/c++/5.2.0/functional:1873
0x4026a1e6: wifi_get_opmode at ??:?
0x402018ec: ESP8266WiFiGenericClass::getMode() at /Users/cmarrin/.platformio/packages/toolchain-xtensa@src-71082da57b6781e9b0698b5fba9df356/xtensa-lx106-elf/include/c++/5.2.0/functional:1873
0x40206cd5: WiFiManager::WiFi_enableSTA(bool, bool) at /Users/cmarrin/Projects/git/m8rscript/lib/WiFiManager/WiFiManager.cpp:2407 (discriminator 4)
0x402024b8: ESP8266WiFiSTAClass::status() at /Users/cmarrin/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp:685
0x4020d9b1: WiFiManager::startConfigPortal(char const*, char const*) at /Users/cmarrin/Projects/git/m8rscript/lib/WiFiManager/WiFiManager.cpp:2407 (discriminator 4)
0x40238100: HardwareSerial::peek() at /Users/cmarrin/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/HardwareSerial.cpp:145
0x4020dc40: WiFiManager::autoConnect(char const*, char const*) at /Users/cmarrin/Projects/git/m8rscript/lib/WiFiManager/WiFiManager.cpp:2407 (discriminator 4)
0x40206adf: void WiFiManager::DEBUG_WM<__FlashStringHelper const*, char const*>(WiFiManager::wm_debuglevel_t, __FlashStringHelper const*, char const*) at /Users/cmarrin/Projects/git/m8rscript/lib/WiFiManager/WiFiManager.cpp:2407 (discriminator 4)
0x40206b08: void WiFiManager::DEBUG_WM<__FlashStringHelper const*>(WiFiManager::wm_debuglevel_t, __FlashStringHelper const*) at /Users/cmarrin/Projects/git/m8rscript/lib/WiFiManager/WiFiManager.cpp:2407 (discriminator 4)

0x402082fc: void std::swap<std::_Any_data>(std::_Any_data&, std::_Any_data&) at /Users/cmarrin/Projects/git/m8rscript/lib/WiFiManager/WiFiManager.cpp:2407 (discriminator 4)
 (inlined by) std::function<void (WiFiManager*)>::swap(std::function<void (WiFiManager*)>&) at /Users/cmarrin/.platformio/packages/toolchain-xtensa@src-71082da57b6781e9b0698b5fba9df356/xtensa-lx106-elf/include/c++/5.2.0/functional:2160 (discriminator 4)
 (inlined by) std::function<void (WiFiManager*)>::operator=(std::function<void (WiFiManager*)> const&) at /Users/cmarrin/.platformio/packages/toolchain-xtensa@src-71082da57b6781e9b0698b5fba9df356/xtensa-lx106-elf/include/c++/5.2.0/functional:2075 (discriminator 4)
 (inlined by) WiFiManager::setAPCallback(std::function<void (WiFiManager*)>) at /Users/cmarrin/Projects/git/m8rscript/lib/WiFiManager/WiFiManager.cpp:2111 (discriminator 4)
```

So it looks like its getting a memory corruption or something. I print out the heap just before calling this code and it's at 38KB, so I don't think I'm running out of memory.

One more detail is that I'm building with C++14, so I fear I may be getting some core libraries from the wrong place or something.

Has anyone seen any issues like this?

tablatronix, 2020-04-08:
Do a full erase and update your esp lib

tablatronix, 2020-04-10:
It looks like it is getting way past your callback

Do you have serial logs?

```
0x40100d26: malloc at /Users/cmarrin/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/umm_malloc/umm_malloc.cpp:511
0x401005ac: pvPortMalloc at /Users/cmarrin/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/heap.cpp:291
0x40253e33: wifi_softap_start at ??:?
0x4026a48a: wifi_set_opmode_current at ??:?
0x40201ba3: ESP8266WiFiGenericClass::mode(WiFiMode, WiFiState*) at /Users/cmarrin/.platformio/packages/toolchain-xtensa@src-71082da57b6781e9b0698b5fba9df356/xtensa-lx106-elf/include/c++/5.2.0/functional:1873
0x4026a1e6: wifi_get_opmode at ??:?
```

tablatronix, 2020-04-10:
Might be an environment thing, I use pio standalone and it works, 

Although it wont compile with that capture for some reason.

```C++
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

bool _enteredConfigMode = false;

void setup(){
	Serial.begin(115200);
WiFiManager wifiManager;

	wifiManager.setAPCallback([](WiFiManager* wifiManager) {
		Serial.printf("Entered config mode:ip=%s, ssid='%s'\n", 
                        WiFi.softAPIP().toString().c_str(), 
                        wifiManager->getConfigPortalSSID().c_str());
		_enteredConfigMode = true;
	});
	wifiManager.resetSettings();
	if (!wifiManager.autoConnect()) {
		Serial.printf("*** Failed to connect and hit timeout\n");
		ESP.reset();
		delay(1000);
	}
}

void loop(){}
```

``PHP
*WM: [1] resetSettings 
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 
*WM: [1] SETTINGS ERASED 
*WM: [1] AutoConnect 
*WM: [1] No Credentials are Saved, skipping connect 
*WM: [2] Starting Config Portal 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  ESP_3C188C
*WM: [2] AP has anonymous access! 
*WM: [1] SoftAP Configuration 
*WM: [1] -------------------- 
*WM: [1] ssid:             ESP_3C188C
*WM: [1] password:         
*WM: [1] ssid_len:         10
*WM: [1] channel:          1
*WM: [1] authmode:        
*WM: [1] ssid_hidden:     
*WM: [1] max_connection:   4
*WM: [1] country:          CN
*WM: [1] beacon_interval:  100(ms)
*WM: [1] -------------------- 
*WM: [1] AP IP address: 192.168.4.1
Entered config mode:ip=192.168.4.1, ssid='ESP_3C188C'
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan completed in 2184 ms
*WM: [2] Config Portal Running, blocking, waiting for clients... 
```

cmarrin, 2020-04-10:
My setup is pretty "custom" right now. I'm using the development version of WiFiManager. I'm also using a 5.2 version of the toolchain so I can compile c++14 and I'm compiling debug with gdb_init in place. With that I've added some extra log messages to WiFiManager and the Arduino WiFi libraries. Here's the log output:

```
00 excvaddr:0x00000000 depc:0x00000000'
*WM: [1] <form action='/wifi'    method='get'><button>Configure WiFi</button></form><br/>
<form action='/info'    method='get'><button>Info</button></form><br/>
<form action='/exit'    method='get'><button>Exit</button></form><br/>
 
Free heap: 37304

*WM: [1] AutoConnect 
*WM: [1] No Credentials are Saved, skipping connect 
*WM: [2] Starting Config Portal 
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
del if0
usl
mode : null
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  ESP_AD5703
****************************** enableAP:enter
****************************** enableAP:1
****************************** enableAP:2
****************************** mode:1
****************************** mode:1=5
****************************** mode:6
****************************** mode:7
****************************** mode:8
****************************** mode:12
****************************** mode:14
modewifi evt: 8
 : softAP(86:0d:8e:ad:57:03)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
$T0b#e6
```

I added the "mode:14" message to ESP8266WiFiGenericClass::mode() just before the call to wifi_set_opmode_current(). I also have "-DDEBUG_ESP_CORE -DDEBUG_ESP_WIFI" turned on which I think is what is showing some of these diagnostics.

The final "$T0b#e6z" is the system trying to connect with GDB after the crash. I've turned that off and that's where I got the crashdump stack trace from the original message.

I've also tried turning everything off, using the stock toolchain and just building the auto connect example in a new project and it still happens. So if definitely seems like some sort of build environment issue.

cmarrin, 2020-04-10:
> Do a full erase and update your esp lib
> 
> Also try staging, I had alot of crashes in 2.6.x

Sorry, but my PlatformIO knowledge is not great. 

Is adding:

```
platform_packages =
    framework-arduinoespressif8266 @ https://github.com/esp8266/Arduino.git
```
to platformio.ini what you mean by trying staging?

When I do that with my simple test app, it first updates a bunch stuff, then builds and uploads and fails in the same way

cmarrin, 2020-04-10:
@tablatronix when you say " I use pio standalone and it works," do you mean you build and upload from the command line? I tried:

`platformio run -t upload
`
and the same thing happens

tablatronix, 2020-04-10:
Well I mean my platformio IDE is not vscode, but its still pio..

staging is github master.
`platform = https://github.com/platformio/platform-espressif8266.git#feature/stage`

tablatronix, 2020-04-10:
Are you using esp IDF arduino toolchain ? I use straight arduino..

cmarrin, 2020-04-10:
@tablatronix When I try the platform URL you sent, I get:

`Error: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', '--branch', 'feature/stage', 'https://github.com/platformio/platform-espressif8266.git', '/Users/cmarrin/.platformio/platforms/_tmp_installing-0iQvME-package']
`

cmarrin, 2020-04-10:
I don't see any feature branch in that repo. Just master, develop and a release branch

cmarrin, 2020-04-10:
Should have given the simpler part of the error message:

```
warning: Could not find remote branch feature/stage to clone.
fatal: Remote branch feature/stage not found in upstream origin

```

tablatronix, 2020-04-10:
Thats in my platormio.ini env
Not sure how vscode env works, or how you pull in github repos.

tablatronix, 2020-04-10:
What happens if you use a callback function and not a lambda ?

cmarrin, 2020-04-10:
My example right now doesn't set the callback at all. The AutoConnect sample doesn't do it so I assumed it was an ok thing to do.

I was think it could be crashing in malloc because the ssid length was not set so it way trying to allocate an enormous buffer. But I've tried not giving it an SSID and just letting it pick one on its own and the same thing happens.

One thing for sure. I was thinking the code I had around all this was somehow corrupting memory. But since it happens in the most basic example, that's not the case.

One other tidbit. I have another project I'm building in the Arduino IDE. It is using the same development version of WiFiManager and it's working fine. This for sure has to do specifically with starting the AP mode. When I was first testing this project after adding WiFiManager support everything was working fine. That was because I replaced some older code that hardcoded the SSID, so WiFiManager was just picking up those setting and turning on STA mode. When I did resetSettings to test the config portal I started having this problem

tablatronix, 2020-04-10:
Always erase flash for problems with crashes, it gets corrupt when changing versions after updates etc.

So the callback above is not related?