tzapu/WiFiManager #1034: Switched to Dev Branch and Wifi disconnects immediately after

jbronikowski, 2020-04-09:
Hello,

Trying to understand the new logic in the Dev branch. I recently switched over from the master to the dev branch and when i run it, the ESP gets an IP right away but then disconnect configportal and disconnects wifi. Any insight would be greatly appreciated. This works fine on Master

```
09:49:52.973 -> *WM: [2] <- HTTP WiFi save  
09:49:52.973 -> *WM: [3] Method: POST
09:49:52.973 -> *WM: [3] Sent wifi save page 
09:49:52.973 -> *WM: [2] processing save 
09:49:54.992 -> *WM: [2] Connecting as wifi client... 
09:49:54.992 -> *WM: [3] STA static IP:
09:49:54.992 -> *WM: [2] setSTAConfig static ip not set, skipping 
09:49:54.992 -> *WM: [1] CONNECTED:
09:49:54.992 -> *WM: [1] Connecting to NEW AP: YoshRen
09:49:54.992 -> *WM: [3] Using Password: aspentrevor
09:49:54.992 -> *WM: [3] WiFi station enable 
09:49:54.992 -> *WM: [3] enableSTA PERSISTENT ON 
09:49:55.181 -> *WM: [1] connectTimeout not set, ESP waitForConnectResult... 
09:50:00.752 -> *WM: [2] Connection result: WL_CONNECTED
09:50:00.752 -> *WM: [3] lastconxresult: WL_CONNECTED
09:50:00.752 -> *WM: [1] Connect to new AP [SUCCESS] 
09:50:00.752 -> *WM: [1] Got IP Address: 
09:50:00.752 -> *WM: [1] 10.10.100.109 
09:50:00.752 -> Should save config
09:50:00.752 -> *WM: [2] disconnect configportal 
09:50:01.750 -> *WM: [2] restoring usermode AP
09:50:01.750 -> *WM: [2] wifi status: WL_DISCONNECTED
09:50:01.750 -> *WM: [2] wifi mode: AP
09:50:01.750 -> *WM: [1] config portal exiting 
09:50:01.750 -> connected...yeey :)
09:50:01.750 -> local ip
09:50:01.750 -> (IP unset)
```

tablatronix, 2020-04-10:
Dev branch now preserves your user code esp modes, so it puts them back when its done.

You do not have wifi sta mode set.
Autoconnect, will force it on.

cp save does not, I am going to change the default and add a getter, cause I get too many issues about it. Consider it "fixed" in 5 minutes.

"
"09:50:01.750 -> *WM: [2] restoring usermode AP
09:50:01.750 -> *WM: [2] wifi status: WL_DISCONNECTED
09:50:01.750 -> *WM: [2] wifi mode: AP
"

tablatronix, 2020-04-10:
It also preseves persistent, so user can go back to exactly the same wifi modes as before running wm, but since this is drastically different than previous versions ( which are bugged! in this )

tablatronix, 2020-04-10:
e9d75a08b8ad59508a70441fad1cbba90ec0dcaf

tablatronix, 2020-04-10:
I changed it to force sta mode after using wm now, it will prevent these complaints or confusion but will also turn off ap mode if you are using it, so yo have to turn it back on in that case. Should help for now and prevent these issues or questions.

closed 2020-04-10 by tablatronix