tzapu/WiFiManager #1036: Can't add mDNS to OnDemandNonBlocking example

TMaxedCode, 2020-04-10:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [X] Development

**Esp8266/Esp32:**
- [X] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [X] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [X] 2.4.0
- [ ] staging (master/dev)

### Description
I am trying out the OnDemandNonBlocking example in the development branch and everything works fine. I don't see any code in the example showing that the esp8266 is connected to the internet but instead it is written in the library. I also browsed through the library and pretty much got lost because there is too much code. How would I be able to add mDNS support on your OnDemandNonBlocking example? Thank you.

tablatronix, 2020-04-10:
Set wm.hostname and use autoconnect, then service mdns in loop as you normally do.

tablatronix, 2020-04-10:
Also check all the public functions in the .h file they all have comments.

TMaxedCode, 2020-04-11:
@tablatronix Is there any docs on how to use wm.hostname? Sorry, I've never used it once.

tablatronix, 2020-04-11:
`
    // set a custom hostname, sets sta and ap dhcp client id for esp32, and sta for esp8266
    bool          setHostname(const char * hostname);
`

TMaxedCode, 2020-04-12:
Thanks but isn't auto connect blocking? If I use setHostname and autoconnect, it won't be OnDemandNonBlocking anymore

tablatronix, 2020-04-12:
Good point, maybe I can add some mdns helpers instead...

you can still use autoconnect.

```C++
    // if this is set, portal will be blocking and wait until save or exit, 
    // is false user must manually `process()` to handle config portal,
    // setConfigPortalTimeout is ignored in this mode, user is responsible for closing configportal
    void          setConfigPortalBlocking(boolean shouldBlock);
```

TMaxedCode, 2020-04-13:
How can I make that autoconnect on demand as well? Sorry, I forgot to specifically state that it needs to be on demand as well.

tablatronix, 2020-04-13:
Same thing you were doing before, it doesnt change anything

TMaxedCode, 2020-04-16:
Sorry for asking again but I'm really stuck. How would I be able to implement OnDemand and NonBlocking at the same time while using mdns? Can you provide a documentation for this or an example code? Thank you.

tablatronix, 2020-04-16:
This might work, wm does not include dns for you, that might have been your problem

```c++

/**
 * OnDemandNonBlocking.ino
 * example of running the webportal or configportal manually and non blocking
 * trigger pin will start a webportal for 120 seconds then turn it off.
 * startCP = true will start both the configportal AP and webportal
 */
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

// include MDNS
#ifdef ESP8266
#include <ESP8266mDNS.h>
#elif defined(ESP32)
#include <ESPmDNS.h>
#endif

// select which pin will trigger the configuration portal when set to LOW
#define TRIGGER_PIN 0

WiFiManager wm;

unsigned int  timeout   = 120; // seconds to run for
unsigned int  startTime = millis();
bool portalRunning      = false;
bool startAP            = false; // start AP and webserver if true, else start only webserver

void setup() {
  WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP  
  // put your setup code here, to run once
  Serial.begin(115200);
  Serial.println("\n Starting");
  pinMode(TRIGGER_PIN, INPUT_PULLUP);

   wm.setHostname("MDNS_EXAMPLE_ESP");
   wm.setEnableConfigPortal(false);
   wm.autoConnect();
}

void loop() {
  MDNS.update();
  doWiFiManager();
  // put your main code here, to run repeatedly:
}

void doWiFiManager(){
  // is auto timeout portal running
  if(portalRunning){
    wm.process();
    if((millis()-startTime) > (timeout*1000)){
      Serial.println("portaltimeout");
      portalRunning = false;
      if(startAP){
        wm.stopConfigPortal();
      }  
      else{
        wm.stopWebPortal();
      } 
   }
  }

  // is configuration portal requested?
  if(digitalRead(TRIGGER_PIN) == LOW && (!portalRunning)) {
    if(startAP){
      Serial.println("Button Pressed, Starting Config Portal");
      wm.setConfigPortalBlocking(false);
      wm.startConfigPortal();
    }  
    else{
      Serial.println("Button Pressed, Starting Web Portal");
      wm.startWebPortal();
    }  
    portalRunning = true;
    startTime = millis();
  }
}

```

tablatronix, 2020-04-17:
were you able to test this ?

TMaxedCode, 2020-04-18:
> 
> 
> were you able to test this ?

Sorry for late reply, not yet. I will notify you as soon as I test it. Thanks for your solution.

tablatronix, 2020-04-18:
No prob

TMaxedCode, 2020-04-19:
It doesn't work for me. It says unable to connect to it.

tablatronix, 2020-04-20:
so ping hostname.local does nothing ?

TMaxedCode, 2020-05-09:
@tablatronix Sorry for late reply but it does nothing.

tablatronix, 2020-05-09:
I see what I did wrong, let me figure something out, this only would work if there were creds saved

tablatronix, 2020-05-09:
hmm this works for me only testing on sta mode, not sure is mdns works in softap

I am having trouble testing because my router does multicast across wired to wireless networks.

```
/**
 * OnDemandNonBlocking.ino
 * example of running the webportal or configportal manually and non blocking
 * trigger pin will start a webportal for 120 seconds then turn it off.
 * startCP = true will start both the configportal AP and webportal
 */
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

// include MDNS
#ifdef ESP8266
#include <ESP8266mDNS.h>
#elif defined(ESP32)
#include <ESPmDNS.h>
#endif

// select which pin will trigger the configuration portal when set to LOW
#define TRIGGER_PIN 0

WiFiManager wm;

unsigned int  timeout   = 120; // seconds to run for
unsigned int  startTime = millis();
bool portalRunning      = false;
bool startAP            = false; // start AP and webserver if true, else start only webserver

void setup() {
  WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP  
  // put your setup code here, to run once
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  delay(1000);
  Serial.println("\n Starting");

  pinMode(TRIGGER_PIN, INPUT_PULLUP);

  // wm.resetSettings();
  wm.setHostname("MDNSEXAMPLE");
  // wm.setEnableConfigPortal(false);
  // wm.setConfigPortalBlocking(false);
  wm.autoConnect();
}

void loop() {
  MDNS.update();
  doWiFiManager();
  // put your main code here, to run repeatedly:
}

void doWiFiManager(){
  // is auto timeout portal running
  if(portalRunning){
    wm.process();
    if((millis()-startTime) > (timeout*1000)){
      Serial.println("portaltimeout");
      portalRunning = false;
      if(startAP){
        wm.stopConfigPortal();
      }  
      else{
        wm.stopWebPortal();
      }
   }
  }

  // is configuration portal requested?
  if(digitalRead(TRIGGER_PIN) == LOW && (!portalRunning)) {
    if(startAP){
      Serial.println("Button Pressed, Starting Config Portal");
      wm.setConfigPortalBlocking(false);
      wm.startConfigPortal();
    }  
    else{
      Serial.println("Button Pressed, Starting Web Portal");
      wm.startWebPortal();
    }  
    portalRunning = true;
    startTime = millis();
  }
}

```

TMaxedCode, 2020-05-11:
it still doesn't work for some reason. I can get in using the local ip while it is in sta mode but then using mdnsexample.local doesn't seem to work

tablatronix, 2020-05-11:
Are you sure your network domain is .local ? or you are on the same wifi ?

If you are on wired, mdns may fail across segments, some broadband routers have alt dns domains, mine has attlocal.net for example

TMaxedCode, 2020-05-11:
Ok I will check it. I am on the same wifi as the esp8266.

TMaxedCode, 2020-05-16:
I keep on trying and it still doesn't work. Entering the sta ip works but not the mdns.

tablatronix, 2020-05-16:
Ok ill upload this example and start from scratch. You can try that

tablatronix, 2020-05-16:
What os are you on?

tablatronix, 2020-05-18:
I added mdns to that example are you using that?

tablatronix, 2020-05-19:
I have no idea how this used to work but it does not now..

    #ifdef ESP8266MDNS_H
Does not work to check if mdns was included..

I have to fix this, precompiler directives are so annoying.

TMaxedCode, 2020-05-19:
Sorry for really late replies. I rarely check my github. I'm using my nodemcu and im using windows 10.

tablatronix, 2020-05-19:
Everytime i think I understand precprocessor macros, ill never get it.

Why can I not do 

```
#include <ESP8266mDNS.h>

#ifdef ESP8266MDNS_H
#endif

```

I dont get it

TMaxedCode, 2020-05-20:
I am not good with writing libraries but then did you include it in your .h file? I usually log every action to the console so its easier to debug whats happening.

closed 2021-06-11 by CoderUni