tzapu/WiFiManager #1037: no connection between serverscript and esp

tolletje223, 2020-04-14:
i have an problem to get my sketch working with my esp. I cannot make an connection with the server apparantly, and it will not return messages thus.
(https://www.youtube.com/watch?v=k4Ux_I4Gl30&t=11s)
I tried almost everything a newby could have done, so now i am reaching out  , cause the wifimanager is not working, i believe?
i tried several hosts in the sketch, used http and https hostst, switched to other esp boards,placed a Serial.setDebugOutput(true)in the sketch to find out what might have been the issue. For now i have a connection with the router i believe cause it gives me no' failure 'in the serial. (First i had a no connection failure all the time..)
when i try to obtain the message from the serverscript online , it failes, the script however reports the EOT , so the script is working. But the connection is not made to retrieve it ?
Could you please look into it and see if there is anything wrong? Any suggestions would be great :)

-router is seeing the esp, gives ip adress 
-http or https domain not working  (dailydeals.one/woodiebox/matrix)(lichtkrant.epizy.com/test)
-Oled states that is has a connection with wifi
-connected to the host outside my network (mobile)

i use a NodeMcu Lua CH340G (nodeMcu 1.0 esp 12e module setting) with two matrix led max7219 
esp3266 addon 2.6.0 and downgraded it also to 2.4.0 (same results)
4-1 module and a 7 pin oled 1.3 inch display.
wifimanager version 0.15 used , but also tried 0.14 version

**CODE:**

void ConnectToServer(){
  // Use WiFiClient class to create TCP connections
  const char* host      = "dailydeals.one";                 // Here coms your Webserver / Host adress
  String url       = "/woodiebox/matrix/myText.txt";         // The Path and Filename
  WiFiClient client;
  const int httpPort = 80;
  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed");
    return;
  }   
  // This will send the request to the server
  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" + 
               "Connection: close\r\n\r\n");
  unsigned long timeout = millis();
  while (client.available() == 0) {
    if (millis() - timeout > 5000) {
      Serial.println(">>> Client Timeout !");
      client.stop();
      return;
    }
  }
  // Read all the lines of the reply from server
  while(client.available()){
    String line = client.readStringUntil('\r');
    //Serial.print(line);                             // Diag, Show's all the lines from HTTP answere
    if(line.endsWith("EOT")){                         // line with EOT at the end? (adding in PHP-Script to the end of text)
      line = line.substring(1,line.length() -3);      // Shortens the string by by 3 characters "EOT"
      if (line == "WiFiReset"){                       // Secret phrase to reset WiFi settings. Next start in config Mode
        wifiManager.resetSettings();
      }
      if(line.startsWith("Lumos=")){                  // Secret phrase to set intensity of dot matrix
        line = line.substring(6,line.length());
        SetIntensity(line);
        return;
      }
      line.toCharArray(curMessage, sizeof(curMessage));  
    }
  }
}

void SetIntensity(String lumos ){           // Set the intensity of dot matrix. Text for Website: "Lumos=0" to "Lumos=15"
  long l = lumos.toInt();
  byte i = (byte) l;
  if ((i>=0) && (i<=15)){  //&& (i!= i2) geht nicht, Schreibt immer weil i2 von Eeprom immer 0 ?? // Values from 0 to 15
    EEPROM.write(0, i);       // Write the new value to EEPROM
    EEPROM.commit();
    Parola.setIntensity(i);   
  } 
}
  
void configModeCallback (WiFiManager *myWiFiManager) {      // Show infos to log in with build in accesspoint. (Config Mode) When not possible to connect to WiFi 
  u8g2.clearBuffer();   u8g2.drawStr(0,11,"Config Mode");   
  u8g2.drawStr(0,27,"AP: "); 
  myWiFiManager->getConfigPortalSSID().toCharArray(oledTxt, sizeof(oledTxt));
  u8g2.drawStr(28,27,oledTxt);  
  u8g2.drawStr(0,43,"IP address: ");    
  sprintf(oledTxt, "%03d.%03d.%03d.%03d", WiFi.softAPIP()[0], WiFi.softAPIP()[1], WiFi.softAPIP()[2], WiFi.softAPIP()[3]);
  u8g2.drawStr(0,59, oledTxt);  u8g2.sendBuffer(); 
}

void OLED_Screen1(){                        // Wellcome screen
  u8g2.setFont(u8g2_font_7x14B_tr);         // choose a small font for status dispay 
  u8g2.clearBuffer(); 
  u8g2.drawStr(0,11,"    Welcome to"); 
  u8g2.setFont(u8g2_font_profont22_tf);  
  u8g2.drawStr(6,32, "ScrollText");  
  u8g2.setFont(u8g2_font_7x14B_tr); 
  u8g2.drawStr(0,48,"    Connecting.."); 
  u8g2.setFont(u8g2_font_6x10_tf); 
  u8g2.drawStr(0,64,"v1.3");  u8g2.sendBuffer(); 
}

void OLED_Screen2(){                      // Connected screen
  u8g2.setFont(u8g2_font_7x14B_tr);       // choose a small font for status dispay 
  u8g2.clearBuffer();
  u8g2.drawStr(0,11,"WiFi connected");   
  u8g2.drawStr(0,27,"SSID: ");  
  WiFi.SSID().toCharArray(oledTxt, sizeof(oledTxt));
  u8g2.drawStr(37,27,oledTxt);  
  u8g2.drawStr(0,43,"IP address: ");    
  sprintf(oledTxt, "%03d.%03d.%03d.%03d", WiFi.localIP()[0], WiFi.localIP()[1], WiFi.localIP()[2], WiFi.localIP()[3]);
  u8g2.drawStr(0,59, oledTxt);  u8g2.sendBuffer();
}

void OLED_Screen3(){                        // Scan now screen
   u8g2.setFont(u8g2_font_profont22_tf);    // choose a big font for "Scan Now" text
   u8g2.clearBuffer();
   u8g2.drawStr(4,15,"scan QR");   
   u8g2.drawStr(5,35,"-> for own");   
   u8g2.drawStr(35,55,"Text!"); u8g2.sendBuffer();
}

void drawPic() {                          // show QR-Code
  // picture loop
  u8g2.clearBuffer();
  u8g2.firstPage();  
  do {
      u8g2.drawXBM( 0, 0, qr_code_width, qr_code_height, qr_code_bits);
  } while( u8g2.nextPage() );
}

**Server script:**
<!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Strict//EN" "http://www.w3.org/TR/xhtml1/DTD/xhtml1-strict.dtd">
<html xmlns="http://www.w3.org/1999/xhtml" xml:lang="de" lang="de">
<meta http-equiv="Content-Type" content="text/html; charset=iso-8859-1">
<style type="text/css">
    body {
     margin: 20px auto;
	 background-color: LightGoldenRodYellow
    }
</style>
<Center>
<font face='Arial,Helvetica'>
<font color="blue">LED Laufschrift</font> <br>
Geben sie hier den Text ein. (1 bis 80 Zeichen).
<br>

<?php
if(isset($_POST['submit'])) {							// Is executed when clicking on Submit
	$data = ($_POST['textfeld']);
	if(strlen($data) > 0 AND strlen($data) < 81 ) {  			// Only if valid number
		$path = 'myText.txt';
		$data2 = $data . "EOT"; 					// Attach EOT for extraction in ESP8266
		$ret = file_put_contents($path, $data2, LOCK_EX);
		if($ret === false) {
			die('Es gab einen Fehler beim schreiben Textzeile');
		}
		else {
		}	
	}
	else {									// Too few or too many characters
		echo '<font color="red">';
		echo "Das waren ". strlen($data). " Zeichen. Bitte minimum 1 Zeichen maximum 80 Zeichen eingeben";
		echo '</font>';
	}
}	
?>

<br>

<form action="<?php htmlentities($_SERVER['PHP_SELF']);?>" method="post" accept-charset="ISO-8859-1">
<input type="text" size="60" name="textfeld" style="font-size:18pt">
<br>
<br>
<input type="submit" name="submit" value="Senden" style="font-size:18pt">
</form>

<?php // d.waldesbuehl@hispeed.ch
if(strlen($data) > 0 AND strlen($data) < 81 ) {					// With valid input
	echo (strlen($data) ." Zeichen wurden geschrieben <br />");
	echo ('Text: "'. utf8_encode($data) . '"');
}
?>

**Serial Output:**

scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 7
cnt 
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Status:
*WM: 6
*WM: Using last saved values, should be faster
state: 5 -> 0 (0)
rm 0
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 7
cnt 

connected with Ziggo67C8334, channel 1
dhcp client start...
ip:192.168.178.76,mask:255.255.255.0,gw:192.168.178.1
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.178.76
pm open,type:2 0

tolletje223, 2020-04-14:
performed a ping too:
C:\Users\daily>ping 192.168.178.76

Pinging 192.168.178.76 with 32 bytes of data:
Reply from 192.168.178.76: bytes=32 time=62ms TTL=255
Reply from 192.168.178.76: bytes=32 time=74ms TTL=255
Reply from 192.168.178.76: bytes=32 time=89ms TTL=255
Reply from 192.168.178.76: bytes=32 time=17ms TTL=255

Ping statistics for 192.168.178.76:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 17ms, Maximum = 89ms, Average = 60ms

**Serial output:** -after upgrading board to esp  2.6.3

connected with Ziggo67C8334, channel 1
dhcp client start...
ip:192.168.178.76,mask:255.255.255.0,gw:192.168.178.1
*WM: Connection result: 
*WM: 3
station: 00:1c:bf:0e:36:63 leave, AID = 1
rm 1
bcn 0
del if1
pm open,type:2 0
**mode : sta(ec:fa:bc:bc:53:fb)**  // was put in place after upgrading board to 2.3.6 version

tablatronix, 2020-04-14:
Sounds like a problem with your client code not wifi

tolletje223, 2020-04-14:
@tablatronix : Do you have any suggestions how to fix this?

tablatronix, 2020-04-14:
I do not see anything from ConnectToServer in your serial log

    Serial.println("connection failed");
      Serial.println(">>> Client Timeout !");

add some debugging your your client code.. shrug try something easier like get ntp time or something

tolletje223, 2020-04-14:
@tablatronix 
No , well thats the point, i can not connect to the server! I had a serial print line connection failed, that vanished as soon as i upgraded my board to 2.6.3 version. The designer of the sketch had a serverscript that was placed on a NAS , does that change anything within my code do you think?
Still, i can not figure out how i can  get my connection to the server . 
since it does not say connection failed or time out for client , i pressume it has a connection , but it can not retrieve the info i want...
Debugging client code you say, how is that done? 
What does this : **pm open,type:2 0** mean btw ? Do you know? Cant find it anywere .

thanks for your suggestions and help, i can't find the troublemaker that is causing this and any help i am gratefull for

tablatronix, 2020-04-14:
Thats just low level wifi debugging.

Does the serial print(line)output anything?

You need to add more prints to see where its getting to

tolletje223, 2020-04-14:
@tablatronix 
Still bugging around with the problem, i was trying to connect to the server URL from the designer. Guess what, it received the message i typed there. So when i placed the URL from the server from the designer into my sketch , it worked. I had my textmessage printed on my LED matrix.
Thuss, the code is OK. It still has something to do with connecting to **my** html page . maybe serverscript ?

Serial gives  me no more info btw, it still keeps ending at **pm open,type:2 0**

For i know the only difference between me and the designers connection is that he posted his script on his NAS , and i mine on my external website server. But that could not  make any difference does it?

tolletje223, 2020-04-14:
By getting the sourceview of the pages i get these differences 

**serverscriptcode My website:**

<!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Strict//EN" "http://www.w3.org/TR/xhtml1/DTD/xhtml1-strict.dtd">
--
  | <html xmlns="http://www.w3.org/1999/xhtml" xml:lang="de" lang="de">
  | <meta http-equiv="Content-Type" **content="text/php**; charset=iso-8859-1">
  | <style type="text/css">
  | body {
  | margin: 20px auto;
  | background-color: LightGoldenRodYellow
  | }
  | </style>
  | <Center>
  | <font face='Arial,Helvetica'>
  | <font color="blue">LED Laufschrift</font> <br>
  | Geben sie hier den Text ein. (1 bis 80 Zeichen).
  | <br>
  |  
  |  
  | <br>
  | 
  | <form action="" method="post" accept-charset="ISO-8859-1">
  | <input type="text" size="60" name="textfeld" style="font-size:18pt">
  | <br>
  | <br>
  | <input type="submit" name="submit" value="Senden" style="font-size:18pt">
  | </form>

**serverscriptcode designer site**:

  | <!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Strict//EN" "http://www.w3.org/TR/xhtml1/DTD/xhtml1-strict.dtd">
-- | --
  | <html xmlns="http://www.w3.org/1999/xhtml" xml:lang="de" lang="de">
  | <meta http-equiv="Content-Type" **content="text/html**; charset=iso-8859-1">
  | <style type="text/css">
  | body {
  | margin: 20px auto;
  | background-color: LightGoldenRodYellow
  | }
  | </style>
  | <Center>
  | <font face='Arial,Helvetica'>
  | <font color="blue">LED Laufschrift</font> <br>
  | Geben sie hier den Text ein. (1 bis 120 Zeichen).
  | <br>
  |  
  |  
  | <br>
  | 
  | <form action="" method="post" accept-charset="ISO-8859-1">
  | <input type="text" size="60" name="textfeld" style="font-size:18pt">
  | <br>
  | <br>
  | <input type="submit" name="submit" value="Senden" style="font-size:18pt">
  | </form>

tablatronix, 2020-04-14:
Are you getting stuck in while(client.available()){ ?
does it ever exit?

tolletje223, 2020-04-14:
@tablatronix 

I dont know, i use the same serverscript , so far i know, and with connection made to developers nas , no problems!! 
But when i use the script on my own hosted sites (several) it does not return the messages , so logical , for me , it must mean that the server script was altered . In the code the while(client.available()){ must be right. **I can connect with developers nas(site)..**
My settings for the modem/router and multiple hosts i tried , were also correct.  

I posted a request to the developers adress for serverscript comparances.  I am breaking my head on this issue for a month or so now. 
I dont want to piggyback on his server for my project to work , i need my own host to operate properly offcourse . i think i better wait for a response to see if the script was altered or not ..

tablatronix, 2020-04-14:
Could be the way the server returns whitespace readline until \r

Like I said you need to dump debug logs from your micro to see what data is coming, and use a static file to fetch data from for testing

tolletje223, 2020-04-15:
@tablatronix 

Solved!

I used another free webhosting , made a account to upload my script and it worked. Probarly on my own sites the software that runs the site blocked the script. Anyway, thanks for your input , you've been helpfull 👍

closed 2020-04-15 by tolletje223