tzapu/WiFiManager #1038: Need help vith DEV configportal.

usane1, 2020-04-17:
Hi!
I need a little help with nonblocking configportal.
I've managed to make my project from the nonblocking example, but there is one more thing i want to achieve. I don't know if its possible. So far if i set up the wifi then it works fine. If the router failing and recovering the ESP reconnect fine and it doesnt bother the main loop.
However, can i achive somehow that if the program in the ESP  start without wifi (router shutdown) then if the wifi is up after the program started, the ESP connect to it if it was configured earlier?

closed 2020-04-20 by usane1

tablatronix, 2020-04-20:
ESP already does that automatically,  you can just check for connected now and then. But esp autoreconnect in the background

tablatronix, 2020-04-20:
` if(WiFi.status() == WL_CONNECTED)`