tzapu/WiFiManager #1044: https portal

Nimo11, 2020-05-02:
Hello,

I will find it very useful to be able to use the portal in https. I use the non-blocking portal and it would be more secure. Do you have it in your road map.

Thank you for this very useful work

tablatronix, 2020-05-02:
Yes ideally you would provide your own websever and pas it into wm

Nimo11, 2020-05-05:
You could maybe change handleRequest to this :

```
void WiFiManager::handleRequest() {
  if (WiFiManager::AuthenticationRequired&&!server->authenticate(WiFiManager::AuthUser,WiFiManager::AuthPassword))
  {
    server->requestAuthentication(HTTPAuthMethod::DIGEST_AUTH);
  }
  _webPortalAccessed = millis();
}
```

tablatronix, 2020-05-05:
What’s that do?

Nimo11, 2020-05-08:
It add authentication ability. 

Changing handleRequest with this code and adding in header file this property :

  bool AuthenticationRequired;
  const char* AuthUser;
  const char* AuthPassword;

So developers can decide to ask authentication simply.

Sorry for this too fast reply.

tablatronix, 2020-05-08:
Does this work with captive portal ?

Nimo11, 2020-05-10:
Yes, all requests are processed. But the behavior can be modified by changing the value of AuthenticationRequired after the autoconnect

tablatronix, 2020-11-23:
Ok this definitely does not work for captive portals. I am gonna try to add a landing page and see if that works.

Nimo11, 2021-01-10:
I try it this week and it work me with dev branch. I restart with acces point and i have login needed, After configuration i have a login page on my portal. What's wrong with your test?

tablatronix, 2021-01-10:
Https authenticaiton does not work on my devices. It never pops up