tzapu/WiFiManager #1045: Can´t connect Nodemcu with WiFiManager

wicherc, 2020-05-02:
#### Hardware: Nodemcu ESP8266 Lolin
**WiFimanager Branch/Release: 0.15.0
- [ ] Master
- [x ] Development

**Esp8266/Esp32:**
- [x ] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ x] Other: Nodemcu ESP8266 V3 Lolin
**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0 
- [ ] staging (master/dev)

When I try to connect to the network, don´t recognize the previous ssid and pwd stored. Then the AP is created and when I select my ssid don´t connect. It´s the same at a cold start or under reset condition.
If I change the router channel, from the options 1,6,11 or automatic, I get connected after some intents. The modem is SAGENCOM fast3890 and the connection can be for 1 or 20hrs and then fall down.
Can help me?
Regards

### Settings in IDE

Module:  NodeMcu
Additional libraries: 
ESP8266WiFi.h
DNSServer.h
ESP8266WebServer.h
WiFiManager.h
BlynkSimpleEsp8266.h
### Sketch

`void setup() {
  Serial.begin(19200);
  Serial.println("Comienza");
  WiFiManager wifiManager;
  //  wifiManager.resetSettings();
  wifiManager.autoConnect("cultivAr", "metrolAdmin");
  if (!wifiManager.autoConnect()) {
    Serial.println("failed to connect and hit timeout");
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(1000);
  }
  Serial.print("Conectado, ip: ");
  Serial.println(WiFi.localIP());

  Serial.println("Conectado");
  Blynk.config(auth);
  Blynk.connect();
  //  Blynk.begin(auth, ssid, pass);
}

void loop() {
  Blynk.run();
}`

### Debug Messages

Comienza
16:42:48.181 -> *WM: 
16:42:48.181 -> *WM: AutoConnect
16:42:48.274 -> *WM: Connecting as wifi client...
16:42:48.274 -> *WM: Status:
16:42:48.274 -> *WM: 6
16:42:48.274 -> *WM: Using last saved values, should be faster
16:43:48.288 -> *WM: Connection result: 
16:43:48.288 -> *WM: 255
16:43:48.288 -> *WM: 
16:43:48.288 -> *WM: Configuring access point... 
16:43:48.288 -> *WM: cultivAr
16:43:48.288 -> *WM: metrolAdmin
16:43:48.791 -> *WM: AP IP address: 
16:43:48.791 -> *WM: 192.168.4.1
16:43:48.791 -> *WM: HTTP server started
16:47:15.391 -> *WM: Request redirected to captive portal
16:47:15.672 -> *WM: Request redirected to captive portal
16:47:22.819 -> *WM: Handle root
16:47:26.815 -> *WM: Scan done
16:47:26.815 -> *WM: DUP AP: TeleCentro Wifi
..........................................
16:47:26.915 -> *WM: Claudio
16:47:26.915 -> *WM: -16
16:47:26.915 -> *WM: TeleCentro Wifi
16:47:26.949 -> *WM: -75
..........................................
16:47:27.086 -> *WM: Sent config page
16:47:32.118 -> *WM: WiFi save
16:47:32.118 -> *WM: Sent wifi save page
16:47:33.107 -> *WM: Connecting to new AP
16:47:33.107 -> *WM: Connecting as wifi client...
16:47:33.107 -> *WM: Status:
16:47:33.140 -> *WM: 0
16:47:33.140 -> *WM: [ERROR] WiFi.begin res:
16:47:33.140 -> *WM: 6
16:48:33.137 -> *WM: Connection result: 
16:48:33.137 -> *WM: 255
16:48:35.106 -> *WM: Failed to connect.
16:48:35.106 -> *WM: Request redirected to captive portal
16:48:35.387 -> *WM: Request redirected to captive portal
16:48:36.930 -> *WM: Request redirected to captive portal
16:48:37.232 -> *WM: Request redirected to captive portal
------------ and each 1 sec ---------------------
16:54:06.950 -> *WM: Request redirected to captive portal
16:54:07.222 -> *WM: Request redirected to captive portal
16:54:37.009 -> *WM: Request redirected to captive portal
16:54:37.272 -> *WM: Request redirected to captive portal
16:55:06.926 -> *WM: Request redirected to captive portal
16:55:07.207 -> *WM: Request redirected to captive portal

and so on

tablatronix, 2020-05-03:
Do full erase

wicherc, 2020-05-03:
Sorry, but how I can do this? I supose you don´t refer to the sketch.
Can you help me, please?
Thank you!

tablatronix, 2020-05-03:
Arduino boards menu

tablatronix, 2020-05-03:
Usually if you have connection failures when you have saved credentials it is corrupt memory a full flash erase will be needed to fix it, if it doesn’t repost

wicherc, 2020-05-04:
Ok. Do you refer at this option? The next flash operation will erase all the memory?
[image]

Thanks a lot for your help

tablatronix, 2020-05-04:
yup

wicherc, 2020-05-04:
I just do it, but it´s the same. If I don't make a change with the channel of my router I can't get connected.
Thanks

tablatronix, 2020-05-04:
interesting

you say you are using development? those serial logs do not look like dev

wicherc, 2020-05-04:
I'm sorry, but can't understand about this logs. What is the diference btwn Master and Dev?

tablatronix, 2020-05-05:
Alot

wicherc, 2020-05-05:
Ok. Now I understood. I'm workink with the Code from Master branch

tablatronix, 2020-05-06:
It might be a bug or corrupt flash , try develop branch let me know how it goes.

Like i said this is a known issue when flash is corrupt so try that first. Also many bugs are fixed version to version in esp so try rolling back or using staging lib

wicherc, 2020-05-06:
Thanks. I'll try using the develop branch and post here the results.

wicherc, 2020-05-07:
Here the results:
The dev version has a lot diferences with the log. 
Here the info from the portal:
esp8266
Uptime: 1 Mins 29 Secs
Chip ID: aa22b2
Flash Chip ID: 1458415
Flash Size: 4194304 bytes
Real Flash Size: 4194304 bytes
SDK Version: 2.2.2-dev(38a443e)
Core Version: 2_7_0
Boot Version: 31
CPU Frequency: 80MHz
Memory - Free Heap: 36392 bytes available
Memory - Sketch Size:
Used / Total bytes
354624 / 960832

Last reset reason:External System

WiFi
Access Point IP: 192.168.4.1
Access Point MAC: 86:0D:8E:AA:22:B2
SSID: Claudio
BSSID: EC:BE:DD:A7:59:48
Station IP: (IP unset)
Station Gateway: (IP unset)
Station Subnet: (IP unset)
DNS Server:(IP unset)
Hostname: ESP-AA22B2
Station MAC: 84:0D:8E:AA:22:B2
Connected:No
Autoconnect:Enabled

At the first intend the result was OK:
Conectando
*WM: [1] AutoConnect 
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping 
*WM: [1] Connecting to SAVED AP: Claudio
*WM: [3] Using Password: xxxxxxxxxxxx
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 192.168.0.143
Conectado, ip: 192.168.0.143
Conectando a Blynk
Conectado a Blynk
*WM: [3] unloading 

After 4 HW resets, the reconnection was OK, but at the 5th the device couldn't reconnect and the log was:
Conectando
*WM: [1] AutoConnect 
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping 
*WM: [1] Connecting to SAVED AP: Claudio
*WM: [3] Using Password: xxxxxxxxxxx
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_NO_SSID_AVAIL
*WM: [3] lastconxresult: WL_NO_SSID_AVAIL
*WM: [1] AutoConnect: FAILED 
*WM: [2] Starting Config Portal 
*WM: [2] AccessPoint set password is VALID 
*WM: [1] xxxxxxxxx 
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  cultivAr
*WM: [1] SoftAP Configuration 
*WM: [1] -------------------- 
*WM: [1] ssid:             cultivAr
*WM: [1] password:         xxxxxxxx
*WM: [1] ssid_len:         8
*WM: [1] channel:          1
*WM: [1] authmode:         3
*WM: [1] ssid_hidden:     
*WM: [1] max_connection:   4
*WM: [1] country:          CN
*WM: [1] beacon_interval:  100(ms)
*WM: [1] -------------------- 
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan completed in 2185 ms
*WM: [2] Config Portal Running, blocking, waiting for clients... 
*WM: [2] Portal Timeout In 175 seconds

And then the portal was out after 175secs and the device goes on autorestart
Thanks

tablatronix, 2020-05-07:
Yeah alot of changes have been made including extended logging which can be changed of course

wicherc, 2020-05-07:
With this I can discard the flash corrupt?

tablatronix, 2020-05-07:
The full erase is like a format

wicherc, 2020-05-07:
If the full erase is made compiling the sketch and transferring it to the device with the option I signaled before in other post, I do it. Most deep reading docs and testing options I get more confused

tablatronix, 2020-05-07:
Not sure how the arduino ide works, I guess it just does it all at once. I use platformio which uses esptool to erase flash

tablatronix, 2020-05-07:
Yes it looks like it does whatever is selected, you can change it back to sketch, you only have to full erase when changing versions (as the partition may change )or if you have issues from a bug that corrupts the credential memory

wicherc, 2020-05-07:
I compiled the program with the option of full erase and I made flash with this version. The result is the same.
After several tests, I found a clue that I had previously followed and I have commented here.
If at the time of connecting or resetting the device, I connect to the administration page of my router and modify the WiFi channel, then the device connects immediately. No matter which channel is selected, the connection is established just by changing this parameter. I cannot understand this as I am ignorant on this topic.

tablatronix, 2020-05-07:
The development branch lets you set the country see the dev example this fixes issues with the esp using chinese channels. Might be related, else something is up with your router maybe it kust takes a while to allow you reconnect

wicherc, 2020-05-07:
I was looking the country code for Argentina and are the same parameters than CN.
I will try to connect in another place to use another router.
Thanks for your support!

tablatronix, 2020-05-07:
I think something might be flaky with your router, I remember someone was having issues on the esp library, they had to turn off some kind of multimedia traffic setting I forget what it was called but it caused connection issues

wicherc, 2020-05-09:
All right.
I dusted off an old 300Mbps router that I had saved. I have connected it as a repeater of the main router and to this I connected the ESP device. It has been working for two days. It has service outages from what I could detect, but here, the automatic reconnect code works perfectly. Obviously there is something suspicious in the first router and it is not the distance that it must cover to the device, because they are only 60cm.
I greatly appreciate your patience and support of my doubts. At the moment I think we could close the case.
Thank you.

tablatronix, 2020-05-09:
Yeah try to find that issue, it was a setting on routers, some multimedia routing protocol that was causing issues

budi99ok, 2020-06-23:
I had smooth connecting after changing WMM configuration. Prev WMM setting, optimize to 10 user.. realy hard to connect... (Tenda router)
[image]