tzapu/WiFiManager #1046: Event handler not de-registered when WiFiManager destroyed

liebman, 2020-05-04:
This is on the development branch. The WiFiManager object was destroyed much earlier.  The event callback gets called and eventually crash.

https://github.com/tzapu/WiFiManager/blob/293f705b0ff29f5c78443a1f181a9872875aeac6/WiFiManager.cpp#L208-L211

```
Exception Cause: Not found

0x400ecb8f: Print::printf(char const*, ...) at /Users/chris.l/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.cpp:234
0x400014fd: ?? ??:0
0x4000150d: ?? ??:0
0x400ecb8f: Print::printf(char const*, ...) at /Users/chris.l/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.cpp:234
0x400e5ee9: void WiFiManager::DEBUG_WM<char const*, unsigned char>(WiFiManager::wm_debuglevel_t, char const*, unsigned char) at /Users/chris.l/.pio-workspaces/MatrixClock-f51995dc94/libdeps/esp32_liebman/WifiManager/WiFiManager.cpp:2448
0x400e5fd2: WiFiManager::WiFiEvent(system_event_id_t, system_event_info_t) at /Users/chris.l/.pio-workspaces/MatrixClock-f51995dc94/libdeps/esp32_liebman/WifiManager/WiFiManager.cpp:2448
0x400e1f46: void std::_Mem_fn_base<void (WiFiManager::*)(system_event_id_t, system_event_info_t), true>::operator()<system_event_id_t, system_event_info_t, void>(WiFiManager*, system_event_id_t&&, system_event_info_t&&) const at /Users/chris.l/.pio-workspaces/MatrixClock-f51995dc94/libdeps/esp32_liebman/WifiManager/WiFiManager.cpp:2448
 (inlined by) void std::_Bind<std::_Mem_fn<void (WiFiManager::*)(system_event_id_t, system_event_info_t)> (WiFiManager*, std::_Placeholder<1>, std::_Placeholder<2>)>::__call<void, system_event_id_t&&, system_event_info_t&&, 0u, 1u, 2u>(std::tuple<system_event_id_t&&, system_event_info_t&&>&&, std::_Index_tuple<0u, 1u, 2u>) at /Users/chris.l/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1074
 (inlined by) void std::_Bind<std::_Mem_fn<void (WiFiManager::*)(system_event_id_t, system_event_info_t)> (WiFiManager*, std::_Placeholder<1>, std::_Placeholder<2>)>::operator()<system_event_id_t, system_event_info_t, void>(system_event_id_t&&, system_event_info_t&&) at /Users/chris.l/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1133
 (inlined by) std::_Function_handler<void (system_event_id_t, system_event_info_t), std::_Bind<std::_Mem_fn<void (WiFiManager::*)(system_event_id_t, system_event_info_t)> (WiFiManager*, std::_Placeholder<1>, std::_Placeholder<2>)> >::_M_invoke(std::_Any_data const&, system_event_id_t&&, system_event_info_t&&) at /Users/chris.l/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1871
0x400da8c5: std::function<void (system_event_id_t, system_event_info_t)>::operator()(system_event_id_t, system_event_info_t) const at /Users/chris.l/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:250
 (inlined by) WiFiGenericClass::_eventCallback(void*, system_event_t*) at /Users/chris.l/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:438
0x400da965: _network_event_task(void*) at /Users/chris.l/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:250
0x40088bfd: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

tablatronix, 2020-05-04:
haha

tablatronix, 2020-05-11:
Do you know which event, I am having a hard time reproducing to make sure a fix is working.

What does your code look like ?

liebman, 2020-05-11:
Hmm - don't know, but my guess would be something that happens at DHCP renew maybe?  It was hours after the WiFiManager was destroyed.  My code is changed to work around it by not creating a WiFiManager unless I need to config.  Than reset after the config.  If I get some time next weekend I'll try to reproduce it again and find the event.

tablatronix, 2020-05-11:
were you up to date because there is a flag in the event handler to avoid this..
odd

tablatronix, 2020-05-11:
Well I added an event id, nor sure if I need to protect its memory or not.. let me know if it fixes your problem.

closed 2020-05-11 by tablatronix

liebman, 2020-05-11:
If your talking about `_hasBegun` that won't help once the WiFiManger object has been destroyed (and memory returned to heap or stack)...... It would see some random value. That would only work if its static.....

liebman, 2020-05-11:
I'll give your update a try this weekend.

tablatronix, 2020-05-11:
oh your right its not static, hmm I think I just reused it, it was for something else originally oops.

well this should take care of it, I think

commit 65151901 (2020-05-11): fixes #1046