tzapu/WiFiManager #1047: Accesspoint not visible after "autoconnect()"

JochenKoehler69, 2020-05-05:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

## POST SERIAL OUTPUT !

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [x] Master
- [ ] Development

**Esp8266/Esp32:**
- [x] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [x] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [x] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

### Description

I can't see the ESP as accesspoint on any WIFI-device after starting the autoconnect() function.
Is there any additional debug functionality I can active to may be detect the error?

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp
#include <WiFiManager.h>

// Set web server port number to 80
WiFiServer server(80);

// Variable to store the HTTP request
String header;

void setup() {
#ifdef DEBUG
	uart_div_modify(0, UART_CLK_FREQ / 115200);
	Serial.begin(115200);
	gdbstub_init();
#else
	Serial.begin(115200);
#endif

  // WiFiManager
  // Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wifiManager;

  // Uncomment and run it once, if you want to erase all the stored information
  // wifiManager.resetSettings();

  // set custom ip for portal
  //wifiManager.setAPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));

  // fetches ssid and pass from eeprom and tries to connect
  // if it does not connect it starts an access point with the specified name
  // here  "AutoConnectAP"
  // and goes into a blocking loop awaiting configuration
  // wifiManager.autoConnect("AutoConnectAP", "password");
  // or use this for auto generated name ESP + ChipID
  wifiManager.autoConnect();

  // if you get here you have connected to the WiFi
  Serial.println("Connected.");

  server.begin();
}

```

### Debug Messages

```
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Status:
*WM: 0
*WM: No saved credentials
*WM: Connection result: 
*WM: 0
*WM: 
*WM: Configuring access point... 
*WM: ESP206098
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
```

tablatronix, 2020-05-05:
Try a full erase, and retry it, sometimes the softap wont start , this version does not detect failures, also make sure your esp lib is up to date. Might also try our development branch

JochenKoehler69, 2020-05-05:
I tried the full erase already (by uncommenting `wifiManager.resetSettings()` once  - without success. I try now to update the esp8266 libs ...

JochenKoehler69, 2020-05-05:
Just updated all libraries and ESP lib to 2.7.0 and also changed to 0.15.0-beta. 
The serial output changed a little:

> *WM: 
*WM: AutoConnect
bcn 0
del if1
usl
mode : sta(4c:11:ae:03:25:12)
add if0
*WM: Connecting as wifi client...
*WM: No saved credentials
*WM: Connection result: 
*WM: 0
del if0
usl
mode : softAP(4e:11:ae:03:25:12)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
*WM: 
*WM: Configuring access point... 
*WM: ESP206098
bcn 0
del if1
usl
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started

But accesspoint is still not visible from any other WIFI device ...

tablatronix, 2020-05-05:
You need to do a real erase