tzapu/WiFiManager #1048: Dev - ESP32 - too many initializers for 'const wifi_country_t'

Craigzyc, 2020-05-11:
I get this error even in a completely empty sketch with only #include <WiFiManager.h>
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [ *] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [ *] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ *] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)
ESP32 1.0.4

### Description

too many initializers for 'const wifi_country_t'

### Settings in IDE

Module:  			ESP32 Dev Module

Additional libraries: 

### Sketch

```
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Messages

Arduino: 1.8.12 (Mac OS X), Board: "ESP32 Dev Module, QIO, 80MHz, 4MB (32Mb), 921600, None"

In file included from /Documents/Arduino/libraries/WiFiManager/WiFiManager.h:94:0,
                 from /Documents/Arduino/sketch_may11a/sketch_may11a.ino:1:
/Documents/Arduino/libraries/WiFiManager/strings_en.h:348:100: error: too many initializers for 'const wifi_country_t'
 const wifi_country_t WM_COUNTRY_US{"US",1,11,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};
                                                                                                    ^
/Documents/Arduino/libraries/WiFiManager/strings_en.h:348:100: error: invalid conversion from 'int' to 'wifi_country_policy_t' [-fpermissive]
/Documents/Arduino/libraries/WiFiManager/strings_en.h:349:100: error: too many initializers for 'const wifi_country_t'
 const wifi_country_t WM_COUNTRY_CN{"CN",1,13,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};
                                                                                                    ^
/Documents/Arduino/libraries/WiFiManager/strings_en.h:349:100: error: invalid conversion from 'int' to 'wifi_country_policy_t' [-fpermissive]
/Documents/Arduino/libraries/WiFiManager/strings_en.h:350:100: error: too many initializers for 'const wifi_country_t'
 const wifi_country_t WM_COUNTRY_JP{"JP",1,14,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};
                                                                                                    ^
/Documents/Arduino/libraries/WiFiManager/strings_en.h:350:100: error: invalid conversion from 'int' to 'wifi_country_policy_t' [-fpermissive]
exit status 1
Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

tablatronix, 2020-05-11:
When was the last time you pulled?

CraigZych, 2020-05-11:
Yesterday

tablatronix, 2020-05-11:
Are you sure cause this was fixed a while back, what esp32 version you using ?

tablatronix, 2020-05-11:
https://github.com/espressif/arduino-esp32/blob/d5e2bb12ca02ae9066e9dad84d9dbf268aca6fa3/tools/sdk/include/esp32/esp_wifi_types.h#L46

CraigZych, 2020-05-11:
I will give it another try tonight with a fresh pull

tablatronix, 2020-05-11:
Sounds more like you are on an old esp32 version, this was changed in 2.5.2 I think

CraigZych, 2020-05-11:
How does that versioning relate to the esp32? I tried it on 1.0.4 and 1.0.3 of the arduino core for esp32

tablatronix, 2020-05-11:
oh my bad, let me check, should be latest 1.0.4

CraigZych, 2020-05-11:
I just tried. 
Esp8266 on 2.7.1, no issues
Esp32 on 1.0.4 shows that issue

So is this an issue with the arduino core then?

tablatronix, 2020-05-11:
not sure you said you are using develop branch of wm and its up to date ?

I will try to reproduce on arduino , I only use platformio.

can you look in strings.h and make sure it matches the struct ?

```C++
const wifi_country_t WM_COUNTRY_US{"US",1,11,CONFIG_ESP32_PHY_MAX_TX_POWER,WIFI_COUNTRY_POLICY_AUTO};
```

Craigzyc, 2020-05-12:
if we are talking about strings_en.h yes it matches that. 
Interestingly though, if I comment out the if statement so that both the esp8266 and esp32 use the version intended for the ESP8266 it doesn't throw that error. 

`const wifi_country_t WM_COUNTRY_US{"US",1,11,WIFI_COUNTRY_POLICY_AUTO};`

After that I'm only left with the error:
`Documents/Arduino/libraries/WiFiManager/WiFiManager.h:357:5: error: 'wifi_event_id_t' does not name a type
     wifi_event_id_t wm_event_id;
     ^`

tablatronix, 2020-05-12:
you are missing the same argument count
CONFIG_ESP32_PHY_MAX_TX_POWER

tablatronix, 2020-05-12:
i just added that last one uh oh

Craigzyc, 2020-05-12:
ok, my bad. I get it now. I didn't understand what it meant with esp32 with staging. I manually reinstalled the esp32 into arduino using the manual method on the development branch, and it works.

Sorry to waste your time.

closed 2020-05-12 by Craigzyc

tablatronix, 2020-05-12:
Yeah if you use staging you have to manually update it via git or download. Its bleeding edge unstable code