tzapu/WiFiManager #1050: Char* conversion for default_value errornous when adding WiFiManagerParameter();

zen85, 2020-05-11:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [X] Development (OTA)

**Esp8266/Esp32:**
- [X] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [X] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [X] 2.6.3
- [ ] staging (master/dev)

### Description

Hi, 

When we add a custom WiFiManagerParameter with a normal char* to initialize the default value it has a strange conversion error. Following code: 

```
char mqtt_broker[12] = "192.168.1.108";
custom_mqtt_broker = WiFiManagerParameter("mqttbroker", "MQTT Broker", mqtt_broker, sizeof(mqtt_broker));
Serial.println(custom_mqtt_broker.getValue());
Serial.println(mqtt_broker);`
```
puts out on the Serial: 
```
⸮⸮⸮?168.1.10
192.168.1.108
```

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp

#include <FS.h>                   //this needs to be first, or it all crashes and burns...
#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson
#include <WiFiManager.h> //https://github.com/tzapu/WiFiManager WiFi Configuration Magic
#include <PubSubClient.h>
#include <ESP8266WiFi.h>

#define USEOTA
// enable OTA
#ifdef USEOTA
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#endif

#define TRIGGER_PIN 0 //Pin used to start AccessPoint to configure device - Just ground GPIO for 2 seconds

WiFiManager wm;
ESP8266WebServer server(80);

WiFiClient espClient;
PubSubClient client(espClient);
[... 399 lines cut ...]
<form action='/exit'    method='get'><button>Exit</button></form><br/>
 
Setup mode...
*WM: [1] AutoConnect 
*WM: [2] esp_wifi_set_country: US 
*WM: [1] AutoConnect: ESP Already Connected 
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping 
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 192.168.1.109
Connected to wifi network!
IOT_ESP_6184038
192.168.1.109
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip: 
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan completed in 2182 ms
-------------
⸮⸮⸮?168.1.10
192.168.1.108
192.168.1.108

attempting connecting to mqtt

```

rvt, 2020-05-12:
You need to let c++ decide  for the size: 
So use ‘char mqtt_broker[] = "192.168.1.108";’
And  male sure that the length parameter for WiFiManagerParameter is large enough for a
An IP address: 16 chars

zen85, 2020-05-12:
> You need to let c++ decide for the size:
> So use ‘char mqtt_broker[] = "192.168.1.108";’
> And male sure that the length parameter for WiFiManagerParameter is large enough for a
> An IP address: 16 chars

unfortunatly this did not work and does not change anything no matter what i do there... the error occurs when i do the following:

- i setup the device, put in my wifi and/or save params.
- the device connects... everything works... also after restarting.
- when i enter the params page again and set a new or the same value and save it... suddenly "custom_mqtt_broker.getValue()" messes up the value and instead of "192.168.1.108" says "⸮⸮⸮?168.1.10"

it is so weird that this sequence of events works totally fine till it breaks...

tablatronix, 2020-05-12:
Are you sure the parameters are still in scope?

rvt, 2020-05-12:
You are re-assigning custom_mqtt_broker in your setup pfunction , don’t do it like that.

zen85, 2020-05-13:
ok... progress :)

> Are you sure the parameters are still in scope?
> 
> Maybe reduce your test sketch so we can reproduce, there is too much going on there, try the examples

i did that and then 

> You are re-assigning custom_mqtt_broker in your setup pfunction , don’t do it like that.

got rid of the weirdness... but i cant find a way to update the value of custom_mqtt_broker on runtime... so the value i get back at custom_mqtt_broker.getValue() will always be 192.168.1.108 no matter what i do in the portal and therefor the param-interface will always show that value. is there a method to do this cleanly?

```
#include <FS.h>                   //this needs to be first, or it all crashes and burns...
#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson
#include <WiFiManager.h> //https://github.com/tzapu/WiFiManager WiFi Configuration Magic
#include <PubSubClient.h>
#include <ESP8266WiFi.h>

WiFiManager wm;
ESP8266WebServer server(80);

//############################################# CUSTOM PARAMETERS FOR THE WIFI MANAGER ##########################################

char mqtt_broker[16] = "192.168.100.108";

WiFiManagerParameter custom_mqtt_broker("mqttbroker", "MQTT Broker", mqtt_broker, 16);

//################################################### GENERAL VARIABLES #########################################################
bool blockWM = true; // Change this to false if you want your code to continue to run on the loop void even if you are not conected to any wifi.
//###############################################################################################################################

void saveParamCallback() {
  Serial.println("****************Should save params");
  Serial.println(mqtt_broker);
  Serial.println(custom_mqtt_broker.getValue());

  strcpy(mqtt_broker, custom_mqtt_broker.getValue());
  
  Serial.println(mqtt_broker);
  Serial.println(custom_mqtt_broker.getValue());
  
  DynamicJsonDocument json(1024);
 
  json["mqtt_broker"] = mqtt_broker;
 
    
  File configFile2 = SPIFFS.open("/config.json", "w");
    if (!configFile2) {
      Serial.println("failed to open config file for writing");
    }
    
  serializeJson(json, Serial);
  serializeJson(json, configFile2);

  configFile2.close();
  //end save
  }

void handleNotFound()
{
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET) ? "GET" : "POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i = 0; i < server.args(); i++)
[... 130 lines cut ...]
  wm.setMenu(menu); // custom menu, pass vector

  // set country
  wm.setCountry("US"); // setting wifi country seems to improve OSX soft ap connectivity, may help others as well

  Serial.println("Setup mode...");
  //wifiManager.resetSettings();
  
  setupDeviceWM();

  wm.setClass("invert");
  Serial.println("-------------");
  Serial.println(String(custom_mqtt_broker.getValue()));
  Serial.println(custom_mqtt_broker.getValue());
  Serial.println(String(mqtt_broker));
  Serial.println(mqtt_broker);

}

void loop()
{
  
  loopDeviceWM();

}

tablatronix, 2020-05-13:
I have no idea, your sketch still contains other stuff not needed to reproduce, hell you have 2 webservers running...

zen85, 2020-05-13:
you are right - you need to know my motivation to know why i left that part in:

the basic idea behind the sketch was to take the sketch posted on the end of this:
https://github.com/tzapu/WiFiManager/issues/656

and to merge it with the FS Example

zen85, 2020-05-13:
i refactored and still got to the root of the problem:

when i configure the first time everything is cool. after entering the configportal while on wireless with the local ip the fields are empty... therefor when i change a value in the custom parameters i rewrite the existing ones with "" - i wanted to get around of that and so my code seems, and quite probably is, got kind of weird...

so here is this version for reproducing that:
`#include <FS.h>          // this needs to be first, or it all crashes and burns...
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager
#include <ArduinoJson.h> // https://github.com/bblanchon/ArduinoJson

#ifdef ESP32
  #include <SPIFFS.h>
#endif

//define your default values here, if there are different values in config.json, they are overwritten.
char mqtt_server[40];
char mqtt_port[6]  = "8080";
char api_token[32] = "YOUR_API_TOKEN";

//default custom static IP
char static_ip[16] = "10.0.1.56";
char static_gw[16] = "10.0.1.1";
char static_sn[16] = "255.255.255.0";

//flag for saving data
bool shouldSaveConfig = false;

//callback notifying us of the need to save config
void saveConfigCallback () {
  Serial.println("Should save config");
  shouldSaveConfig = true;
}

void setupSpiffs(){
  //clean FS, for testing
  // SPIFFS.format();

  //read configuration from FS json
  Serial.println("mounting FS...");

  if (SPIFFS.begin()) {
    Serial.println("mounted file system");
    if (SPIFFS.exists("/config.json")) {
      //file exists, reading and loading
      Serial.println("reading config file");
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile) {
        Serial.println("opened config file");
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);

        configFile.readBytes(buf.get(), size);
       DynamicJsonDocument jsonBuffer(1024);
        //DynamicJsonBuffer jsonBuffer;
        deserializeJson(jsonBuffer, buf.get());
        //JsonObject& json = jsonBuffer.parseObject(buf.get());
        serializeJson(jsonBuffer, Serial);
        //json.printTo(Serial);

        if (!jsonBuffer.isNull()) {        
    
          Serial.println("\nparsed json");

          strcpy(mqtt_server, jsonBuffer["mqtt_server"]);
          strcpy(mqtt_port, jsonBuffer["mqtt_port"]);
          strcpy(api_token, jsonBuffer["api_token"]);

          // if(json["ip"]) {
          //   Serial.println("setting custom ip from config");
          //   strcpy(static_ip, json["ip"]);
[... 98 lines cut ...]
    if (!configFile) {
      Serial.println("failed to open config file for writing");
    }

    serializeJson(jsonBuffer, Serial);
    serializeJson(jsonBuffer, configFile);

    //jsonBuffer.prettyPrintTo(Serial);
    //jsonBuffer.printTo(configFile);
    configFile.close();
    //end save
    shouldSaveConfig = false;
  }

  Serial.println("local ip");
  Serial.println(WiFi.localIP());
  Serial.println(WiFi.gatewayIP());
  Serial.println(WiFi.subnetMask());
}

void loop() {
  // put your main code here, to run repeatedly:

  wm.process();
}`

tablatronix, 2020-05-13:
ok this is pretty much the example, so where exactly is the problem, where in the code can we see the issue ? can you add a line in there or serial print ?

zen85, 2020-05-13:
> ok this is pretty much the example, so where exactly is the problem, where in the code can we see the issue ? can you add a line in there or serial print ?

ok - above sketch never saved changed parameters to spiff... could not work anyway... but please try this one. there is an obvious serial print saying what i expect and this is not matching up with what i get at the end of the setup:
```

#include <FS.h>                   //this needs to be first, or it all crashes and burns...
#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson
#include <WiFiManager.h> //https://github.com/tzapu/WiFiManager WiFi Configuration Magic

WiFiManager wm;

//############################################# CUSTOM PARAMETERS FOR THE WIFI MANAGER ##########################################

char mqtt_broker[16] = "192.168.100.108";

WiFiManagerParameter custom_mqtt_broker("mqttbroker", "MQTT Broker", mqtt_broker, 16);

//################################################### GENERAL VARIABLES #########################################################
bool blockWM = true; // Change this to false if you want your code to continue to run on the loop void even if you are not conected to any wifi.
//###############################################################################################################################

void saveParamCallback() {
  Serial.println("****************Should save params");
  Serial.println(mqtt_broker);
  Serial.println(custom_mqtt_broker.getValue());

  strcpy(mqtt_broker, custom_mqtt_broker.getValue());
  
  Serial.println(mqtt_broker);
  Serial.println(custom_mqtt_broker.getValue());
  
  DynamicJsonDocument json(1024);
 
  json["mqtt_broker"] = mqtt_broker;
 
    
  File configFile2 = SPIFFS.open("/config.json", "w");
    if (!configFile2) {
      Serial.println("failed to open config file for writing");
    }
    
  serializeJson(json, Serial);
  serializeJson(json, configFile2);

  configFile2.close();
  //end save
  }

/*
  This void sets the device with the WiFiManager from tzapu with custom parameters.
  Needs to be called only in the setup void.
*/

void setupDeviceWM()
{
  //Serial.println("setupDeviceWM() is running");

  // callbacks
  wm.setConfigPortalBlocking(blockWM);
  
  String ssid = "IOT_ESP_" + String(ESP.getChipId());

  if (wm.autoConnect(ssid.c_str()))
  {
    //if you get here you have connected to the WiFi
    Serial.println("Connected to wifi network!");
    Serial.println(ssid);
    Serial.println(WiFi.localIP());
    
    WiFi.mode(WIFI_STA);
    wm.startWebPortal();
  }
  else
  {
    Serial.println("Could not connect with WiFi!");
  }

  // call the code down to activate wifi so users can configure the device, event if it's connected to the local network
  //wm.startConfigPortal("IOT_Device");
  //
  }

/*
  This void needs to be called in the loop void so it can handle the WM and the webportal.
*/

void loopDeviceWM()
{
  wm.process();
  
}

int setupSpiffs() {
  //read configuration from FS json
  Serial.println("mounting FS...");

  if (SPIFFS.begin()) {
    Serial.println("mounted file system");
    if (SPIFFS.exists("/config.json")) {
      //file exists, reading and loading
      Serial.println("reading config file");
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile) {
        Serial.println("opened config file");
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);

        configFile.readBytes(buf.get(), size);
        DynamicJsonDocument jsonBuffer(1024);
        deserializeJson(jsonBuffer, buf.get());
        serializeJson(jsonBuffer, Serial);
      
        if (!jsonBuffer.isNull()) {
          Serial.println("\nparsed json");

          strcpy(mqtt_broker, jsonBuffer["mqtt_broker"]);

          Serial.println("loaded Json");
          Serial.println("now mqtt broker is: ");
          Serial.println(mqtt_broker);
          
          return 0;
          
        } else {
          Serial.println("failed to load json config");
          return 1;
        }
        configFile.close();
      }
    }
    else
    {
      Serial.println("config json doesnt exist");
      return 1;
    }
   
  } else {
    Serial.println("failed to mount FS");
    return 1;
  }
  //end read

}

void setup()
{

  Serial.begin(115200);
  delay(3000);

  //diese funktion gibt 0 zurück wenn eine config json da ist. 
  if(setupSpiffs() == 0){
    Serial.println("SO, NOW:");
    Serial.println(mqtt_broker);
    //custom_mqtt_broker = WiFiManagerParameter("mqttbroker", "MQTT Broker", mqtt_broker, 16);
  }
  else
  {
    
  }
  // at this point the parameters are either empty because there is no config or full because there is a config
  

  wm.setSaveParamsCallback(saveParamCallback);

 
  wm.addParameter(&custom_mqtt_broker);

  std::vector<const char *> menu = {"wifi", "info", "param", "close", "sep", "erase", "restart", "exit"};
  wm.setMenu(menu); // custom menu, pass vector

  // set country
  wm.setCountry("US"); // setting wifi country seems to improve OSX soft ap connectivity, may help others as well

  Serial.println("Setup mode...");
  //wifiManager.resetSettings();
  
  setupDeviceWM();

  wm.setClass("invert");
  Serial.println("**** Issue here **** - the next four lines should be the mqtt broker adress after changing in portal and restarting but apparently custom_mqtt_broker.getValue() never gets it value of the config.json");
  Serial.println(String(custom_mqtt_broker.getValue()));
  Serial.println(custom_mqtt_broker.getValue());
  Serial.println(String(mqtt_broker));
  Serial.println(mqtt_broker);

}

void loop()
{
  
  loopDeviceWM();

}

tablatronix, 2020-05-13:
Why are you copying it back into the same variable though ?

`          strcpy(mqtt_broker, jsonBuffer["mqtt_broker"]);`

tablatronix, 2020-05-13:
you also should not be doing this much stuff in your callbacks, just set some flag and then save your spiffs later.

zen85, 2020-05-13:
> Why are you copying it back into the same variable though ?
> 
> ` strcpy(mqtt_broker, jsonBuffer["mqtt_broker"]);`

this is just done in setup... "mqtt_broker" is always alright. it behaves perfectly even when i put it in the code... put when i enter the configportal > setup it shows the first value it had and the changes are not reflected there... if i could just put something like "custom_mqtt_broker = mqtt_broker" everything would work :) . if i comment that out nothing is updated... 

the callback also works fine i think. the config.json reflects everything i change. otherwise "strcpy(mqtt_broker, jsonBuffer["mqtt_broker"]);" would not work. its just the UI thats unusable because it will always say "192.168.100.108".

tablatronix, 2020-05-13:
Looks fine to me

```php
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip: 
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan completed in 2182 ms
**** Issue here **** - the next four lines should be the mqtt broker adress after changing in portal and restarting but apparently custom_mqtt_broker.getValue() never gets it value of the config.json
192.168.100.108
192.168.100.108
192.168.100.108
192.168.100.108
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.1.84 
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [2] WiFi Scan completed in 2183 ms
*WM: [2] <- HTTP Param 
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [3] Sent param page 
*WM: [2] <- HTTP WiFi save  
*WM: [3] Method: POST
*WM: [2] Parameters 
*WM: [2] -------------------- 
*WM: [2] mqttbroker: 192.168.100.112
*WM: [2] -------------------- 
****************Should save params
192.168.100.108
192.168.100.112
192.168.100.112
192.168.100.112
{"mqtt_broker":"192.168.100.112"}*WM: [3] Sent param save page 
```
I had to remove spiffs, it was crashing my esp, maybe your problem is with spiffs..

Not really understanding what you are trying to do though..
I am going to assume your issue is loading or saving to spiffs or json, so remove all the other WM code and use plain variables and see if you have the same problem.

Is this a typo?
`custom_mqtt_broker.getValue() never gets it value of the config.json`

because that doesn't make sense, wm params have nothing to do with json.. they are whatever you set them to or what you submit them as.

zen85, 2020-05-13:
yes... it works so far... if you restart the esp now it does not anymore

and it gets extremly close to working if i put: 
  if(setupSpiffs() == 0){
    Serial.println("SO, NOW:");
    Serial.println(mqtt_broker);
    custom_mqtt_broker = WiFiManagerParameter("mqttbroker", "MQTT Broker", mqtt_broker, sizeof(mqtt_broker));
  }

tablatronix, 2020-05-13:
If it is not set after resetting then you are not reading it from spiffs properly and then setting it..

zen85, 2020-05-13:
> If it is not set after resetting then you are not reading it from spiffs properly and then setting it..
> Look for a bug there, either the spiffs read or the json decode

but it is set properly... restart it after you did what you just did. and then enter the portal on the wifi, go to setup... now you will still see 192.168.100.108 and not 192.168.100.112 while the value of mqtt_broker is correctly 192.168.100.112.

tablatronix, 2020-05-13:
I cannot test that part, my spiffs is not working, reread my previous port I edited it

tablatronix, 2020-05-13:
oops I commented out the spiffs part

I see it now

    custom_mqtt_broker = WiFiManagerParameter("mqttbroker", "MQTT Broker", mqtt_broker, sizeof(mqtt_broker));

tablatronix, 2020-05-13:
```
  wm.setBreakAfterConfig(true);
  wm.setSaveParamsCallback(saveParamCallback);

 
  wm.addParameter(&custom_mqtt_broker);
  Serial.println(String(custom_mqtt_broker.getValue())); // <--- what does this say ?
```

I am curious is it corrupt only after submitting and getting it? Or is this also corrupt ?

zen85, 2020-05-13:
if i dont do that nothing is updated at all and the value will always be 192.168.1.108.... if i do that i get at least ???.168.100.112

> Serial.println(String(custom_mqtt_broker.getValue())); // <--- what does this say ?

SO, NOW:
192.168.100.102
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter: mqttbroker
@⸮⸮?168.100.102 <---------

tablatronix, 2020-05-13:
interesting

tablatronix, 2020-05-13:
oh sorry, I have that commented out, do not do   // if(setupSpiffs() == 0){
remove that whole block

tablatronix, 2020-05-13:
Then change this
          strcpy(mqtt_broker, jsonBuffer["mqtt_broker"]);
to use a different VARIABLE 
say mqtt_broker_RESTORE[];

then change your spiffs restore to 
```
  // if(setupSpiffs() == 0){
  //   Serial.println("SO, NOW:");
  //   Serial.println(mqtt_broker);
  //   //custom_mqtt_broker = WiFiManagerParameter("mqttbroker", "MQTT Broker", mqtt_broker_RESTORE, 16);
  // }
  // else
  // {
```

zen85, 2020-05-13:
> Then change this
> strcpy(mqtt_broker, jsonBuffer["mqtt_broker"]);
> to use a different VARIABLE
> say mqtt_broker_RESTORE[];
> 
> then change your spiffs restore to
> 
> ```
>    if(setupSpiffs() == 0){
>      Serial.println("SO, NOW:");
>      Serial.println(mqtt_broker);
>      //custom_mqtt_broker = WiFiManagerParameter("mqttbroker", "MQTT Broker", mqtt_broker_RESTORE, 16); // <-- NEW VAR
>   }
> ```
> 
> Sorry it took me like 30 minutes and 2 reboots to get my damn serial ports working, and now I have to format spiffs to actually run your code.

heyhey.... its completly amazing that there is somebody out there... i sunk about 20 hours in this now and it feels truly heartwarming that you are taking care...

so i changed my code to this but i still have the same outcome:
```
#include <FS.h>                   //this needs to be first, or it all crashes and burns...
#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson
#include <WiFiManager.h> //https://github.com/tzapu/WiFiManager WiFi Configuration Magic

WiFiManager wm;

//############################################# CUSTOM PARAMETERS FOR THE WIFI MANAGER ##########################################

char mqtt_broker[] = "192.168.100.108";
char mqtt_broker_RESTORE[sizeof(mqtt_broker)];

WiFiManagerParameter custom_mqtt_broker("mqttbroker", "MQTT Broker", mqtt_broker, sizeof(mqtt_broker));

//################################################### GENERAL VARIABLES #########################################################
bool blockWM = true; // Change this to false if you want your code to continue to run on the loop void even if you are not conected to any wifi.
//###############################################################################################################################

void saveParamCallback() {
  Serial.println("****************Should save params");
  Serial.println(mqtt_broker);
  Serial.println(custom_mqtt_broker.getValue());

  strcpy(mqtt_broker, custom_mqtt_broker.getValue());
  
  Serial.println(mqtt_broker);
  Serial.println(custom_mqtt_broker.getValue());
  
  DynamicJsonDocument json(1024);
 
  json["mqtt_broker"] = mqtt_broker;
 
    
  File configFile2 = SPIFFS.open("/config.json", "w");
    if (!configFile2) {
      Serial.println("failed to open config file for writing");
    }
    
  serializeJson(json, Serial);
  serializeJson(json, configFile2);

  configFile2.close();
  //end save
  }

/*
  This void sets the device with the WiFiManager from tzapu with custom parameters.
  Needs to be called only in the setup void.
*/

[... 117 lines cut ...]
  wm.setMenu(menu); // custom menu, pass vector

  // set country
  wm.setCountry("US"); // setting wifi country seems to improve OSX soft ap connectivity, may help others as well

  Serial.println("Setup mode...");
  //wifiManager.resetSettings();
  
  setupDeviceWM();

  wm.setClass("invert");
  Serial.println("**** Issue here **** - the next four lines should be the mqtt broker adress after changing in portal and restarting but apparently custom_mqtt_broker.getValue() never gets it value of the config.json");
  Serial.println(String(custom_mqtt_broker.getValue()));
  Serial.println(custom_mqtt_broker.getValue());
  Serial.println(String(mqtt_broker));
  Serial.println(mqtt_broker);

}

void loop()
{
  
  loopDeviceWM();

}

tablatronix, 2020-05-13:
Do you know how to init format spiffs?

tablatronix, 2020-05-13:
nm,   // SPIFFS.format();

tablatronix, 2020-05-13:
And you have the same problem with the autoconnectwithparams example?

tablatronix, 2020-05-13:
ah crap is the old json lib

zen85, 2020-05-13:
> nm, // SPIFFS.format();

i like that neat little tool: https://github.com/nodemcu/nodemcu-flasher

> And you have the same problem with the autoconnectwithparams example?

no

yes... arduinojson6...

tablatronix, 2020-05-14:
ok I have the same issue , no idea what is the deal, let me look into it

tablatronix, 2020-05-14:
in the mean time use setValue, it has something to do with calling it twice and it not setting the or overriding the obj memory properly.

This should work fine, adjust length to match whatever you set it to before etc.
` custom_mqtt_broker.setValue(mqtt_broker_RESTORE,20);`

tablatronix, 2020-05-14:
So here is the reproduction

```c++
void setup()
{

  Serial.begin(115200);

  char mqtt_broker[] = "192.168.100.108";
  WiFiManagerParameter custom_mqtt_broker("mqttbroker", "MQTT Broker", mqtt_broker, 20);
  custom_mqtt_broker = WiFiManagerParameter("mqttbroker", "MQTT Broker", mqtt_broker, 20);
  wm.addParameter(&custom_mqtt_broker);
  Serial.println("custom_mqtt_broker IS");
  Serial.println(String(custom_mqtt_broker.getValue()));

  wm.addParameter(&custom_mqtt_broker);

....

```

tablatronix, 2020-05-14:
Maybe someone can look at this for me and help me fix it or understand wtf is going on in the param class

tablatronix, 2020-05-14:
```
custom_mqtt_broker FIRST
Address of x is 0x3fff008c
192.168.100.108
preexisting ID

custom_mqtt_broker SECOND
Address of x is 0x3ffeffac
`
custom_mqtt_broker THIRD
Address of x is 0x3ffeffac
192.168.100.108
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter: 
```

hmm ok so something I do not understand about constructors.. NOT the same memory..

rvt, 2020-05-14:
WiFiManagerParameter is missing a copy assignment operator. https://en.cppreference.com/w/cpp/language/copy_assignment

This:` WiFiManagerParameter custom_mqtt_broker("mqttbroker", "MQTT Broker", mqtt_broker, 20);`

is different from: `custom_mqtt_broker = WiFiManagerParameter("mqttbroker", "MQTT Broker", mqtt_broker, 20);` 

`_value` get's deleted in the destructor during the assignment. (compiler creates it's own assignment but that is incorrect for this situation).

tablatronix, 2020-05-14:
Thanks, Yeah I was reading about correct assignment but did not really understand what it meant. I am mostly curious how to handle this, how to make it work as expected or prevent it. As it is something users are going to try. Any relevant suggested reading? My c++ knowledge has some holes. Ill start googling.m now that I know what to look for

rvt, 2020-05-14:
I agree that users will try to do this so we must prevent users from doing it.

I will make a pull request to handle this.

dontsovcmc, 2020-05-14:
Maybe it's better to deny coping of WiFiParameter and add function "setValue()"? 
No reasons to copy WiFiParameters.

tablatronix, 2020-05-14:
We have setValue and we can also check _id and deny I can throw a compiler error. It would be cool to just allow it though if its not too much trouble. At this point I really want to see how

tablatronix, 2020-05-14:
So is this correct or the gist?

```c++
WiFiManagerParameter& operator=(const WiFiManagerParameter& rhs);

...

WiFiManagerParameter& WiFiManagerParameter::operator=(const WiFiManagerParameter& rhs){
  (*this->_value) = (*rhs._value);
  return *this;
}
```

rvt, 2020-05-14:
We could use `WiFiManagerParameter(const WiFiManagerParameter&) = delete;`

> No reasons to copy WiFiParameters.

Somebody will always find a reason :)

dontsovcmc, 2020-05-14:
I found the reason: More code, more bugs.

Here is one of them:

We have memory leak if call setValue() in our program.
_value  = new char[_length + 1]; //_value is't null cause setValue called early in constructor

rvt, 2020-05-14:
I like this one better:

    ```
WiFiManagerParameter& operator=( WiFiManagerParameter rhs) {
      if (&rhs == this) {
        return *this;
      }
      std::swap(_id , rhs._id);
      std::swap(_label , rhs._label);
      std::swap(_value , rhs._value);
      std::swap(_value, rhs._value);
      std::swap(_length , rhs._length);
      std::swap(_labelPlacement , rhs._labelPlacement);
      std::swap(_customHTML , rhs._customHTML);

      return *this;
    }
```

> _value = new char[_length + 1]; //_value is't null cause setValue called early in constructor

That will be deleted again in the destructor

The copy assignment could be further optimized by using move semantics.
**But the whole dynamic allocation within an embedded system should be prevent at all cost.**

@dontsovcmc Perhaps you made a good point, just prevent copy assignment in the first place?

That will show also good design methodology for embedded software developers.

dontsovcmc, 2020-05-14:
@rvt "That will be deleted again in the destructor."
No,  I mean:
```
char value[16] = "192.168.100.108";
WiFiManagerParameter custom_mqtt_broker("mqttbroker", "MQTT Broker", value, 20);
custom_mqtt_broker.setValue(new_value); //here _value = new char[_length + 1];
```

rvt, 2020-05-14:
@dontsovcmc yes you are absolutely right, setValue does not release memory of previously allocated memory

zen85, 2020-05-14:
@tablatronix @rvt 
sorry i left the discussion, it was 3 in the morning where i live and i had to work.
When i woke up i checked immididatly and tried the solution before i had to leave and you guys made my day. it finally works. thank you so much!

I have to admit that i did not know about setValue() and never stumpled upon it in my extensive research. I am just wondering why this is considered a "meantime"-solution since it does exactly what i needed? is it not optimal to solve it like that?

also: the working sketch for anybody making it down here:
```
#include <FS.h>                   //this needs to be first, or it all crashes and burns...
#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson
#include <WiFiManager.h> //https://github.com/tzapu/WiFiManager WiFi Configuration Magic

WiFiManager wm;

//############################################# CUSTOM PARAMETERS FOR THE WIFI MANAGER ##########################################

char mqtt_broker[] = "192.168.100.108";
//char mqtt_broker_RESTORE[sizeof(mqtt_broker)];

WiFiManagerParameter custom_mqtt_broker("mqttbroker", "MQTT Broker", mqtt_broker, sizeof(mqtt_broker));

//################################################### GENERAL VARIABLES #########################################################
bool blockWM = true; // Change this to false if you want your code to continue to run on the loop void even if you are not conected to any wifi.
//###############################################################################################################################

void saveParamCallback() {
  Serial.println("****************Should save params");
  Serial.println(mqtt_broker);
  Serial.println(custom_mqtt_broker.getValue());

  strcpy(mqtt_broker, custom_mqtt_broker.getValue());
  
  Serial.println(mqtt_broker);
  Serial.println(custom_mqtt_broker.getValue());
  
  DynamicJsonDocument json(1024);
 
  json["mqtt_broker"] = mqtt_broker;
 
    
  File configFile2 = SPIFFS.open("/config.json", "w");
    if (!configFile2) {
      Serial.println("failed to open config file for writing");
    }
    
  serializeJson(json, Serial);
  serializeJson(json, configFile2);

  configFile2.close();
  //end save
  }

/*
  This void sets the device with the WiFiManager from tzapu with custom parameters.
  Needs to be called only in the setup void.
*/

void setupDeviceWM()
{
  //Serial.println("setupDeviceWM() is running");

  // callbacks
  wm.setConfigPortalBlocking(blockWM);
  
  String ssid = "IOT_ESP_" + String(ESP.getChipId());

  if (wm.autoConnect(ssid.c_str()))
  {
    //if you get here you have connected to the WiFi
    Serial.println("Connected to wifi network!");
    Serial.println(ssid);
    Serial.println(WiFi.localIP());
    
    WiFi.mode(WIFI_STA);
    wm.startWebPortal();
  }
  else
  {
    Serial.println("Could not connect with WiFi!");
  }

  // call the code down to activate wifi so users can configure the device, event if it's connected to the local network
  //wm.startConfigPortal("IOT_Device");
  //
  }

/*
  This void needs to be called in the loop void so it can handle the WM and the webportal.
*/

void loopDeviceWM()
{
  wm.process();
  
}

int setupSpiffs() {
  //read configuration from FS json
  Serial.println("mounting FS...");

  if (SPIFFS.begin()) {
    Serial.println("mounted file system");
    if (SPIFFS.exists("/config.json")) {
      //file exists, reading and loading
      Serial.println("reading config file");
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile) {
        Serial.println("opened config file");
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);

        configFile.readBytes(buf.get(), size);
        DynamicJsonDocument jsonBuffer(1024);
        deserializeJson(jsonBuffer, buf.get());
        serializeJson(jsonBuffer, Serial);
      
        if (!jsonBuffer.isNull()) {
          Serial.println("\nparsed json");

          strcpy(mqtt_broker, jsonBuffer["mqtt_broker"]);

          Serial.println("loaded Json");
          Serial.println("now mqtt broker is: ");
          Serial.println(mqtt_broker);
          
          return 0;
          
        } else {
          Serial.println("failed to load json config");
          return 1;
        }
        configFile.close();
      }
    }
    else
    {
      Serial.println("config json doesnt exist");
      return 1;
    }
   
  } else {
    Serial.println("failed to mount FS");
    return 1;
  }
  //end read

}

void setup()
{

  Serial.begin(115200);
  delay(3000);

  //diese funktion gibt 0 zurück wenn eine config json da ist. 
  
   if(setupSpiffs() == 0){
     Serial.println("SO, NOW:");
     Serial.println(mqtt_broker);
      //custom_mqtt_broker = WiFiManagerParameter("mqttbroker", "MQTT Broker", mqtt_broker_RESTORE, 16); // <-- NEW VAR
      custom_mqtt_broker.setValue(mqtt_broker, sizeof(mqtt_broker));
  }

  
  // at this point the parameters are either empty because there is no config or full because there is a config
  
  wm.setBreakAfterConfig(true);
  wm.setSaveParamsCallback(saveParamCallback);

 
  wm.addParameter(&custom_mqtt_broker);
  Serial.println(String(custom_mqtt_broker.getValue())); // <--- what does this say ?

  std::vector<const char *> menu = {"wifi", "info", "param", "close", "sep", "erase", "restart", "exit"};
  wm.setMenu(menu); // custom menu, pass vector

  // set country
  wm.setCountry("US"); // setting wifi country seems to improve OSX soft ap connectivity, may help others as well

  Serial.println("Setup mode...");
  //wifiManager.resetSettings();
  
  setupDeviceWM();

  wm.setClass("invert");
  Serial.println(String(custom_mqtt_broker.getValue()));
  Serial.println(custom_mqtt_broker.getValue());
  Serial.println(String(mqtt_broker));
  Serial.println(mqtt_broker);

}

void loop()
{
  
  loopDeviceWM();

}

tablatronix, 2020-05-14:
Thanks all, yeah that is why I asked if this would leak memory, I assumed it would from the NEW value each time if there was any destruction allowed via assignment.

Ill let you all discuss this a little

tablatronix, 2020-05-15:
Made copy assignment private, so compiler will error out if you try this again.

Leaving open for improvements or alternatives.

rvt, 2020-05-15:
> Made copy assignment private, so compiler will error out if you try this again.

Not sure what compiler you are using but with c++11 you can do this:
`WiFiManagerParameter& operator=(const WiFiManagerParameter&) = delete;`

so you don't have to set it private if you just want to prevent all users (including the class self) prevent using the copy assignment.

tablatronix, 2020-05-15:
ooh

jonathanvanpeteghem, 2022-06-15:
Hello, 

I am curently using .setValue() 
The reason is that when i update the parameters, and later go back to the configportal (via button for example) then I want to see the updated values. And .setValue seems the only way for that.

But my question is, is this now safe to use? Forgive my ignorance. I have read the post 1050 / 1341 and 1346. But I cannot understand if the memory leak issue is now gone or not? 

In my case I always restart after going to the configportal (because, when returning from configportal i have to go back to an assync webserver, and that only works with a restart, since the cleanup from the wm webserver leaves "stuff behind")
==> so i guess with my restart, i wont notice the memory leak, since my restart cleaned it up.

But in anycase, it would be kind if you can let me know if the .setValue() is now leak safe.

Best Regards to Tablatronix! Thank you so much for making this wifimanager something that we can use daily.

tablatronix, 2022-06-15:
This is 2 years old, so no idea, pretty sure there have been at least 2 PRs to this