tzapu/WiFiManager #1051: Refactor: static ip flags

dontsovcmc, 2020-05-11:
Sorry if I don't understand all use cases, but doubling static ip flags '_disableIpFields' '_staShowStaticFields' looks terrible.

There are only 2 cases:
1. User want to use static ip address to connect to internet
2. User want to use DHCP

I can refactor it, if needed.

P.S. It will be interesting to research statistic of using WifiManager methods.. I think there are only few popular cases. Others are brake the development.

P.P.S. optionalIPFromString also is outdated.

tablatronix, 2020-05-11:
It is so you can force showing the fields if you are setting them in code or from elsewhere

tablatronix, 2020-05-11:
P.P.S. optionalIPFromString also is outdated. ?

dontsovcmc, 2020-05-11:
1. i can't understand why somebody hardcode static ip and don't want to show it on the menu
2. i don't have any problem. i think we should simplify source code.
3. i don't understand the function of _disableIpFields 
4. optionalIPFromString uses to ESP8266 core 2.1.0 message. it's outdated, so we can remove it and use 
```
IPAddress ip;
  if (ip.fromString(server->arg(FPSTR(S_ip)))) {
    _sta_static_ip = ip;
  }
```

dontsovcmc, 2020-05-14:
@tablatronix offtopic: Why getWiFiSSID()= WiFi_SSID() & getWiFiPass()=WiFi_psk() functions were added? There are the same...

tablatronix, 2020-05-14:
They are public helpers, have better names