tzapu/WiFiManager #1052: wm.process() only returns true once after saving new connection

Craigzyc, 2020-05-12:
In the on demand non blocking. What can we test to know if there is an active connection?
I would have expected wm.process() to return true whenever there is a connection but that doesn't seem to be the case.

Thank you for any help

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [ *] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [ *] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [* ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [* ] staging (master/dev)

### Description
Using the non blocking on demand, how are we supposed to tell if its connected?
I did something along the lines of this inside the loop. 
`if(wm.process()){
    Serial.println("Connected");
}`
The print gets executed once when you make a new connection. It does not get executed on a reboot with a saved connection even when the serial output shows connected. Should it not always be "true" when connected?

### Settings in IDE

Additional libraries: 

### Sketch

```cpp

#include <WiFiManager.h> 
WiFiManager wm;

void setup() {
    WiFi.mode(WIFI_STA); 
    wm.setConfigPortalBlocking(false);
    wm.setCountry("US");     
    wm.autoConnect("StepperDriver");
}

void loop() {
    if(wm.process()){
        Serial.println("Connected");//only makes it here once after a brand new connection is saved
}
```

### Debug Messages

*WM: [1] AutoConnect 
*WM: [2] esp_wifi_set_country: US 
*WM: [2] ESP32 event handler enabled 
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping 
*WM: [1] Connecting to SAVED AP: Samsung Galaxy Note9 5849
*WM: [3] Using Password: /////////
*WM: [3] WiFi station enable 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 192.168.43.192

tablatronix, 2020-05-12:
Same way you normally use autoconnect

`    if(wm.autoConnect("AutoConnectAP")){
        Serial.println("connected...yeey :)");
    }
`
its in the example.

Craigzyc, 2020-05-13:
I understand that. I was looking for a way to test if the connection is active, that only tells me if its started. 
if(Wifi.status() == WL_CONNECTED) does what I was looking for.

Thank you

closed 2020-05-13 by Craigzyc