tzapu/WiFiManager #1057: C++ nerds, help!

tablatronix, 2020-05-19:
I thought this worked, it does not.

Why can I not do.

```
#include <ESP8266mDNS.h>

#ifdef ESP8266MDNS_H
// HOLY CRAP MDNS IS LOADED!!!! yyeehaw
#endif
```

I have several flags like this checking for things loaded, i swear they worked at some point..

tablatronix, 2020-05-19:
I have also seen libraries that check for defines in the users sketch! No idea how they do it..

//sketch.ino
#define DEBUG
#include library.h

//library.?
#ifdef DEBUG
// user enabled debug!!!
#endif 

I think this works in blynk library

rvt, 2020-05-21:
Try checking for` ESP8266MDNS_LEGACY_H`. But why would you want to do it like this?

tablatronix, 2020-05-21:
How else would you do it?

dontsovcmc, 2020-05-22:
You example shouldn't work... I don't know any defines for including files.
Sometimes who code in C writes in headers smth like this:

```
#ifndef _MASTERI2C_h
#define _MASTERI2C_h

....
#endif
```

In this case you can check this define in the code

dontsovcmc, 2020-05-22:
in ESP8266mDNS.h

```
#if !defined(NO_GLOBAL_INSTANCES) && !defined(NO_GLOBAL_MDNS)
// Maps the implementation to use to the global namespace type
//using MDNSResponder = Legacy_MDNSResponder::MDNSResponder; //legacy
using MDNSResponder = esp8266::MDNSImplementation::MDNSResponder; //new

extern MDNSResponder MDNS;
#endif
```

so, you can define NO_GLOBAL_MDNS  to turn off MDNS

tablatronix, 2020-05-22:
I was more interested in detecting if assets were included and using macros to change behavior, whether they are included from user sketch or your h file, checking for include guards does not work. I think it is possible to check defined for stuff in the make environment, platform macros, and build flags, but not include guards, must be how the precompiler combines them all together at once.

I have read that you can find a way to include a 3rd asset and use it but I never got that to work either..

rvt, 2020-05-22:
I would not magically do something just because you found a macro identifier from a third party available (there are exceptions). For sure I would try to avoid at all cost to use the macro preprocessors in implementations.

Nowadays c++ compiler can detect dead code and and not include it in the binary.
For different targets, for example esp8266 vs esp32 it is better to separate the concerns in different classes than prinkle one large codebase with preprocessor directives (if/elif).

As always, there are exceptions....

dontsovcmc, 2020-05-22:
You mean I use my own html? Yes, but I rewrite you code =(. And can't update it without manual diff check with main branch. That's why I want to refactor wm.

If you want to change strings_en.h to another, I think it's more easy to use python scripting in Platformio.
Another way to separate strings into "system" and custom. Custom strings can be changes outside the library.

tablatronix, 2020-05-22:
@dontsovcmc wrong thread

dontsovcmc, 2020-05-22:
> How would you implement user optional features such as filesystem and network interfaces

One of the way to use adapter patterns: https://en.wikipedia.org/wiki/Adapter_pattern ?
So wm should have a some wrapper that proxy requests to target object.

tablatronix, 2020-05-23:
Right, there are a couple patterns we could use here, but more specifically, how do you handle dependencies, and includes specifically external ones.. 

Can you handle that inside classes? Or prevent the compiler from using them at all without having main include them ? pimpl, facade etc ?

rvt, 2020-05-23:
@tablatronix without knowing to much if the details my initial though would be to make interfaces and abstract classes with implementations for each type of device . 

You could also not handle something and let the calling application decide.

For example with SPIFFS vs LittleFS. You could also provide way to let the calling app know you want to save something and you can do with a callback.
All you have to do is provide a method that can accept a reference to a Stream.
In your lib just include `#include <Stream.h>` and provide a method that like like this: `void MyObject::save(Stream& device) {...}` and you do´t care about the filesystem anymore.

tablatronix, 2020-05-23:
Right so by `calling application` you mean the user.. Which is exactly why I posed this question.

You are either asking the user code to either create additional classes or handle callbacks, I am asking how do we NOT.

I agree that this library should be a little more oop

wifi handling
httpd handling
api handling
strings templating
http post var handling
settings

But that is not this discussion.

Obviously if I was going to rewrite the entire library ( I am not ) I would add helper classes for FS and httpd, and you either pass them in, or have another class they have to call,  for Spiffs it would be a memory address for start, and maybe an object for interfacing pairs or something.

I am asking how do you NOT `#include spiffs.h` when the user is not using it without asking them to include or not include additional assets. 

Let the compiler optimize it out ? There has to be some memory wasted still right ?

rvt, 2020-05-23:
If you include spiffs but don't use it the compiler should optimise that all out, given that the library included is proper written.

I just tested that between LittleFS AND SPIFFS.
I normally use LittleFS but when I include `FS.h` binary size does not increase after compilation.
The moment I do `SPIFFS.begin() ` binary size increase by 30K.