tzapu/WiFiManager #1058: Add back button

tablatronix, 2020-05-19:
Sometimes it is hard to find the back button in captive portals, add a back/home button

tablatronix, 2020-05-20:
3c0c3ff93e819f05298df40b06a48147864dcf13

`_showBack` to enable, no setter yet still testing how this should work best, might be better as part of the header

dontsovcmc, 2020-05-22:
f**king flags... design should be more flexible. In this case everybody can easily add any buttons, styles...  WifiManager has terrible class interface, so there are a lot of request about feature, that will be used in 0.1% cases.

First of all, it's needed to separate work with network, webserver, templates.

tablatronix, 2020-05-22:
wm was never designed to be a web interface for personal projects with templating etc

tablatronix, 2020-05-22:
#158 

async server supports some basic templating built in

dontsovcmc, 2020-05-22:
I know, but you increase complexity without good code and autotests, so you have a lot of bugs inside wm, inside framework and etc..
I want to help refactor wm to be flexible and simple.

tablatronix, 2020-05-22:
I have said it before, make an issue and write a roadmap for refactoring, all I have been doing is abstracting stuff out and cleaning it up, so it can be structured and maintained.

I also disagree with it being any more complex or buggy then when I started...

dontsovcmc, 2020-05-22:
I will write some great ideas ;)

tablatronix, 2020-05-22:
Just be aware that the majority of users just install a library, open an example and copy paste code, they are not going to be building assets externally, or frameworks, or even uploading files..

If there is a arduino webserver capable library solution for building webapp, id liek to see it.
I have been building my own vanilla js frameworks for my projects, newer micros might have room for jquery and bootstrap but just because you can doesn't mean you should

dontsovcmc, 2020-05-22:
Of course. It should be simple for beginners and flex for advanced.

closed 2022-01-22 by tablatronix