tzapu/WiFiManager #1062: Popup when selecting a network

FilipDem, 2020-05-24:
**Esp8266/Esp32:**
- [X] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [X] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [X] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

### Description

I easily integrated the library in my sketch. However one cosmetic "problem" is remaining... Connecting to the AP works fine. However when selecting a network from the scanned WiFi list, a "pop-up" is shown as indicated in the screenshot.

A Ubuntu20.04 was used to set the WiFi settings .

[image]

### Settings in IDE

Module:  			Generic ESP8266

Additional libraries: 

Not relevant

### Debug Messages

Not relevant (popup even comes up without debug options).

tablatronix, 2020-05-24:
interesting, which wm version/branch is this ?

FilipDem, 2020-05-24:
I have version 2.0.1-alpha installed (through the library manager of the Arduino IDE).

[image]

tablatronix, 2020-05-24:
Ok let me check and see if I can reproduce, or fix

tablatronix, 2020-05-24:
This is to auto select password when clicking an item 
code is

` onclick='c(this)'`

```
const char HTTP_SCRIPT[]           PROGMEM = "<script>function c(l){"
"document.getElementById('s').value=l.innerText||l.textContent;"
"p = l.nextElementSibling.classList.contains('l');"
"alert(p);"
"document.getElementById('p').disabled = !p;"
"if(p)document.getElementById('p').focus();}</script>";
```

tablatronix, 2020-05-24:
af3c3671a283f89fb501ed0b37669a4c23b4e77f

tablatronix, 2020-05-24:
I will push a new version soon

closed 2020-05-24 by tablatronix

FilipDem, 2020-05-25:
I looked also in the code, but the alert missed my attention.
Thanks! And nice library, easy to use!

tablatronix, 2020-05-25:
I had to re create the release, the tags were messed up, should up date in arduino in everal hours