tzapu/WiFiManager #1065: Won't join stored WiFI until hard reset

alfo, 2020-05-25:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [x] Development

**Esp8266/Esp32:**
- [x] ESP8266
- [x] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01 
- [x] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [x] 2.7.1
- [ ] staging (master/dev)

### Description

After a successful storing of WiFi details, the first time the ESP8266 boots up (from no power, or new sketch upload), it will fail to join the new WiFi with the stored details (and therefore will switch to AP mode).

Pressing the hard reset button to reboot and/or using `ESP.restart()`, the next time the ESP boots it will work fine.

See debug messages at the bottom of the issue:

### Settings in IDE

Module:  			Wemos D1 v2
Module:  			Wemos D1 32

(tried both)

### Sketch

```cpp

// LED will blink when in config mode

#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

//for LED status
#include <Ticker.h>
Ticker ticker;

#ifndef LED_BUILTIN
#define LED_BUILTIN 13 // ESP32 DOES NOT DEFINE LED_BUILTIN
#endif

int LED = LED_BUILTIN;

void tick()
{
  //toggle state
  digitalWrite(LED, !digitalRead(LED));     // set pin to the opposite state
}

//gets called when WiFiManager enters configuration mode
void configModeCallback (WiFiManager *myWiFiManager) {
  Serial.println("Entered config mode");
  Serial.println(WiFi.softAPIP());
  //if you used auto generated SSID, print it
  Serial.println(myWiFiManager->getConfigPortalSSID());
  //entered config mode, make led toggle faster
  ticker.attach(0.2, tick);
}

void setup() {
  WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP
  // put your setup code here, to run once:
  Serial.begin(115200);
  
  //set led pin as output
  pinMode(LED, OUTPUT);
  // start ticker with 0.5 because we start in AP mode and try to connect
  ticker.attach(0.6, tick);

  //WiFiManager
  //Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wm;
  //reset settings - for testing
  // wm.resetSettings();

  //set callback that gets called when connecting to previous WiFi fails, and enters Access Point mode
  wm.setAPCallback(configModeCallback);

  //fetches ssid and pass and tries to connect
  //if it does not connect it starts an access point with the specified name
  //here  "AutoConnectAP"
  //and goes into a blocking loop awaiting configuration
  if (!wm.autoConnect()) {
    Serial.println("failed to connect and hit timeout");
    //reset and try again, or maybe put it to deep sleep
    ESP.restart();
    delay(1000);
  }

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");
  ticker.detach();
  //keep LED on
  digitalWrite(LED, LOW);
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Messages

```
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping 
*WM: [1] Connecting to SAVED AP: Cheese and Biscuits
*WM: [3] Using Password: mypassword
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_CONNECT_FAILED
*WM: [3] lastconxresult: WL_STATION_WRONG_PASSWORD
*WM: [1] AutoConnect: FAILED 
*WM: [2] Starting Config Portal 
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  ESP_FE8F75
*WM: [2] AP has anonymous access! 
*WM: [1] SoftAP Configuration 
*WM: [1] -------------------- 
*WM: [1] ssid:             ESP_FE8F75
*WM: [1] password:         
*WM: [1] ssid_len:         10
*WM: [1] channel:          1
*WM: [1] authmode:        
*WM: [1] ssid_hidden:     
*WM: [1] max_connection:   4
*WM: [1] country:          CN
*WM: [1] beacon_interval:  100(ms)
*WM: [1] -------------------- 
*WM: [1] AP IP address: 192.168.4.1
Entered config mode
192.168.4.1
ESP_FE8F75
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan completed in 2184 ms
*WM: [2] Config Portal Running, blocking, waiting for clients... 
```
Then I press the hard reset button/or call `ESP.restart();`
```
rl l⸮r$⸮n⸮l⸮b|⸮⸮⸮rb⸮b⸮nnlnnbbp⸮$blrlp⸮n⸮l⸮bn⸮n⸮$⸮b⸮⸮nn'l⸮l`⸮nnl`nr⸮⸮⸮n$`r⸮⸮n$`⸮⸮⸮Č⸮l$`⸮⸮nb⸮*WM: [1] AutoConnect 
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping 
*WM: [1] Connecting to SAVED AP: Cheese and Biscuits
*WM: [3] Using Password: mypassword
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 10.0.1.156
connected...yeey :)
*WM: [3] unloading 

```

alfo, 2020-05-25:
I should also mention that if I change the WiFiManager version to 0.1.5, everything works as expected, so it's a problem specific to ~v2.x.x

tablatronix, 2020-05-25:
There has been alot of talk of this and wifi working everyother restart.

I think this is an esp bug, did you do a full erase?

`*WM: [3] lastconxresult: WL_STATION_WRONG_PASSWORD`

tablatronix, 2020-05-25:
Try to full erase memory from esptool or aruino

also enable all debugging in esp

You can try this and see if it changes, some esps hang mid connect, we do not know why

 ```
  // This is sometimes necessary, it is still unknown when and why this is needed but it may solve some race condition or bug in esp SDK/lib
  // wm.setCleanConnect(true); // disconnect before connect, clean connect
 ```

tablatronix, 2020-05-25:
#1067

alfo, 2020-05-25:
Ah, I think clearing the flash completely has fixed it, my bad. Must be because I've switched between 0.15 and 2.0.1 in the same project while developing.

alfo, 2020-05-25:
Actually, I'm an idiot, sorry. This only fixed it once, which it seems is the same as the "every other problem".

tablatronix, 2020-05-25:
Thanks , yeah I have no idea what causes this.

I was never able to reproduce it, there are a few issues on it, I will try to reference them all and see what could be the common cause..

mhama, 2021-01-17:
My workaround is that disabling config portal and calling autoConnect() again until wifi connects. (tested on M5Paper)

On the second or third try, autoConnect() outputs failure log like first attempt (WL_STATION_WRONG_PASSWORD) , but the connection is actually established. I don't know why but it's working now.

To show config portal, I added checking of button press on launch and call startConfigPortal() if the button has pressed.

```cpp
wifiManager.setEnableConfigPortal(false);
```

```cpp
  while(true) {
    setupWiFi(); // call autoConnect()
    if (WiFi.isConnected()) {
      break;
    }
    canvas.drawString("Waiting to retry WiFi connection...", 30, 30);
    canvas.pushCanvas(0,0,UPDATE_MODE_DU4);
    delay(3000);
  }
```

costyn, 2021-02-09:
I have the same thing on my project. Except it's when I've just uploaded the firmware (platformio + VSCode) through USB. If I use OTA the issue doesn't happen. If I hit the reset button on the ESP it comes back straight away.

I have noticed the same behavior before I was using WifiManager, and just the regular ESP Wifi.h. I thought it was a bug in my router perhaps, not allowing a client to join so quickly after leaving. A throttling function perhaps? Which could explain the different behaviors people are seeing?

Patronics, 2021-05-03:
I'm also running into the same issue, the connection is failing with every other restart, and every time the program is reuploaded. Is there any news regarding a potential fix for this issue?

Patronics, 2021-05-03:
Nevermind, the setConnectRetries() function worked around this issue for me, as described here: https://github.com/tzapu/WiFiManager/issues/979#issuecomment-831057723