tzapu/WiFiManager #1066: 2.0.2-alpha does not prefill selected SSID from the list

joysfera, 2020-05-25:
I have removed the rest of entries because this is some kind of a javascript/HTML bug that is not related to my ESP32 Pico D4 hardware, or the Arduino-ESP32 1.0.4 core, or the Arduino 1.8.12 IDE.

### Description

If you pick an AP by "clicking" on it the AP name is not copied over to the SSID input field.

Thus no matter what Password you then enter the subsequent WiFi client login check fails because the AP name is either not set at all or an incorrent (the old/previous) AP name is used.

tablatronix, 2020-05-25:
Is this the alpha dev version ?
What browser?

tablatronix, 2020-05-25:
let me make sure my last commit didnt cause this, I didnt test it

tablatronix, 2020-05-25:
I did shit, it was a one line comment, but there are no lines in the source doh

tablatronix, 2020-05-25:
Sorry about that, pushed fix

closed 2020-05-25 by tablatronix