tzapu/WiFiManager #1067: CONNECTION PROBLEMS, restart, or every other restart

tablatronix, 2020-05-25:
There has been several issues with problems with wifi not connecting. But after restarts it works.
Or every other restafrt it fails

Sometimes there are errors like wrong password, or ap not found

This issue is for investigating this to find some common issue.

I suspect it is a race condition or router issue with certain SDKs etc. It could also be a simple bug like autoconnect is connecting while esp is auto-connecting and it hangs. Or set sta is not returning fast enough and is asynchronous, there has been proof of this and issues by me created in esp libs.

tablatronix, 2020-05-25:
I will look into wm as the cause of this issue, since it is occurring on both platforms

TheNitek, 2020-05-25:
I already linked it in #979 but just to have it here again. I don't think it's bug in WifiManager but in the SDK: https://github.com/espressif/arduino-esp32/issues/2501 (nevertheless one that the WifiManager can work around)

TheNitek, 2020-05-25:
BTW I also suspect the router model to be important for this: In my case it's a Fritz!Box 6490.

alfo, 2020-05-25:
I’ve been having issues with both the ESP8266 and the ESP32 using v2.x.x of WiFiManager but notably the problem goes away on v0.1.5 on my ESP8266, so it might be worth having a look at what changed inside the autoConnect function?

I agree it’s probably true that WiFi routers also play a part in this, in my case I’m using a Ubiquity nanoHD access point (with two APs, one downstairs, broadcasting on the same SSID). Maybe that has something to do with it?

1technophile, 2020-05-26:
I confirm also that [OpenMQTTGateway ](https://docs.openmqttgateay.com) users were complaining about this issue with Ubiquity routers.

tablatronix, 2020-05-26:
One difference is we used to always disconnect, 
```C++
//trying to fix connection in progress hanging
    ETS_UART_INTR_DISABLE();
    wifi_station_disconnect();
    ETS_UART_INTR_ENABLE();
```

But that adds 2 second startup time

This was removed in dev and made optional

` ``C++
    // clean connect, always disconnect before connecting
    void          setCleanConnect(bool enable); // default false
```

tablatronix, 2020-05-26:
A couple things to try that might be causing a race condition.

Try adding this delay and log to this function

```
bool WiFiManager::wifiConnectDefault(){
  bool ret = false;
  DEBUG_WM(F("Connecting to SAVED AP:"),WiFi_SSID(true));
  DEBUG_WM(DEBUG_DEV,F("Using Password:"),WiFi_psk(true));
  ret = WiFi_enableSTA(true,storeSTAmode);

 // add these 2 lines
  delay(500); // <-- DELAY
  DEBUG_WM(DEBUG_DEV,"Mode after delay: "+getModeString(WiFi.getMode())); <-- LOG

  if(!ret) DEBUG_WM(DEBUG_ERROR,"[ERROR] wifi enableSta failed");
  ret = WiFi.begin();
  if(!ret) DEBUG_WM(DEBUG_ERROR,"[ERROR] wifi begin failed");
  return ret;
}
```

Another thing to try 

Remove the sta, begin will do this by itself also

```C++
bool WiFiManager::wifiConnectDefault(){
  bool ret = false;
  DEBUG_WM(F("Connecting to SAVED AP:"),WiFi_SSID(true));
  DEBUG_WM(DEBUG_DEV,F("Using Password:"),WiFi_psk(true));
  // ret = WiFi_enableSTA(true,storeSTAmode); // <-- REMOVE
  // if(!ret) DEBUG_WM(DEBUG_ERROR,"[ERROR] wifi enableSta failed");
  ret = WiFi.begin();
  if(!ret) DEBUG_WM(DEBUG_ERROR,"[ERROR] wifi begin failed");
  return ret;

thewavelength, 2020-08-18:
I have the same issue. Sometimes, totally not reproducible so please don't ask, I even get a crash which seems to be related to this. This happens now as long as I erase the flash and flash the image again.

```
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  10.0.1.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan completed in 2308 ms
*WM: [2] Config Portal Running, blocking, waiting for clients... 
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x00000000  PS      : 0x00060e30  A0      : 0x80157969  A1      : 0x3ffb4f00  
A2      : 0x3ffda6a0  A3      : 0x3ffdae1c  A4      : 0x3ffda134  A5      : 0x3ffda8d4  
A6      : 0x260d000a  A7      : 0x300d000a  A8      : 0x8015780c  A9      : 0x3ffb4ec0  
A10     : 0x3ffda6b0  A11     : 0x3ffdae1c  A12     : 0x3ffb4f0c  A13     : 0x00000044  
A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffb4f00 0x40157966:0x3ffb4f40 0x40164e71:0x3ffb4f60 0x40169e79:0x3ffb4fa0 0x4016f116:0x3ffb4fc0 0x4015836f:0x3ffb4fe0 0x40088b9d:0x3ffb5010
```

tablatronix, 2022-01-11:
I have been looking into this issue here
https://github.com/espressif/arduino-esp32/issues/2501

And made a PR here for workaround via autoreconnect bugfix
https://github.com/espressif/arduino-esp32/pull/6113

I also have an open issue with IDF against the actual cause under review

tablatronix, 2022-01-24:
MY PR was merged, but if you need a solution without a patch

I added a test solution to wm, hardcoded, no setter. I can add a define or setter if its useful

```
    bool          _aggresiveReconn        = false; // use an agrressive reconnect strategy, WILL delay conxs
                                                   // on some conn failure modes will add delays and many retries to work around esp and ap bugs, ie, anti de-auth protections
```

This adds esp32 event handler for auth fail, and bumps reconnects up and adds delays, I have connect everytime, and my conx time goes to 5s unfortunately until this issue is also fixed in IDF that is what we have..

tablatronix, 2022-06-17:
Theres seems to be 2 conditions here with different failures, this one for reauth timeout there is also a 4 way handshake timeout or assoc failure which is possibly a different issue

dddhhhrrr, 2022-06-20:
is there a workaround for this one? I tried the _aggressiveReconn = true fix and it connects but gives me an IP = 0.0.0.0

tablatronix, 2022-06-21:
Hmm it shouldnt do you have logs?

tablatronix, 2022-06-21:
hmm, that is odd, I may have seen this reported elsewhere, or as a different bug...
can you turn on esp debug level?

dddhhhrrr, 2022-06-22:
I updated the ESP core version, it seems that it fixed it...

tablatronix, 2022-06-22:
Yeah I couldn't fine the issue, but i specifically recall seeing a IDF bug with failing to set ip

Robotto, 2022-10-14:
> I updated the ESP core version, it seems that it fixed it...

I'm still experiencing this issue with ESP core version 3.0.2 (via arduino board manager) on an ESP-12E
wm version 2.0.13-beta (via arduino library manager)

Is there a different combination of core and library versions where this works?

tablatronix, 2022-10-14:
Can you provide logs with esp debugging on ?

The esp core should be retrying these connections itself now.

tablatronix, 2022-10-22:
Ok this is a slightly different issue, have you tried the aggressive recon mod above?

kjyv, 2022-10-22:
It's already included by default, but also tried setting it to fault

tablatronix, 2022-10-23:
oh yeah lol

Robotto, 2022-10-23:
> Can you provide logs with esp debugging on ?
> 
> The esp core should be retrying these connections itself now.

Hm... While messing around trying to get debug output to work, i discovered that the ESP module is an ESP-12 and not an ESP-12E .. changed this and the issue seems to have vanished...  (never got debug output to work)

lyricnz, 2024-11-24:
(Crossposting from closed issue)

FWIW I have the same symptom (connects every second boot, with 100% reliability).  Adding the retries didn't help for me. 

```
Connecting to WiFi
*wm:[2] AutoConnect 
*wm:[4] No Hostname to set 
*wm:[3] ESP32 event handler enabled 
*wm:[3] Connecting as wifi client... 
*wm:[4] STA static IP:
*wm:[3] setSTAConfig static ip not set, skipping 
*wm:[2] Connect Wifi, ATTEMPT # 1 of 3
*wm:[2] Connecting to SAVED AP: iot
*wm:[4] Using Password: REMOVED
*wm:[4] WiFi_enableSTA enable
*wm:[4] Mode after delay:  STA
*wm:[2] connectTimeout not set, ESP waitForConnectResult... 
*wm:[3] [EVENT] WIFI_REASON:  202
*wm:[3] [EVENT] WIFI_REASON:  202
*wm:[3] Connection result: WL_CONNECT_FAILED
*wm:[2] Connect Wifi, ATTEMPT # 2 of 3
*wm:[2] Connecting to SAVED AP: iot
*wm:[4] Using Password: REMOVED
*wm:[4] WiFi_enableSTA enable
*wm:[4] Mode after delay:  STA
*wm:[2] connectTimeout not set, ESP waitForConnectResult... 
*wm:[3] Connection result: WL_CONNECT_FAILED
*wm:[2] Connect Wifi, ATTEMPT # 3 of 3
*wm:[2] Connecting to SAVED AP: iot
*wm:[4] Using Password: REMOVED
*wm:[4] WiFi_enableSTA enable
*wm:[4] Mode after delay:  STA
E (8862) wifi:sta is connecting, return error
[  4270][E][WiFiSTA.cpp:317] begin(): connect failed! 0x3007
*wm:[2] connectTimeout not set, ESP waitForConnectResult... 
*wm:[3] Connection result: WL_CONNECT_FAILED
*wm:[4] lastconxresulttmp: WL_STATION_WRONG_PASSWORD
*wm:[4] lastconxresult: WL_STATION_WRONG_PASSWORD
*wm:[2] AutoConnect: FAILED for  3654 ms
*wm:[3] Starting Config Portal 
*wm:[4] WiFi station disconnect 
*wm:[4] WiFi_enableSTA enable
*wm:[3] Disabling STA 
*wm:[3] Enabling AP 
*wm:[2] StartAP with SSID:  Info-Orbs_88
```