tzapu/WiFiManager #1068: hostname example

sanchosk, 2020-05-27:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [x] Master
- [x] Development

**Esp8266/Esp32:**
- [x] ESP8266
- [x] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [x] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [x Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [x] 2.4.0
- [ ] staging (master/dev)

### Description

Missing example for setting hostname.
I've found within your issue list a suggestion where you refer to wm.hostname(), but this seem not to work.
I'd like to have an example of portal with wifi ssid, password and hostname.
Is this feasible?

Btw, VERY nice wifi manager - I only found it recently, but I'll use it everywhere from now on.

### Settings in IDE

Module:  			ESP32-Cam

Additional libraries: ESP32 camera

alfo, 2020-05-28:
You can do this with WiFi.hostname() on ESP8266 and WiFi.setHostname() on ESP32

sanchosk, 2020-05-28:
I know.
What I wanted to achieve was to use the parameters to also set the hostname.
However, my skills are insufficient, so I failed.
And when I use WiFi.setHostname() after wm, the whole sketch locks...

alfo, 2020-05-28:
I have this working. You load your parameters from JSON, then set WiFi.hostname(), then instantiate wm

sanchosk, 2020-05-28:
Can you, perhaps, share the code?

alfo, 2020-05-28:
https://github.com/alfo/IoTBaseFile/

commit b90af1ac (2022-08-18): fixes #1068 esp32 S2

Esp32 has seeveral variations on when you can sethostname, might vary with versions so will try to add a best practice for all
closed 2022-08-18 by tablatronix