tzapu/WiFiManager #1069: IS there a way to either keep AP+STA for timeout duration, or for AP at each reboot?

Hoel, 2020-05-27:
----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [x] Master
- [ ] Development

**Esp8266/Esp32:**
- [x] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [x] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [x] 2.4.0
- [ ] staging (master/dev)

### Description

Problem description
I would like to make the portal always accessible at reset for the duration of the timeout, if the credential have already been filled previously it is fine that the network is still connected immediately (AP + STA) but even in this case i would like that the AP portal stays accessible for the duration of the timeut. The goal is to allow to change credentials at any time after a reset (or enter them for the first time).

What is the best way to do this? For some reason i did not found a way to do it in the library, i dont mind modifying the source but its not clear to me how it is best done.

### Settings in IDE

Module: esp8285 generic module

tablatronix, 2020-05-28:
You can use the ondemand examples to do this. Just remember that AP+STA is not very stable.

You can also just make the portal available on your lan with no ap..

Hoel, 2020-05-28:
ah thank you, i also modified slightly the source to keep AP+STA during the timeout, even if the STA is connected directly, which is what i needed.

closed 2020-05-28 by Hoel