tzapu/WiFiManager #1070: Password showing in Serial Monitor

jg1xmv, 2020-06-01:
Hi,

Hopefully I am in the right place to ask this question as it is not really an issue but a question about configuration.

My Wifi password is shown in the Serial Monitor when connecting, is it possible to not show it or have it show ******* instead?

Thanks.

*WM: [1] AutoConnect 
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping 
*WM: [1] Connecting to SAVED AP: SSID HERE
*WM: [3] Using Password: **password**
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS

tablatronix, 2020-06-01:
It will not show up if you pick a lower debug level

jg1xmv, 2020-06-02:
Thanks for the suggestion, my debug level is at "None" and it still shows-up.

tablatronix, 2020-06-02:
How are you setting wm debug level?

jg1xmv, 2020-06-02:
I left it as is by default, I did not change anything.

jg1xmv, 2020-06-02:
Ok, I understand what you mean now.
I set it using wifiManager.setDebugOutput(false);
The password does not show anymore.
I'll study the various debug levels a bit more.
Thanks for your help!

tablatronix, 2020-06-02:
it is hardcoded atm in the .h file you can change it using a compiler flag also

jg1xmv, 2020-06-02:
Understood, thank you very much.