tzapu/WiFiManager #1071: IS there a way to disable automatic redirection to portal after AP connect?

Hoel, 2020-06-01:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [x] Master
- [ ] Development

**Esp8266/Esp32:**
- [x] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [x] Other ESP8285

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [x] 2.4.0
- [ ] staging (master/dev)

### Description

Problem description

I need to pass an UUID as parameter of the portal URL, and the only way i found to do it is to use two QR code, the first QR code sets the mobile to the correct AP, and the second QR code loads the portal with the UUID as URL parameter.
For this to work i need to prevent the mobile from loading automatically the PA portal after conncetion (192.168.4.1 in this case), is there a way to do this in wifimanager?

tablatronix, 2020-06-01:
In the alpha/dev version you can disable the captive portal

Hoel, 2020-06-01:
ok nice, i will give a try. Just to clarify : i dont want to totally disable the captive portal, i want to disable the fact that it is called automatically after the AP is connected, so i can call it manually in a second time (via QR code)

tablatronix, 2020-06-01:
ConfigPortal is what runs.

CaptivePortal is what auto redirects on wifi connection

Hoel, 2020-06-01:
ahh it makes sense, thank you for the insights.

closed 2020-06-01 by Hoel