tzapu/WiFiManager #1072: do not disable internal ap when device connected to external ap

alirezaimi, 2020-06-05:
Hi
Is there any option exist in this library that after my board connected to external access point, do not disable internal ap of board ? 
I'm asking this because i want both ap and sta mode active simultaneously in my esp8266 board for when external ap or internet connection is dead, i can connect to board with internal ap of board .

alirezaimi, 2022-04-22:
Hi
in esp8266 board the ability of using both AP and sta in same time is exist and I want to use it, but in this library after connecting to AP, the internal sta disabled by default . I want to choose it optionally , how ???

tablatronix, 2022-04-22:
just turn it on like usual WiFi.softAp()

alirezaimi, 2022-04-25:
Hi @tablatronix 
When should i do this ? 
I mean in  setup I got these lines :

    // -- Initializing the configuration.
    iotWebConf.init();

    // -- Set up required URL handlers on the web server.
    server.on("/", handleRoot);
    server.on("/config", []{ iotWebConf.handleConfig(); });
    server.onNotFound([](){ iotWebConf.handleNotFound(); });

and everything is handled by library itself, when i turn on softap again after the library disable it ?

tablatronix, 2022-04-25:
hmm you mentioned ap mode not a webserver..
you can use the wm.server()

someone else just asked this in another issue, let me check