tzapu/WiFiManager #1074: Custom parameters are not stored when using startConfigPortal() (works first time on autoconnect())

pnodseth, 2020-06-07:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

## POST SERIAL OUTPUT !

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos
Version: 2.0.2 Alpha

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [X ] Development

**Esp8266/Esp32:**
- [X] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [X] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

### Description
First of all, thanks for all the great work you are doing with this library! So useful! :D 
Custom parameters are not stored when using startConfigPortal() and connecting to same wifi as previously connected to. The values that was stored on first connection with th

Problem description
I am running into a problem when using startConfigPortal(). 
When I connect to a new wifi using autoconnect during setup, the custom parameters I have are stored just like they should.

But after that,  if I trigger startConfigPortal() and connect to the same wifi but insert other custom parameters, they are not stored. The initially saved parameters are still there.

### Settings in IDE

Module:  			NodeMcu

Additional libraries: 

### Sketch

```cpp

#include <FS.h> //this needs to be first, or it all crashes and burns...

#include <ESP8266WiFi.h> //https://github.com/esp8266/Arduino

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h> //https://github.com/tzapu/WiFiManager
#include <ArduinoJson.h> //https://github.com/bblanchon/ArduinoJson

char deviceName[40];
char userSecret[40];

#define WIFI_TRIGGER_PIN 0

WiFiManagerParameter custom_deviceName("deviceName", "Device Name", deviceName, 40);
WiFiManagerParameter custom_userSecret("userSecret", "User Secret", userSecret, 40);
WiFiManager wifiManager;
//flag for saving data
bool shouldSaveConfig = false;

//callback notifying us of the need to save config
void saveConfigCallback()
{
  Serial.println("Should save config");
  shouldSaveConfig = true;
}

/* Wifimanager end */

void setup()
{
  Serial.println("Starting setup...");

  wifiManagersetup();

}

void loop()
{

  /* IF we want device to go to wifi setup portal, press onboard reset button */
  if (digitalRead(WIFI_TRIGGER_PIN) == LOW)
  {
    Serial.println("starting config portal...");
    delay(2000);
    wifiManager.resetSettings();
    delay(2000);
    wifiManager.startConfigPortal("nodeMCUWifi");
  }
  else
  {
    //...stuff here
  }
}

void wifiManagersetup()
{

  //read configuration from FS json
  Serial.println("mounting FS...");
  SPIFFS.begin();
  if (SPIFFS.begin())
  {
    Serial.println("mounted file system");
    if (SPIFFS.exists("/config.json"))
    {
      //file exists, reading and loading
      Serial.println("reading config file");
      File configFile = SPIFFS.open("/config.json", "r");

      if (configFile)
      {
        Serial.println("opened config file");
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);

        configFile.readBytes(buf.get(), size);
        StaticJsonDocument<256> json;
        DeserializationError jsonError = deserializeJson(json, buf.get());
        serializeJsonPretty(json, Serial);
        if (!jsonError)
        {
          Serial.println("\nparsed json");

          strcpy(deviceName, json["deviceName"]);
          strcpy(userSecret, json["userSecret"]);

          Serial.print("nu kjøm han");
          Serial.println(deviceName);
        }
        else
        {
          Serial.println("failed to load json config");
        }
        configFile.close();
      }
    }
  }
  else
  {
    Serial.println("failed to mount FS");
  }
  //end read

  //set config save notify callback
  wifiManager.setSaveConfigCallback(saveConfigCallback);
  wifiManager.setSaveParamsCallback(saveConfigCallback);

  //add all your parameters here
  wifiManager.addParameter(&custom_deviceName);
  wifiManager.addParameter(&custom_userSecret);

  if (!wifiManager.autoConnect("nodeMCUWifi"))
  {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(5000);
  }

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");

  //save the custom parameters to FS
  if (shouldSaveConfig)
  {
    //read updated parameters
    strcpy(deviceName, custom_deviceName.getValue());
    strcpy(userSecret, custom_userSecret.getValue());
    Serial.println("saving config");
    StaticJsonDocument<256> json;
    json["userSecret"] = userSecret;
    json["deviceName"] = deviceName;

    File configFile = SPIFFS.open("/config.json", "w");
    if (!configFile)
    {
      Serial.println("failed to open config file for writing");
    }

    serializeJsonPretty(json, Serial);
    serializeJson(json, configFile);
    configFile.close();
    //end save
  }
}

```

### Debug Messages

```
messages here
```

tablatronix, 2020-06-07:
Is your callback being claled, you didnt post any serial logs...

saveConfigCallback

pnodseth, 2020-06-07:
Sorry about that!
Here is the log:

*WM: [2] <- HTTP WiFi save  
*WM: [3] Method: POST
*WM: [2] Parameters 
*WM: [2] -------------------- 
*WM: [2] deviceName: TestingDevice
*WM: [2] userSecret: TestingSecret
*WM: [2] -------------------- 
Should save config
*WM: [3] Sent wifi save page 
*WM: [2] processing save 
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping 
*WM: [1] CONNECTED:
*WM: [1] Connecting to NEW AP: <REMOVED>
*WM: [3] Using Password: <REMOVED>
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] Connect to new AP [SUCCESS] 
*WM: [1] Got IP Address: 
*WM: [1] 192.168.1.183 
Should save config
*WM: [2] disconnect configportal 
*WM: [2] restoring usermode STA
*WM: [2] wifi status: WL_CONNECTED
*WM: [2] wifi mode: STA
*WM: [1] config portal exiting

tablatronix, 2020-06-07:
Well you are not checking for shouldsave in your loop, only in setup, so thats probably why..

pnodseth, 2020-06-08:
Omg, you are correct! Thanks, and sorry :D 
Again, really appreciate your effort with this library!!

closed 2020-06-08 by pnodseth