tzapu/WiFiManager #1076: Changing ssid & password programatically for use by autoConnect()

jg1xmv, 2020-06-08:
When originally connecting I entered a ssid and password via the AP which is then stored in memory by the library for the next start.

How can I change to some new ssid & pw in memory before triggering autoconnect so that it connects to those and not to the originally saved ssid & pw when I first entered those via the AP mode?

I have tried using WiFi.begin(ssidnet,pwnet), where ssidnet & pwnet are the new ones I saved.

It works when using a fixed ip by doing the following:
wifiManager.setSTAStaticIPConfig(fixed ip details here);
WiFi.mode(WIFI_STA);
WiFi.begin(ssidnet, pwnet);

Somehow though, using WiFi.begin(ssidnet, pwnet) before autoconnect() does not seem to make it go and fetch the new ssid & pw I saved but the original ones to which I connected via the AP first time around.

I am wondering of there is a way to make autoConnect() fetch the new ssidnet & pwnet i/o the original ones.

Thanks.

tablatronix, 2020-06-08:
WiFi.begin(ssidnet, pwnet); should work, we do not save those that is all esp code.
Maybe add a connect loop

jg1xmv, 2020-06-09:
Yes, in a way once the ssid and pw are saved I do not need the AP anymore and can just call WiFi.begin instead of autoConnect.

I'll add a check to verify if credentials are already saved and then either use autoConnect to use the AP or WiFi.begin with the previously saved credentials otherwise.

Thanks.

closed 2020-06-09 by jg1xmv

tablatronix, 2020-06-09:
Well either way, that is all autoconnect does. anyway , it just calls begin, the esp code saves it