tzapu/WiFiManager #1077: AP mode still active after setup as client.

VladoPortos, 2020-06-09:
### Basic Infos
ESP-01 ( 8266 ) Module still create AP after setup and successful log in as client. I think that should not be happening. 

#### Hardware
**WiFimanager Branch/Release:**
- [ X] Master
- [ ] Development

**Esp8266/Esp32:**
- [ X] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [X ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

### Description

ESP-01 ( 8266 ) Module still create AP after setup and successful log in as client. I think that should not be happening.

### Settings in IDE

Module:  			Generic ESP8266 Module

Additional libraries: 

include <FS.h>   // Include the SPIFFS library
include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino
include <DNSServer.h>
include <ESP8266WebServer.h>
include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager
include <DoubleResetDetect.h>  //https://github.com/jenscski/DoubleResetDetect
include <OneWire.h>
include <ArduinoJson.h>          // https://github.com/bblanchon/ArduinoJson

### Sketch

```cpp
#include <FS.h>   // Include the SPIFFS library
#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager
#include <DoubleResetDetect.h>  //https://github.com/jenscski/DoubleResetDetect
#include <OneWire.h>
#include <ArduinoJson.h>          // https://github.com/bblanchon/ArduinoJson

OneWire  ds(2);  // on pin 2 
ESP8266WebServer server(80);

char hostName[40]="TempMonitor";

// Number of seconds after reset during which a
// subseqent reset will be considered a double reset.
#define DRD_TIMEOUT 10

// RTC Memory Address for the DoubleResetDetector to use
#define DRD_ADDRESS 0x00

DoubleResetDetect drd(DRD_TIMEOUT, DRD_ADDRESS);

//flag for saving data
bool shouldSaveConfig = false;

[... 251 lines cut ...]
11:31:12.317 -> *WM: [3] allocating params bytes: 20
11:31:12.317 -> *WM: [2] Added Parameter: hostName
11:31:12.317 -> *WM: [1] AutoConnect 
11:31:12.317 -> *WM: [1] AutoConnect: ESP Already Connected 
11:31:12.317 -> *WM: [3] STA static IP:
11:31:12.317 -> *WM: [2] setSTAConfig static ip not set, skipping 
11:31:12.317 -> *WM: [1] AutoConnect: SUCCESS 
11:31:12.350 -> *WM: [1] STA IP Address: 10.0.0.166
11:31:12.350 -> *WM: [1] AutoConnect 
11:31:12.350 -> *WM: [1] AutoConnect: ESP Already Connected 
11:31:12.350 -> *WM: [3] STA static IP:
11:31:12.350 -> *WM: [2] setSTAConfig static ip not set, skipping 
11:31:12.350 -> *WM: [1] AutoConnect: SUCCESS 
11:31:12.350 -> *WM: [1] STA IP Address: 10.0.0.166
11:31:12.350 -> connected...yeey :)
11:31:12.350 -> Server started
11:31:12.350 -> 10.0.0.166
11:31:12.350 -> *WM: [3] freeing allocated params! 
11:31:12.350 -> *WM: [3] unloading 
11:31:12.520 -> No more addresses.
11:31:12.520 -> 
11:31:54.197 -> No more addresses.
11:31:54.197 -> 

```

tablatronix, 2020-06-09:
WiFi.mode(WIFI_STA); in your sketch
the esp auto starts ap if its saved

VladoPortos, 2020-06-10:
> 
> 
> `WiFi.mode(WIFI_STA);` in your sketch
> the esp auto starts ap if its saved

Thanks @tablatronix seems like this is the answer, but where in setup exactly to put it ?

alirezaimi, 2020-06-16:
hi, I can't understand what the answer is ?!
I have this problem with esp-8266 12e/f , is this is an option ?

tablatronix, 2020-06-16:
Yes , the esp by default start an ap with the esp id name, you need to turn off WIFI_AP mode

cacs200281, 2020-06-22:
I had the same issue and just solved, in my case, following that instructions from @tablatronix .
Thanks a lot!

closed 2020-06-22 by VladoPortos