tzapu/WiFiManager #1081: wifimanager.setAPcallback

SudhaMaanasa, 2020-06-20:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [ ] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

### Description

Problem description

### Settings in IDE

"wifimanager.setAPcallback is used if we need to do something when device enters configuration mode on failed WiFi connection attempt. Before autoConnect()."
But when I use it in my code under a failed connection of WiFi I couldn't find that the device entering into configuration mode. Is there something I'm missing in achieving this? Someone please help me with this.

tablatronix, 2020-06-21:
No idea are you using an example?

SudhaMaanasa, 2020-06-30:
Yes

tablatronix, 2020-06-30:
Which one, ill check and see if there is something wrong