tzapu/WiFiManager #1082: Exception thrown when trying to start edit settings in onDemand portal

AceoStar, 2020-06-20:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [X] Development

**Esp8266/Esp32:**
- [X] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [X] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [X] 2.4.0
- [ ] staging (master/dev)

### Description

My goal is to launch WifiManager on setup. At a later date, I'd like users to press a button that re-initiates the process, they'll connect to the defined wifi AP and once they hit save, the device will reset. I don't need this process to be non blocking (I think). Eventually, I will save the settings and restart the device once they are done. Basically, when I hit the button, I'd like wifimanager to act the same way it does if you boot the device with no stored wifi credentials.

This seems to work, but when I open Wifi settings, I get the error shown below.

I apologize,  unfortunately, due to the complexity of my sketch, the code below is not 100% matched with what I have on my end. I think the problem may be the way I'm declaring WifiManger? 

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```
#include <Arduino.h>

#include <WiFiManager.h>       //WiFi Manager to allow users to configure device via mobile   //https://github.com/tzapu/WiFiManager

bool buttonPressed = false;

const int buttonPin = D3;     // the number of the pushbutton pin
int buttonState = 0;         // variable for reading the pushbutton status

WiFiManager wifiManager;

unsigned long now = millis();
unsigned long lastTrigger = 0;

ICACHE_RAM_ATTR void buttonCallback(){
  buttonPressed = true;
}

void saveConfigCallback () {
  Serial.println("Should save config");
  
}
void setup() {
    Serial.begin(115200);
    pinMode(buttonPin, INPUT);

  // Serial.setDebugOutput(true);
	attachInterrupt(digitalPinToInterrupt(buttonPin), buttonCallback, CHANGE);

  WiFiManagerParameter key("apiKey", "Key", "token", 40);
  wifiManager.addParameter(&key);

  WiFiManagerParameter service_ids("service_ids", "ID", "ID", 100);
  wifiManager.addParameter(&service_ids);

  wifiManager.setSaveConfigCallback(saveConfigCallback);

  wifiManager.autoConnect("IOT");
  wifiManager.setMinimumSignalQuality(10); // Low signal quality could cause false negatives
  

  Serial.println("Setup complete");

}

void loop() {
  now = millis();
  wifiManager.process();
 // buttonState = digitalRead(buttonPin);
  if (buttonPressed == true) {
    Serial.println("Button pressed");
    

    //wifiManager.setConfigPortalBlocking(false);
    wifiManager.startConfigPortal("IOT");
    //wifiManager.startWebPortal();

    buttonPressed = false;

    delay(50);
  }
}
```

### Debug Messages

```
Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4020f483 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffba0 end: 3fffffc0 offset: 0190
3ffffd30:  402799ca 3ffffd50 4020f483 40211813
3ffffd40:  00000000 4020f483 3ffffda4 40207641
3ffffd50:  3fff3a00 002c002f 80fffda4 007d747b  
3ffffd60:  00000000 8327b219 3ffffda4 40211b7c
3ffffd70:  00000001 00000000 3ffefb6c 40208a41  
3ffffd80:  007d6300 00000e73 80fffe34 402117a7
3ffffd90:  00000e73 3ffffde0 3ffffe28 4021249b  
3ffffda0:  ffffffff 3fff3994 0092009f 0020755c
3ffffdb0:  3fff35b4 005c005f 00fefb6c 40208797  
3ffffdc0:  00000e7c 402799de 3ffffe28 00000e7c
3ffffdd0:  ffffffff 0000000c 00000037 ffffffff
3ffffde0:  00000011 00000047 00000053 00000058  
3ffffdf0:  3ffffe10 8027b2c1 3ffefb6c 40209a64
3ffffe00:  00000001 402799ce 3ffefb6c 4020dcfa  
3ffffe10:  3fff3700 0014001f 80005053 00000005
3ffffe20:  3fff2ad8 3ffffe70 3fff388c 00fd00ff  
3ffffe30:  00fffe90 3fff406c 0e7c0e7f 00209a64
3ffffe40:  00000001 00000001 3fff2a7c 40216410  
3ffffe50:  3ffffe90 3fff288c 3fff2a7c 401000e1
3ffffe60:  3fff2a7c 3fff288c 3fff2a7c 40209a9c
3ffffe70:  66697700 80000069 80000000 fffffffe  
3ffffe80:  3fff2a7c 3fff288c 3fff284c 4020eeca  
3ffffe90:  6669772f 00000069 85fefec8 3fff3c00
3ffffea0:  000000af 80003f69 40100400 00017dcf
3ffffeb0:  3fff288c 00000000 00000001 00000001  
3ffffec0:  00000001 3fff2a7c 79999999 00000000
3ffffed0:  00000000 00000000 4bc6a7f0 3ffefec8  
3ffffee0:  00000001 3fff2870 3fff284c 4020efd7
3ffffef0:  00000000 3fff22f4 00000000 402156c0  
3fffff00:  01b9790c 3ffefc6c 3ffefb6c 40215b5c
3fffff10:  00000000 3ffefc6c 3ffefb6c 4020f1d5  
3fffff20:  00000000 3ffefc6c 3ffefb6c 402097be
3fffff30:  00000000 00000000 00000001 401003f8
3fffff40:  000e000f 80737574 3ffefb6c 3ffefec8  
3fffff50:  00000000 3ffefc6c 3ffefb6c 3ffefec8
3fffff60:  00000000 3ffefc6c 3ffefb6c 4020f483  
3fffff70:  3ffefb68 0000000e 3ffefd88 4021107c
3fffff80:  3ffe88f9 00000000 3ffefd88 402110a0  
3fffff90:  3ffefb68 3ffefb6c 3ffefd18 40202fe0  
3fffffa0:  3fffdad0 00000000 3ffefe88 40212ac0
3fffffb0:  feefeffe feefeffe 3ffe8610 4010109d
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3456, room 16
tail 0
chksum 0x84
csum 0x84
va5432625
~ld

```

tablatronix, 2020-06-21:
Make sure wifimanager is in the same scope either global or redeclare it in your loop if not

tablatronix, 2020-06-21:
Also there is an example for this