tzapu/WiFiManager #1084: Issue with

ArsineMan, 2020-06-23:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

## POST SERIAL OUTPUT !

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ x] Master
- [ ] Development

**Esp8266/Esp32:**
- [ x] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [x ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [x] 2.4.0
- [ ] staging (master/dev)

### Description

Problem description
when autoConnect fails to connect to router (because I put a phony SSID/Password to force the connection to fail) I am immediately directed to AP/Configuration mode. How does the code below reach the Serial.println("Failed to connect and hit timout"); line?  I am hoping that if my connection isn't established I will enter sleep mode and try again later.  My only other thought is to use the configModeCallback to enter sleep mode and try again?  

```
  response = wifiManager.autoConnect("WifiMgr","Password");
  if(!response) {
    Serial.println("failed to connect and hit timeout");
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(1000);
  } ## Settings in IDE
```
Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```
#include <ESP8266WiFi.h>          

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include "WiFiManager.h"          //https://github.com/tzapu/WiFiManager

void configModeCallback (WiFiManager *myWiFiManager) {
  Serial.println("Entered config mode");
  Serial.println(WiFi.softAPIP());
  //if you used auto generated SSID, print it
  Serial.println(myWiFiManager->getConfigPortalSSID());
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);

  WiFiManager wifiManager;

//set callback that gets called when connecting to previous WiFi fails, and enters Access Point mode
  wifiManager.setAPCallback(configModeCallback);
//wifiManager.setConnectTimeout(10);

  bool response;
  response = wifiManager.autoConnect("WifiMgr","Password");
  if(!response) {
    Serial.println("failed to connect and hit timeout");
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(1000);
  } 

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");
 
}

void loop() {

}
```

### Debug Messages

```
*WM: Status:
*WM: 6
*WM: Using last saved values, should be faster
*WM: Connection result:
*WM: 1
Entered config mode
192.168.4.1
WifiMgr
*WM:
*WM: Configuring access point...
*WM: WifiMgr
*WM: Password
*WM: AP IP address:
*WM: 192.168.4.1
*WM: HTTP server started

```

tablatronix, 2020-06-23:
If you do not want to start the configportal why are you even using wifimanager?

ArsineMan, 2020-06-23:
I definitely want to start the ConfigPortal when SSID/Password have yet to be established, I just don't want my battery powered project to remain in configmode (and turned on) if the wifi connection cannot be established.  Ie. if my home accesspoint is down my device will remain on and in ConfigPortal mode.

tablatronix, 2020-06-24:
if it is battery powered, then only use WM when there is no previous saved ssid or use a button for ondemand mode.

tablatronix, 2020-06-24:
It might be a pain to do this in v0, but v2 you can use

```C++
    // check if the module has a saved ap to connect to
    bool          getWiFiIsSaved();
```

ArsineMan, 2020-06-24:
Thanks for the response. Yes I went the route of managing the wifi client mode separate from WM and utilized the startConfigPortal method to utilize your portal and manage ssid's.  I didn't realize the ESP8266wifi class stored previous connection SSID/psk so I was looking all over the damn place for where you were storing it.  Makes my life easier after rtfm.  I appreciate your time on this library it will save me hours.

closed 2020-06-24 by ArsineMan

tablatronix, 2020-06-24:
Yeah you might be able to just check WiFi.SSID() == "" or null, but not sure how reliable that is, in the alpha version I actually check the wifi config from memory.

tablatronix, 2020-06-24:
https://github.com/tzapu/WiFiManager/blob/development/WiFiManager.cpp#L2873-L2902

ArsineMan, 2020-06-24:
Looks good. I assume i'm ok with WiFi.SSID given their code:
```cpp

String ESP8266WiFiSTAClass::SSID() const {
    struct station_config conf;
    wifi_station_get_config(&conf);
    char tmp[33]; //ssid can be up to 32chars, => plus null term
    memcpy(tmp, conf.ssid, sizeof(conf.ssid));
    tmp[32] = 0; //nullterm in case of 32 char ssid
    return String(reinterpret_cast<char*>(tmp));
}

/**
 * Return the current pre shared key associated with the network
 * @return  psk string
 */
String ESP8266WiFiSTAClass::psk() const {
    struct station_config conf;
    wifi_station_get_config(&conf);
    char tmp[65]; //psk is 64 bytes hex => plus null term
    memcpy(tmp, conf.password, sizeof(conf.password));
    tmp[64] = 0; //null term in case of 64 byte psk
    return String(reinterpret_cast<char*>(tmp));
}
```

tablatronix, 2020-06-24:
Oh yeah I forgot I only did that so I could pick persistent or not. So that should work fine.

I need to add an example for this.