tzapu/WiFiManager #1085: STA and AP mod at the same time unwanted

Primus007, 2020-06-23:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [x] Development

**Esp8266/Esp32:**
- [x] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [x] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [x] 2.7.1
- [ ] staging (master/dev)

### Description

At the first run i connect to the AP and configure my wifi. all works fine. next time i start the wifi module i have STA and AP . 
Is there any way to have only STA ? I don't want the AP.

My Scetch:
`  if (!wifiManager.autoConnect("AutoConnectAP")) {
      Serial.println("failed to connect and hit timeout");
      delay(3000);
      //reset and try again, or maybe put it to deep sleep
      ESP.restart();
      delay(5000);
  }`

kc2uez, 2020-07-02:
I am also experiencing this same issue with the alpha lib version 2.0.1, 2.0.2, and 2.0.3. After the device is configured to connect to the router, it connects, but also keeps the AP mode enable WiFi.mode(STA_AP) somehow is enabled instead of just WiFi.mode(STA).

The problem is that the ESP AP is running on the same channel as the router causing wireless network degradation for other devices connected to the router.

I am using the Arduino IDE 1.8.13, esp8266 2.7.1 and the hardware is an ESP8266 Wemos D1 Mini. To get it to work without this "feature" I reverted to 0.15.

tablatronix, 2020-07-02:
Just turn it off, its the default in esp

kc2uez, 2020-07-02:
Ok, but this is something that changed between versions and is not what the documentation says. I imagine a lot of people will not know that now it will go into AP_STA mode by default causing network performance degradation. I just happen to notice because the wifi coverage was bad after switching to the new library version.

tablatronix, 2020-07-02:
hmm, not sure what to do about it, that is how the esp works. you are supposed to set your own user mode in your code, wm no longer disturbs your persistent wifi modes, it was a bug.

Primus007, 2020-07-03:
Yes, please update the examples. this is helpful also.

tobiasfaust, 2020-08-14:
Year, this solution has been costs me > 2h investigation.
Solution is like this:

````
  espClient = WiFiClient();
  WiFi.mode(WIFI_STA);
  WiFiManager wifiManager;
  wifiManager.setTimeout(300);
````