tzapu/WiFiManager #1086: Respond to client save config before going to STA?

OBorce, 2020-06-27:
Is it possible for the portal to wait and get a response whether or not the configuration succeeded?

tablatronix, 2020-06-27:
The portal actually shuts down to connect, so e have to relaunch it and have the client ajax wait and reconnect. It is possible but has not been implemented yet.

OBorce, 2020-06-27:
Not very knowledgeable in this, just starting out, but is it possible to have it as simultaneous AP + STA
thus being able to both serve the portal and try to connect to WiFi at the same time?
so as to not lose connection?

tablatronix, 2020-06-27:
sta is very unstable when in that mode, connections are not as reliable

OBorce, 2020-06-27:
unstable as in it might not connect, or will connect but can disconnected?
if the second I was thinking of using it as in just to confirm that the credentials work, send response to the portal that it has been configured, and restart as STA only.

tablatronix, 2020-06-27:
Connections fails, because you are locked on the same channel as the softap.

OBorce, 2020-06-27:
so there is no solution to knowing that the connection failed without manually checking?

tablatronix, 2020-06-27:
Like I said, I have not come up with a decent solution for this yet. So it has not been implemented. I mean you could always do something in your code..

Zodiac-69, 2020-08-06:
Have you looked at how the guy's on ESPEasy - Letscontrolit do it?
They keep AP and STA mode on after connection to a you wifi network, you can then see the IP address that you are connected to on your Wifi network and still access the unit via your original access unit on 192.168.4.1

tablatronix, 2020-08-06:
That is the solution, but it is not the best solution as I have explained below, it can cause connection failures.

closed 2020-11-16 by OBorce