tzapu/WiFiManager #1088: getParam() example

dontpanic21, 2020-07-03:
Hi,

could you please update the "Advanced.ino" example to show how to retrieve custom parameters on normal startup.

This example includes the function getParam() however I've found that if I attempt to use this function on normal startup (ie after the wifi and parameter info has been saved and the device resets) it causes a crash.

Thanks

tablatronix, 2020-07-03:
are you saving the parameters? You have to save and get them yourself, wm doesnt

dontpanic21, 2020-07-03:
Hi, thanks for the reply.

I'm pretty much just following the Advanced.ino example, I have not added
an *additional* save function to that.

I do have the save webpage enabled, I enter the wifi, password and the
radiobutton parameter and hit "save". At that point in the serial monitor
output I see that the parameter is saved with the expected value. I can
only assume from that that the parameter is saved.

The save function also does a reset, I let it reboot and at that point I'm
trying to use getParam() which is where it crashes.

thanks,
Will.

tablatronix, 2020-07-03:
you have to look at the spiffs examples, you ahve to save the params somewhere.

dontpanic21, 2020-07-03:
Ok thanks, I'll do that, but just to clarify... if you have your extra parameters on the wifi page and you click on "save" it saves the wifi settings but **not** the parameters? you get the parameter values from the callback and then save them yourself?

tablatronix, 2020-07-04:
Yes exactly the wifi stuff is saved by the esp lib wm never saves anything on its own since it is hard to know what a user wants to use , flash , spiffs, tinyfs, cloud services etc

dontpanic21, 2020-07-04:
ok, understand. thanks for the help.

Will.

closed 2021-07-20 by dontpanic21