tzapu/WiFiManager #1090: Trying to add custom parameters but getting error!

ZebNZ, 2020-07-08:
Hi all,

I am trying to add a custom parameter for a username input (char company_username[25]).

Here is my code:
`#include <FS.h>                   //this needs to be first, or it all crashes and burns...

#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager

#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson

//define your default values here, if there are different values in config.json, they are overwritten.
char mqtt_server[40];
char mqtt_port[6] = "8080";
char blynk_token[34] = "YOUR_BLYNK_TOKEN";
char company_username[25] = "Your Company Username";

//flag for saving data
bool shouldSaveConfig = false;

//callback notifying us of the need to save config
void saveConfigCallback () {
  Serial.println("Should save config");
  shouldSaveConfig = true;
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println();

  //clean FS, for testing
  //SPIFFS.format();

  //read configuration from FS json
  Serial.println("mounting FS...");

  if (SPIFFS.begin()) {
    Serial.println("mounted file system");
    if (SPIFFS.exists("/config.json")) {
      //file exists, reading and loading
      Serial.println("reading config file");
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile) {
        Serial.println("opened config file");
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);

        configFile.readBytes(buf.get(), size);
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject(buf.get());
        json.printTo(Serial);
        if (json.success()) {
          Serial.println("\nparsed json");

          strcpy(mqtt_server, json["mqtt_server"]);
          strcpy(mqtt_port, json["mqtt_port"]);
          strcpy(blynk_token, json["blynk_token"]);
          strcpy(company_username, json["datanow_username"]);

        }
        else {
          Serial.println("failed to load json config");
        }
        configFile.close();
      }
    }
  } else {
    Serial.println("failed to mount FS");
  }
  //end read

  // The extra parameters to be configured (can be either global or just in the setup)
  // After connecting, parameter.getValue() will get you the configured value
  // id/name placeholder/prompt default length
  WiFiManagerParameter custom_mqtt_server("server", "mqtt server", mqtt_server, 40);
  WiFiManagerParameter custom_mqtt_port("port", "mqtt port", mqtt_port, 6);
  WiFiManagerParameter custom_blynk_token("blynk", "blynk token", blynk_token, 32);
  WiFiManagerParameter custom_company_username("username", "Company Username", company_username, 25);

  //WiFiManager
  //Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wifiManager;

  //set config save notify callback
  wifiManager.setSaveConfigCallback(saveConfigCallback);

  //set static ip
  wifiManager.setSTAStaticIPConfig(IPAddress(10, 0, 1, 99), IPAddress(10, 0, 1, 1), IPAddress(255, 255, 255, 0));

  //add all your parameters here
  wifiManager.addParameter(&custom_mqtt_server);
  wifiManager.addParameter(&custom_mqtt_port);
  wifiManager.addParameter(&custom_blynk_token);
  wifiManager.addParameter(&custom_company_username);

  //reset settings - for testing
  wifiManager.resetSettings();

  //set minimu quality of signal so it ignores AP's under that quality
  //defaults to 8%
  //wifiManager.setMinimumSignalQuality();

  //sets timeout until configuration portal gets turned off
  //useful to make it all retry or go to sleep
  //in seconds
  //wifiManager.setTimeout(120);

  //fetches ssid and pass and tries to connect
  //if it does not connect it starts an access point with the specified name
  //here  "AutoConnectAP"
  //and goes into a blocking loop awaiting configuration
  if (!wifiManager.autoConnect("AutoConnectAP", "password")) {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(5000);
  }

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");

  //read updated parameters
  strcpy(mqtt_server, custom_mqtt_server.getValue());
  strcpy(mqtt_port, custom_mqtt_port.getValue());
  strcpy(blynk_token, custom_blynk_token.getValue());
  strcpy(company_username, custom_company_username.getValue());

  //save the custom parameters to FS
  if (shouldSaveConfig) {
    Serial.println("saving config");
    DynamicJsonBuffer jsonBuffer;
    JsonObject& json = jsonBuffer.createObject();
    json["mqtt_server"] = mqtt_server;
    json["mqtt_port"] = mqtt_port;
    json["blynk_token"] = blynk_token;
    //json["datanow_username"] = datanow_username;

    File configFile = SPIFFS.open("/config.json", "w");
    if (!configFile) {
      Serial.println("failed to open config file for writing");
    }

    json.printTo(Serial);
    json.printTo(configFile);
    configFile.close();
    //end save
  }

  Serial.println("local ip");
  Serial.println(WiFi.localIP());

}

void loop() {
  // put your main code here, to run repeatedly:
}`

However I am getting this output in the Serial Monitor:
`mounting FS...
mounted file system
reading config file
opened config file
{"mqtt_server":"","mqtt_port":"8080","blynk_token":"rt"}
parsed json

Exception (28):
epc1=0x4021e6da epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffb80 end: 3fffffc0 offset: 0190
3ffffd10:  3fff076c 3ffffe30 3ffeee50 402013fe  
3ffffd20:  3fff076c 3ffffe30 3ffeee50 40202258  
3ffffd30:  40219840 00000000 000003e8 4020cc38  
3ffffd40:  3fff070c 3fff0754 00000000 00000000  
3ffffd50:  3ffeeffc feefeffe feefeffe feefeffe  
3ffffd60:  feefeffe feefeffe feefeffe feefeffe  
3ffffd70:  feefeffe feefeffe feefeffe feefeffe  
3ffffd80:  feefeffe feefeffe feefeffe feefeffe  
3ffffd90:  feefeffe feefeffe feefeffe feefeffe  
3ffffda0:  feefeffe feefeffe feefeffe feefeffe  
3ffffdb0:  feefeffe feefeffe feefeffe feefeffe  
3ffffdc0:  feefeffe feefeffe feefeffe feefeffe  
3ffffdd0:  feefeffe feefeffe feefeffe feefeffe  
3ffffde0:  feefeffe feefeffe feefeffe feefeffe  
3ffffdf0:  feefeffe feefeffe feefeffe feefeffe  
3ffffe00:  feefeffe feefeffe feefeffe feefeffe  
3ffffe10:  feefeffe feefeffe feefeffe feefeffe  
3ffffe20:  feefeffe feefeffe feefeffe feefeffe  
3ffffe30:  feefeffe feefeffe feefeffe feefeffe  
3ffffe40:  feefeffe feefeffe feefeffe feefeffe  
3ffffe50:  feefeffe feefeffe feefeffe feefeffe  
3ffffe60:  feefeffe feefeffe feefeffe feefeffe  
3ffffe70:  feefeffe feefeffe feefeffe feefeffe  
3ffffe80:  feefeffe feefeffe feefeffe feefeffe  
3ffffe90:  feefeffe feefeffe feefeffe feefeffe  
3ffffea0:  feefeffe feefeffe feefeffe feefeffe  
3ffffeb0:  feefeffe feefeffe feefeffe feefeffe  
3ffffec0:  feefeffe feefeffe feefeffe feefeffe  
3ffffed0:  feefeffe feefeffe feefeffe 3ffeee50  
3ffffee0:  00000038 3fff0797 feefef0a feefeffe  
3ffffef0:  feefeffe feefeffe feefeffe feefeffe  
3fffff00:  40219700 feefeffe 3fff07ac 00000200  
3fffff10:  feefeffe feefeffe feefeffe feefeffe  
3fffff20:  feefeffe feefeffe feefeffe feefeffe  
3fffff30:  feefeffe feefeffe feefeffe feefeffe  
3fffff40:  feefeffe feefeffe feefeffe feefeffe  
3fffff50:  feefeffe feefeffe feefeffe feefeffe  
3fffff60:  feefeffe feefeffe feefeffe feefeffe  
3fffff70:  feefeffe feefeffe feefeffe feefeffe  
3fffff80:  00000000 feefeffe feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3ffeef80  
3fffffa0:  3fffdad0 00000000 3ffeef40 4020e9b8  
3fffffb0:  feefeffe feefeffe 3ffe8588 40100c39  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
va5432625
~ld`

When I remove the lines setting up the username feature it works fine! Am I meant to make changes in the .h or .cpp files? (I have only changed the .ino file)

Any help with this would be awesome!

Thanks,

Zeb

ZebNZ, 2020-07-08:
Appears to be faulty ESP8266! Sorry!

closed 2020-07-08 by ZebNZ

tablatronix, 2020-07-08:
Might be bad flash, do a full erase, that exception is usually bad memory access so something is out of scope or corrupt

ZebNZ, 2020-07-08:
See #1092