tzapu/WiFiManager #1092: Code is wrecking my ESP8266s?

ZebNZ, 2020-07-08:
Hi all,

I am running the Auto Connect code with some custom parameters.

For testing I use the " wifiManager.resetSettings(); //reset all saved credentials" line, **but after maybe 20 times of doing this my ESP8266 will just reboot and reboot with error**:
```
mounting FS...
mounted file system
reading config file
opened config file
{"mqtt_server":"","mqtt_port":"8080","blynk_token":"rt"}
parsed json

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4021e868 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd60 end: 3fffffc0 offset: 0190
3ffffef0:  3fff09ec 3fffff4c 3ffeef90 4020144a  
3fffff00:  3fff09ec 3fffff4c 3ffeef90 4020224e  
3fffff10:  402199a0 00000000 000003e8 4020cbfc  
3fffff20:  3fff098c 3fff09d4 00000000 00000000  
3fffff30:  3ffef240 3ffeef90 00000038 3fff0a17  
3fffff40:  feefef0a feefeffe feefeffe 40219860  
3fffff50:  feefeffe 3fff0a2c 00000200 feefeffe  
3fffff60:  feefeffe feefeffe feefeffe feefeffe  
3fffff70:  feefeffe feefeffe feefeffe feefeffe  
3fffff80:  00000000 feefeffe feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3ffef0c0  
3fffffa0:  3fffdad0 00000000 3ffef080 4020e97c  
3fffffb0:  feefeffe feefeffe 3ffe8530 40100dbd  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3664, room 16 
tail 0
chksum 0xee
csum 0xee
v39c79d9b
~ld
```

I then plug in another ESP8266 and the code **works fine for about another 20 cycles, before doing the same thing!**

I have tried **clearing EEPROM** but no success in restoring my ESP8266s!

Any ideas on what's going wrong? I can provide all code if that's helpful? (it is just the Auto Connect code with some custom parameters and works great for around 20 cycles......)

ZebNZ, 2020-07-08:
The other interesting thing is that it uploads to the board fine, but in the Serial Output it says
`{"mqtt_server":"","mqtt_port":"8080","blynk_token":"rt"}`

It should display:
{"datanow_username":"user23","datanow_password":"12345678"}

Would this indicate that the code is not uploading correctly? (if I upload the code to a brand new ESP8266 it displays what it should so code seems to be written correctly)

tablatronix, 2020-07-08:
erase flash

ZebNZ, 2020-07-08:
Hi @tablatronix 

I have flashed this code and the flashed my original code and still no success! 
https://github.com/kentaylor/EraseEsp8266Flash

Any other ideas?

ZebNZ, 2020-07-08:
I have tried running a small but of code with a Serial.println and it works fine, I then upload my WiFiManager code and I get the same error (#1).

```
void setup() {
  Serial.begin(115200);
}

void loop() {
  Serial.println("test");
  delay(1000);
}
```

I have tried clearing the EEPROM again and the flashing my original code but still no success!
```
#include <EEPROM.h>

void setup() {
  Serial.begin(115200);
  EEPROM.begin(512);
  // write a 0 to all 512 bytes of the EEPROM
  for (int i = 0; i < 512; i++) {
    EEPROM.write(i, 0);
  }

  // turn the LED on when we're done
  pinMode(13, OUTPUT);
  digitalWrite(13, HIGH);
  EEPROM.end();
}

void loop() {
  Serial.println("done");
  delay(1000);
}
```

ZebNZ, 2020-07-09:
Fixed!

Go to your Arduino IDE then Tools > Flash > Erase Flash and tick the checkbox that says "All Flash Contents" and flash the code.
[image]

Once the code has finished flashing you can set this option back to "Only Sketch".

Thanks,

Zeb

closed 2020-07-09 by ZebNZ