tzapu/WiFiManager #1093: ESP32 Compatible?

ZebNZ, 2020-07-09:
Hi all,

I see in the .cpp file it has:
#if defined(ESP8266) || defined(ESP32)
My question is can this work with an ESP32? If ti can what do I need to do?

I know I would need to use different libraries.....
#include <ESP8266WiFi.h>          
replace with:
#include <WiFi.h>          

#include <DNSServer.h>

#include <ESP8266WebServer.h>
replace with:
#include <WebServer.h>

#include <WiFiManager.h> 

Thanks,

Zeb

ZebNZ, 2020-07-09:
It looks like in the library it decides what libraries to use based on what you define (ESP8266 or ESP32), so I would only need to change the libraries in the .ino file.

So how do you define what board you're using?

ZebNZ, 2020-07-09:
All working :)

Just initialize your libraries like this:
#include <FS.h>
#include "SPIFFS.h"
#include <WebServer.h>
#include <DNSServer.h>
#include <WiFi.h>
#include <WiFiManager.h>
#include <ArduinoJson.h>

And you might need to add this at the top of the setup()
```
  bool begin(bool formatOnFail=true, const char * basePath="/spiffs", uint8_t maxOpenFiles=10);
  SPIFFS.begin(true); //if it has started
```

Thanks,

Zeb

closed 2020-07-09 by ZebNZ

tablatronix, 2020-07-09:
The development branch fully supports both esp8266 and esp32, there is nothing you should need to do special

ZebNZ, 2020-07-09:
@tablatronix that's awesome! Thanks for your help!