tzapu/WiFiManager #1095: Updating code to JSON 5

ZebNZ, 2020-07-10:
Hi all,

I am using JSON 6 but this code is using JSON 5 commands. How easily can I convert it to JSON 6? is there anything that needs to be changed in the libraries?

Thanks,

Zeb

tablatronix, 2020-07-10:
which code?

ZebNZ, 2020-07-10:
AutoConnect with Parameters

ZebNZ, 2020-07-10:
@tablatronix here's the link
[https://github.com/tzapu/WiFiManager/blob/master/examples/AutoConnectWithFSParameters/AutoConnectWithFSParameters.ino](https://github.com/tzapu/WiFiManager/blob/master/examples/AutoConnectWithFSParameters/AutoConnectWithFSParameters.ino)

tablatronix, 2020-07-10:
Sorry, I forgot this is in hotfixes and has not been released.

check the hotfixes branch there should fixed versions.

tablatronix, 2020-07-10:
- [ ] release .16 with json fixes

ZebNZ, 2020-07-10:
@tablatronix fixed, thank you!

closed 2020-07-10 by ZebNZ

sergionaf, 2020-12-01:
Where can I find the release .16 wiht json fixes ?

abdoshamma, 2021-08-03:
Did you find it anywhere?